ros-navigation/navigation2 #1814: Unit tests for lifecycle node transition callbacks

gimait, 2020-06-19:
## Feature request

#### Feature description
Some of the lifecycle nodes have a low test coverage, and the ones that have some tests (`nav2_costmap_2d` and `nav2_map_server`), are covered indirectly via integration test.
Testing lifecycle nodes is not easy, as specific transition callbacks are not accessible through client interfaces (e.g, `on_error` overrides cannot be tested on a running lifecycle node, since the transitions to ErrorProcessing state are not exposed to client nodes).
To improve the test coverage, unit tests can be written to verify that each transition callback works as expected.

The following lifecycle nodes do not have specific tests (they are only covered by integration tests in some cases), and can benefit from unit testing:
- [ ] `AmclNode`
- [ ] `BtNavigator`
- [ ] `ControllerServer`
- [ ] `Costmap2DROS`
- [ ] `MapSaver`
- [ ] `MapServer`
- [ ] `PlannerServer`
- [ ] `RecoveryServer`
- [ ] `WaypointFollower` 

#### Implementation considerations
The following overrides should be tested on all lifecycle nodes:
`on_configure`, `on_activate`, `on_deactivate`, `on_cleanup`, `on_shutdown`, `on_error`

To unit test these functions, a mock class inheriting from each lifecycle node class should be implemented. This mock class should expose the functions and parameters necessary for testing.

SteveMacenski, 2020-06-19:
I suppose what do you propose to test in the lifecycle transitions? Covering the transition methods via integration tests isn't the best, but I don't think its terrible. We have the up-down tests to test the lifecycle nodes more aggressively by transitioning a server all the way up and down several times to make sure the memory allocations and networking interfaces are correct.

I'm always up for more testing so I think we just need to figure out what makes most sense and as we're writing tests between transitions of each server, what is it we're looking for for "success" or "failure" criteria. It's not immediately clear to me what I should be expecting before and after an `on_configure()` that we could externally test / verify that wouldn't involve looking at specific variables being allocated (which doesn't scale).

gimait, 2020-06-20:
I guess that what I intend to do here is to test the specific implementation of each transition for each node. This will involve testing specific variables in most cases, and while this might not sound as the most useful test, it is quite valuable, since it will make possible to test the code of the lifecycle transitions that is done in this repository (the lifecycle nodes are tested upstream, so I find more useful covering the specific functions of each node).
In any case, I agree that integration tests should be done. I just thought that this would be helpful and easier to define, implement and control.

SteveMacenski, 2020-06-22:
My concern is that over time as new variables are added, I think everyone will forget to update the tests because they're so nit-picky on the specific variables rather than testing that some cumulative state has been reached.

It would be nice if we had a way to test between transitions specific "types" of class variables. E.g. after configuration, check that all params are set but all timers and networking interfaces are null, then after activate that all are valid and that no new memory allocations happened in activate. But I dont think C++ has a way to iterate over all class member variables and check the types to do something like that on a general scale.

gimait, 2020-06-23:
That is a bit what I had in mind. I saw that nodes like `AmclNode` gets some parameters, initializes publishers/subscribers/services, and does other work in `on_configure`. In other transitions, I can imagine that there might be some other logics that can be tested (maybe exception handling, communication with other nodes/services? I don't know atm). But I don't know, I thought it's worth setting up some tests, at least where there is something happening. What do you think?

> But I dont think C++ has a way to iterate over all class member variables and check the types to do something like that on a general scale.

Yes, I agree it's not very scalable. But well, it can help preventing people from deleting or modifying variables, at least they will have to double check that they didn't delete something accidentally. 

I don't know maybe it's not so useful. I just have a tendency to think that all tests are good, even those checking constants. I cannot tell how many times I reported bugs because someone messed up with one or another default variable, for the most misterious reasons haha

SteveMacenski, 2020-06-23:
Perhaps, maybe it worth giving me a bullet poiint proposal for what you'd want to do here for an example server like AMCL or whatever you'd like to do.

gimait, 2020-06-30:
I'm sorry! I haven't had much time for this lately! I'll try to update the description during this week.

SteveMacenski, 2020-08-13:
Completed with my last testing sprint

closed 2020-08-13 by SteveMacenski