ros-navigation/navigation2 #1818: nav2_util/dump_params produces incorrect results for array type params

naiveHobo, 2020-06-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - master

#### Steps to reproduce issue

Run the following test node:
```python3
import rclpy
from rclpy.node import Node

class TestDumpParamsNode(Node):

    def __init__(self):
        Node.__init__(self, 'test_dump_params')
        self.declare_parameter('param_bool_array', [True, False])
        self.declare_parameter('param_int_array',  [1, 2, 3])
        self.declare_parameter('param_double_array', [1.50000, 23.50120, 123.0010])
        self.declare_parameter('param_string_array', ['foo', 'bar'])
        self.declare_parameter('param_byte_array', [b'\x01', b'\x02', b'\x03', b'\x04'])

def main(args=None):
    rclpy.init(args=args)

    node = TestDumpParamsNode()
    try:
        rclpy.spin(node)
    except KeyboardInterrupt:
        pass
    finally:
        node.destroy_node()
        rclpy.shutdown()

if __name__ == '__main__':
    main()
```

Run `dump_params`:
```bash
ros2 run nav2_util dump_params -n test_dump_params
```
#### Expected behavior
```yaml
test_dump_params:
  ros__parameters:
    use_sim_time: False
    param_bool_array: [True, False]
    param_int_array: [1, 2, 3]
    param_double_array: [1.5, 23.5012, 123.001]
    param_string_array: ["foo", "bar"]
    param_byte_array: [0x1, 0x2, 0x3, 0x4]
```

#### Actual behavior
```yaml
test_dump_params:
  ros__parameters:
    use_sim_time: False
    param_bool_array: []
    param_int_array: []
    param_double_array: []
    param_string_array: ["foo", "bar"]
    param_byte_array: []
```

SteveMacenski, 2020-06-22:
You may want to check if this is an rclpy issue or if this still happens when declared through rclcpp

naiveHobo, 2020-06-22:
I did, it's definitely an issue with `dump_params`.

In this line, `num_items` is set as the size of the `string_array_value` vector. This vector is empty if the parameter is of any other type, so the following loop doesn't run at all.
https://github.com/ros-planning/navigation2/blob/0fa309a2f17d2689045187b4f144505a981e5e53/nav2_util/src/dump_params.cpp#L182

SteveMacenski, 2020-06-22:
Ah ok. Does your PR resolve?

naiveHobo, 2020-06-22:
Yup, the PR solves this issue and also refactors some unnecessary code. I've added tests as well so it looks like everything works as expected now.

closed 2020-06-22 by SteveMacenski