ros-navigation/navigation2 #1819: RoundRobin BT node logic doesn't match the description

naiveHobo, 2020-06-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 -->
- Version or commit hash:
  - master

`RoundRobin` BT node's description states:
> This is a custom control node introduced to the Behavior Tree. When this node is ticked, it will tick the first child until it returns `SUCCESS` or `FAILURE`. If the child returns either `SUCCESS` or `FAILURE`, it will tick its next child. Once the node reaches the last child, it will restart ticking from the first child. The main difference between the `RoundRobin` node versus the `Sequence` node is that when a child returns `FAILURE` the `RoundRobin` node will tick the next child but in the `Sequence` node, it will return `FAILURE`.

However, currently the `RoundRobin` node returns failure when a child returns failure:
https://github.com/ros-planning/navigation2/blob/0fa309a2f17d2689045187b4f144505a981e5e53/nav2_behavior_tree/plugins/control/round_robin_node.cpp#L52-L54

#### Expected behavior
If the current child returns `FAILURE`, `RoundRobin` should tick the next child. If all children return `FAILURE`, `RoundRobin` should return `FAILURE`.

#### Actual behavior
If current child returns `FAILURE`, `RoundRobin` returns `FAILURE` and ticks the same child in the next iteration.

SteveMacenski, 2020-06-22:
It also halts children on success, I think that's semantics to make sure anything running under it is stopped. I think that it could be removed as long as https://github.com/ros-planning/navigation2/blob/0fa309a2f17d2689045187b4f144505a981e5e53/nav2_behavior_tree/plugins/control/round_robin_node.cpp#L39 reverts to an `IDLE` state for children if returns success / failure

SteveMacenski, 2020-06-22:
Either way the next node ticks up https://github.com/ros-planning/navigation2/blob/0fa309a2f17d2689045187b4f144505a981e5e53/nav2_behavior_tree/plugins/control/round_robin_node.cpp#L44-L47

naiveHobo, 2020-06-22:
But the next node tick only when the current child returns success. If the current child fails, all children are halted and in the next tick the same child is ticked again.

The contradicting line I'm focusing on in the description is:
> The main difference between the RoundRobin node versus the Sequence node is that when a child returns FAILURE the RoundRobin node will tick the next child but in the Sequence node, it will return FAILURE.

Based on the description, I would expect `tick()` to look like:
```cpp
BT::NodeStatus tick() override
{
  const unsigned num_children = children_nodes_.size();

  setStatus(BT::NodeStatus::RUNNING);

  while (num_failed_children_ < num_children) {

    TreeNode * child_node = children_nodes_[current_child_idx_];
    const BT::NodeStatus child_status = child_node->executeTick();

    switch (child_status) {
      case BT::NodeStatus::SUCCESS:
      {
        // Wrap around to the first child
        if (++current_child_idx_ == num_children) {
          current_child_idx_ = 0;
        }

        num_failed_children_ = 0;
        ControlNode::haltChildren();
        return BT::NodeStatus::SUCCESS;
      }

      case BT::NodeStatus::FAILURE:
      {
        if (++current_child_idx_ == num_children) {
          current_child_idx_ = 0;
        }
        num_failed_children_++;
        break;
      }

      case BT::NodeStatus::RUNNING:
      {
        return BT::NodeStatus::RUNNING;
      }

      default:
      {
        throw BT::LogicError("Invalid status return from BT node");
      }
    }
  }

  return BT::NodeStatus::FAILURE;
}
```

SteveMacenski, 2020-06-22:
I believe you are correct. Now the question is should we update the docs or update the node to do what the docs say

naiveHobo, 2020-06-22:
I think I agree with the logic in the description. `RoundRobin` should go around ticking each child till one returns `SUCCESS` or all fail. Updating the node according to the description would be more in line with the name `RoundRobin`.

SteveMacenski, 2020-06-22:
OK, please update then :-)

closed 2020-06-23 by SteveMacenski