ros-navigation/navigation2 #1824: Docker Images not Building on Docker Hub

Michael-Equi, 2020-06-23:
Looks like no new images are being built here https://hub.docker.com/r/rosplanning/navigation2 and the latest tag seems to be missing.

SteveMacenski, 2020-06-23:
@ruffsl did we disable build triggers during our issues with master?

ruffsl, 2020-06-23:
Looks like docker hub repository link triggers now only work if auto build is explicitly enabled for the tag. I've just re-enabled auto build for the tags. I disabled the auto build before as it then also triggers whenever a new push is made to the respective master branch, and thus the consecutive queue of builds can back up. We could potentially have the CI instead manually trigger docker hub build via web hook by a nightly scheduled cron job, rather than a push to github, to avoid frequent merges from backing up the build queue. 

Also there was never a `latest` tag to begin with for the navigation2 docker hub repo.

Michael-Equi, 2020-06-23:
If there is no latest tag then the docs here for pulling the container should be updated to reflect that. https://navigation.ros.org/build_instructions/index.html

Shru, 2020-06-23:
Only master.release and master.debug tags work. Also there are errors in build instructions(on https://navigation.ros.org/build_instructions/index.html page) in terms of certain packages not being built due to dependency issues. @SteveMacenski While writing the developer guide, can I update these too? I am testing all the steps for any errors.

SteveMacenski, 2020-06-23:
To summarize that:
- @ruffsl enabled it so we should be good to go
- @Michael-Equi mentions that  there isn't a latest tag. @ruffsl can we add a latest tag to build when we push to master? Or is that effectively the same as the `.release` and we should just update docs to point to that?
- @Shru we need to figure out above first

SteveMacenski, 2020-06-23:
The builds are running again.

closed 2020-06-23 by SteveMacenski

ruffsl, 2020-06-24:
We could add a fancy post-push hook to double tag the master.release as latest as well, but I'd just as soon suggest keeping it straightforward and update the docs instead.

SteveMacenski, 2020-06-24:
Sounds good. Want to submit a PR @Shru right now to fix that?

Shru, 2020-06-24:
@SteveMacenski Yes, submitted a PR. Also confirming your point the build instructions are working at this point of time.