ros-navigation/navigation2 #1826: Docker container does not include SLAM Toolbox

Michael-Equi, 2020-06-23:
When running `ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True` launch command with on the latest docker container I get the following error: 

`Caught exception in launch (see debug for traceback): "package 'slam_toolbox' not found, searching: ['/opt/overlay_ws/install/launch_files', '/opt/overlay_ws/install/br-perception', '/opt/overlay_ws/install/br-navigation', '/opt/overlay_ws/install/nav2_system_tests', '/opt/overlay_ws/install/nav2_bringup', '/opt/overlay_ws/install/navigation2', '/opt/overlay_ws/install/nav2_dwb_controller', '/opt/overlay_ws/install/nav2_controller', '/opt/overlay_ws/install/dwb_plugins', '/opt/overlay_ws/install/dwb_critics', '/opt/overlay_ws/install/dwb_core', '/opt/overlay_ws/install/nav_2d_utils', '/opt/overlay_ws/install/dwb_msgs', '/opt/overlay_ws/install/nav_2d_msgs', '/opt/overlay_ws/install/nav2_waypoint_follower', '/opt/overlay_ws/install/nav2_recoveries', '/opt/overlay_ws/install/nav2_planner', '/opt/overlay_ws/install/nav2_navfn_planner', '/opt/overlay_ws/install/nav2_core', '/opt/overlay_ws/install/costmap_queue', '/opt/overlay_ws/install/nav2_costmap_2d', '/opt/overlay_ws/install/nav2_voxel_grid', '/opt/overlay_ws/install/nav2_rviz_plugins', '/opt/overlay_ws/install/nav2_map_server', '/opt/overlay_ws/install/nav2_lifecycle_manager', '/opt/overlay_ws/install/nav2_bt_navigator', '/opt/overlay_ws/install/nav2_behavior_tree', '/opt/overlay_ws/install/nav2_amcl', '/opt/overlay_ws/install/nav2_util', '/opt/overlay_ws/install/nav2_msgs', '/opt/overlay_ws/install/nav2_gazebo_spawner', '/opt/overlay_ws/install/nav2_common', '/opt/underlay_ws/install/vision_opencv', '/opt/underlay_ws/install/opencv_tests', '/opt/underlay_ws/install/image_common', '/opt/underlay_ws/install/gazebo_ros_pkgs', '/opt/underlay_ws/install/gazebo_plugins', '/opt/underlay_ws/install/image_transport', '/opt/underlay_ws/install/image_geometry', '/opt/underlay_ws/install/gazebo_ros', '/opt/underlay_ws/install/gazebo_msgs', '/opt/underlay_ws/install/gazebo_dev', '/opt/underlay_ws/install/cv_bridge', '/opt/underlay_ws/install/camera_info_manager', '/opt/underlay_ws/install/camera_calibration_parsers', '/opt/underlay_ws/install/behaviortree_cpp_v3', '/opt/underlay_ws/install/angles', '/opt/ros/foxy']"`

Without the slam:=True flag the file launches fine.

SteveMacenski, 2020-06-23:
@Michael-Equi please add Slam toolbox's rosdep key to https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/bringup/package.xml. That will have it installed on released distributions.

Michael-Equi, 2020-06-23:
Would you suggest closing the PR and that I just tackle it that way?

SteveMacenski, 2020-06-23:
See my comment in the PR, I think you just need to add it to the ros2_dependencies file

closed 2020-07-01 by Michael-Equi