ros-navigation/navigation2 #1828: [Eloquent] BT.CPP Updated in Eloquent, breaking builds

SteveMacenski, 2020-06-24:
Per http://build.ros2.org/job/Ebin_uB64__nav2_behavior_tree__ubuntu_bionic_amd64__binary/26/console it looks like our eloquent release for navigation2's behavior tree library is failing due to the updates in BT.CPP that broke API. 

These are the same updates we made in master already but need to be updated in Eloquent and released again to make the builds succeed ( see red in README chart).

iluetkeb, 2020-06-25:
Just for understanding: How is that API-breaking changes in BT.CPP, or any package, made it into an already released-distribution?

SteveMacenski, 2020-06-25:
By someone releasing updates, pretty simple 😉

closed 2020-06-29 by SteveMacenski

eliasdc, 2020-07-01:
Looks like they reverted back to the old BT.CPP version for Eloquent.

https://index.ros.org/p/behaviortree_cpp_v3/#eloquent-overview states v3.1.1 instead of v3.5.1. So the builds are failing again on Eloquent. For foxy everything is okay as there the v3.5.1 is used.

Or am I missing something? I can't update my Eloquent package to 3.5.1 (`ros-eloquent-behaviortree-cpp-v3`).

SteveMacenski, 2020-07-01:
https://github.com/ros/rosdistro/blob/master/eloquent/distribution.yaml#L256

Eloquent is running on v3.5.1 and when looking at the debian statuses in the readme https://github.com/ros-planning/navigation2 now everything builds, specifically Nav2 BT http://build.ros2.org/job/Ebin_uB64__nav2_behavior_tree__ubuntu_bionic_amd64__binary/ which was failing until now.

Where do you see builds failing? The PR builder and release mechanism shows that things are building fine. Perhaps you need to `apt update`.

eliasdc, 2020-07-03:
after `apt update` I'm still not seeing the newer version:
```
ros-eloquent-behaviortree-cpp-v3 is already the newest version (3.1.1-1bionic.20191213.041642).
```
But this could be an issue of cache server which aren't updated yet. So I reverted back a few commits and will update my branch once it fails for this reason.