ros-navigation/navigation2 #1830: Port range sensor costmap layer from ROS1

Michael-Equi, 2020-06-26:
# Feature request

#### Feature description
Bring over the range sensor layer for ultrasonic and single point TOF/laser range sensors from here http://wiki.ros.org/range_sensor_layer

#### Implementation considerations
Planning on keeping to the same general framework as the ROS1 version but moving to the nav2_costmap_2d directory with the proper adjustments in order to allow it to operate like the existing ROS2 costmap layers.

SteveMacenski, 2020-06-26:
@DLu do you have any issues / concerns about us absorbing this? (ex. you don't want us to, you know of some issues we should know about, there's some work somewhere to fix some bug / performance, its absolutely perfect and you'd love us to add it here)

ramilmsh, 2020-07-07:
@SteveMacenski @DLu 
I have been using that plugin and I had some problems with it (in addition to general lack of documentation):
one I can remember of the top of my head is data clearing. The range sensors are usually pretty short-range (comparatively). In my case, the robot was set up in a way that it would never knowingly approach the obstacle (min dist to the obstacle was larger than the max range of the sensor for safety reasons), so there was no way to clear it, which would cause the robot to do some weird maneuvers around obstacles that were gone. I tried to solve it by having obstacles time-out, however, the implementation was getting a little time-consuming so I never finished it (ended up making a crutch by manipulating max ranges in the message). If you think it's a good idea, I can try making a pull-request or something.

Michael-Equi, 2020-07-07:
@ramilmsh If you could provide the code you got so far, my team and I may be able to take it further and work out any issues like the one you had. Since range sensors are a core part of our robot we are willing to put in the time it takes to get it working.

SteveMacenski, 2020-07-07:
> (min dist to the obstacle was larger than the max range of the sensor for safety reasons), so there was no way to clear it

Sounds like a unique issue to you, can't help. Clearly, if you set parameters in a way that can't clear things, you're going to have issues. It sounds like you're using the incorrect parameters for your application or the incorrect hardware.

ramilmsh, 2020-07-07:
@SteveMacenski yeah, the hardware ain't great, i had to make do with what i had. my question was if that was a common issue, in which case I could make a contribution, if not - no problem

ramilmsh, 2020-07-07:
@Michael-Equi yeah, I have accidentally deleted the code long ago
but the idea was simple:
an obstacle on the costmap with global coordinates, associated time, and a hash (so that we can check if the obstacle was renewed).
obstacles are added to a simple queue and a hashset (if they already exist, we simply update the time).
on each iteration, we pop obstacles, until there is an unexpired one. for each popped one, we check the hashset, and if this was the most recent entry, we remove it from the costmap.

I have skipped over memory optimizations, but they should be straightforward.

like i said, i have already fixed the issue for us, but if it helps you, it will make me happy. feel free to follow up

simutisernestas, 2020-07-09:
@Michael-Equi do you have any progress already? I would be interested in contributing also.

Michael-Equi, 2020-07-09:
@simutisernestas Not yet. I am still  working on porting the rest of my robot from ROS1 to ROS2. I may get started on this over the weekend but for sure around Monday/Tuesday of next week. I was going to start on it earlier this week but I ended up needing to do a bunch of DevOps stuff for my team. I can let you know once I get started unless you want to get going sooner in which case you can take the lead.

Michael-Equi, 2020-07-09:
@SteveMacenski how would you suggest building in system tests for the range cost map? The turtlebot3 does not have any range sensor on it so I am currently planning on just implementing and testing my own robot. Not sure if you would like to see something else that may be more repeatable by the community.

SteveMacenski, 2020-07-09:
Right now, the system tests aren't really there to test each costmap layer (I don't think the voxel layer is used at all at the moment even due to the overhead of running a gazebo depth camera). So unit and integration tests are fine. This particular layer is easier than most to fake out for tests because its not requiring complex data to be simulated like lasers scans and dense depth images. You could just put data into it and check values for 1 1-D range sensor and that should be sufficient. The only requirement is 80%+ total coverage.

Michael-Equi, 2020-07-10:
Sounds good. I just started transitioning the simulation components of my robot to ROS2 so I can start working on the nav stuff more easily (the ROS1 bridge I have found to not all that robust). Do you know if there is even an implementation of the range sensor gazebo plugin for ROS2 yet? I cant find one but I just want to make sure I'm not missing something.

SteveMacenski, 2020-07-10:
From a quick glance this looks about right: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_ray_sensor.cpp

Though if its not, the ROS1 range sensor code is really simple and should only take an hour or less to port given all the examples in ROS2 already ported.

simutisernestas, 2020-07-10:
@Michael-Equi i've forked the old version and partially ported range sensor layer to use ros2 interface. Partially because i haven't loaded the library, i've just confirmed that it compiles without errors. But a lot of manual work is already done in terms of porting to ros2. Feel free to use that code while integrating to navigation stack (https://github.com/simutisernestas/range_sensor_layer). I will join you next week to help with testing and integration too. Please let me know if it helps.

Michael-Equi, 2020-07-10:
> From a quick glance this looks about right: https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_ray_sensor.cpp
>
> Though if its not, the ROS1 range sensor code is [...]

That works! I didn't realize it was now consolidated with the laser scanner. Looks like the documentation on this is a bit sparse.It would be really nice to have a specific example URDF that has all sensors utilized in one way or another so one could just pull from that instead of having to dig through the source code.

SteveMacenski, 2020-07-10:
:man_shrugging: I always dig through source code. Documentation is often optimistic (what they want it to do rather than what it does) or out of date. Always good to scan through things to make sure it looks like what you're looking for

DLu, 2020-07-17:
I'm fine with this moving into the trunk navigation repo. There's not a great reason for it to be separate except for the fact that it wasn't originally part of the navigation stack. 

I've had very limited experience with fully configured working sonar navigation. The other approach rather than the probabilistic one that is implemented is to just convert the most likely reading to a PointCloud and use that in a standard obstacle layer.

Michael-Equi, 2020-07-26:
I am finally getting around to doing the port. I have started working on it here https://github.com/BytesRobotics/navigation2/tree/range-sensor-costmap and have gotten the range sensor layer to build with nav2.

ramilmsh, 2020-07-26:
@Michael-Equi 
I'd like to ask you to add some more documentation. I had to read the source code to figure out how the probabilistic model works, what parameters do and so on my first time around

Michael-Equi, 2020-07-26:
@ramilmsh There aren't that many parameters and I added the ones I ported to docs/parameters and navigation.ros.org (in a PR). @DLu Would probably be the best person to document the probabilistic model since I didn't need to touch it all that much in the port and I don't want to have any false understandings which make it into the documentation on what is "supposed" to happen internally.

SteveMacenski, 2020-08-13:
Completed

closed 2020-08-13 by SteveMacenski