ros-navigation/navigation2 #1834: Question about how to add a new feature as a plugin to nav_core

rjshim, 2020-06-30:
## Question

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Version or commit hash:
  - Lastest master branch 
- DDS implementation:
  - Default Fast-RTPS

In this repository, most of the functions used for nav_core are made of plugins (sometimes even those plugins are also made of other plugins). When you make a new plugin for a new feature that you want to implement, what you have to do at the end is exporting the path of the plugin.xml file.

My questions here is, in this repository, you added _**pluginlib_export_plugin_description_file(plugin_file_name.xml)**_ lines in **_CMakeLists.txt_** instead of adding **_<planner_name plugin="${prefix}/plugin_file_name.xml" />_** in _**package.xml.**_ I remember I used to use the latter method to use plugins in ROS1 but it seems, and also I tested and found out that,  only the former method works in ROS2 (the latter one does not work. I get an error message saying "The plugin failed to load for some reason"). Do you have any idea about why (or what's the difference between those two methods)?

Any advice or help would be appreciated!
Ryan

SteveMacenski, 2020-06-30:
Its "and" not "or"

https://github.com/ros-planning/navigation2/blob/master/nav2_navfn_planner/CMakeLists.txt#L58
https://github.com/ros-planning/navigation2/blob/master/nav2_navfn_planner/package.xml#L34

rjshim, 2020-06-30:
@SteveMacenski 

Thanks for the quick response Steve. If you see the case below, it doesn't have anything in package.xml.

https://github.com/ros-planning/navigation2/blob/master/nav2_dwb_controller/dwb_plugins/CMakeLists.txt#L79
https://github.com/ros-planning/navigation2/blob/master/nav2_dwb_controller/dwb_plugins/package.xml#L27

and also when I tested the same thing, it still worked without anything exported in package.xml.

SteveMacenski, 2020-06-30:
Ah, I wasn't aware of that myself. Just may be lack of knowledge that that need was removed in some ROS2 version

SteveMacenski, 2020-06-30:
Can this be closed?

rjshim, 2020-07-01:
Oh thanks ! I will :) Thank you for the help and answers.

closed 2020-07-01 by rjshim