ros-navigation/navigation2 #1836: Dump params test fails and brings down CI

SteveMacenski, 2020-06-30:
Right now the test fails to parse something and then crashes, we should A) fix the test so that if it fails, it doesn't crash but just returns fail, and B) fix the test / code so that it works. We discussed adding a retry, that would be a good first step.

https://app.circleci.com/pipelines/github/ros-planning/navigation2/3202/workflows/35825cca-d5b2-46bb-b308-1bf40334f845/jobs/12571/steps

naiveHobo, 2020-07-02:
> We discussed adding a retry, that would be a good first step.

Retries don't seem to be helping locally. The tests are failing most of the times.

A thing to note is that the tests ALWAYS pass when using cyclonedds.

SteveMacenski, 2020-07-02:
yeah sounds like the #1772 issue :-(

SteveMacenski, 2020-07-02:
How many retries would you think it would take to pass 80% of the time after exhausting all of them?

SteveMacenski, 2020-07-06:
@naiveHobo what's the status here? this has become a problem and we should think about disabling the test if we can't get a way for it to pass regularly.

naiveHobo, 2020-07-07:
> How many retries would you think it would take to pass 80% of the time after exhausting all of them?

I haven't been able to come to a number for this, the test fails randomly most of the time locally.

> this has become a problem and we should think about disabling the test if we can't get a way for it to pass regularly.

I think disabling is the way to go right now. The way the `dump_params` tests are written right now seems like the best way to test the utility (using `launch_testing` and checking the process outputs). To get around the failing tests we would have to change the test structure and I don't think that's a good idea in this case. And even that doesn't guarantee that the tests will work since the main problem is failing services due to Fast-RTPS.

Let's disable the test till the time a solution can be found for the failing services.

closed 2020-07-23 by SteveMacenski