ros-navigation/navigation2 #1842: Update default behaviour tree

shrijitsingh99, 2020-07-01:
A lot of new behavior tree nodes have been added thanks to @naiveHobo and @gimait.
It would make sense to add some of these nodes to the default BT so as to promote people to use them, else there is a high chance they will go unnoticed.

The new nodes which have been added are:
1. GoalUpdated Condition 
2. DistanceController Decorator
3. SpeedController Decorator

Let me know if I missed something.

SteveMacenski, 2020-07-02:
I think we also have a TF node I added that hasn't seen much use, but maybe it doesn't belong here either.

So we should discuss a new behavior tree for use in navigation2. I think from those new nodes, its clearly that the default BT should have preemptable recovery behaviors using the goal updated information. We should have a recovery state where we throw all the recoveries at the wall to see what fixes the problem, but also the contextual recoveries for specific subtrees (e.g. if planner fails, do planner-specific recoveries before escalating). 

I like the idea of a retry on the main algorithmic branches so that we don't enter the recovery subtree until we've really tried a few times. I think the throttling should be done with the speed controller.

Any other new ideas or things for the BT? It could also involve new nodes if required or new recovery behaviors (more is good)

SteveMacenski, 2021-02-25:
We now have updated the BT for preemptable recoveries and recovery subtree! Closing since I think we met the spirit of this work. There's always more improvements we can make, but those were the big ones I wanted to discuss in this ticket.

closed 2021-02-25 by SteveMacenski