ros-navigation/navigation2 #1846: Add nav2_system_test for clearing costmaps

SteveMacenski, 2020-07-02:
This would have shown us very quickly the regression in Fast_RTPS like in #1772 if we had some service test coverage. We also have action-based recovery tests in the `recoveries/` directory, I think we need to make a system test for clearing costmaps. This should be relatively simple:

- Create a costmap object 
- Create a test to call the clear service based on it
- test that it was cleared introspecting the object. 

The BT node unit tests test that the BT node works and the costmap unit tests shows that the clearing works, we need this to just close the loop that the intermediate process from the BT to the object works.

tonynajjar, 2020-11-20:
As discussed on Slack, I'll take this

SteveMacenski, 2020-12-14:
@tonynajjar any updates?

tonynajjar, 2020-12-18:
Sorry for not updating earlier. I must admit, I gave it a first try and it turned out to require more prior knowledge than I imagined. I'm starting my vacation from my job today so will try to get back on it.

tonynajjar, 2021-01-06:
Can you clarify what I need to test exactly.

I see the test test_clear_costmap_service.cpp which in fact tests that the BT node works but I don't see a costmap unit test where the actual clearing behaviour is tested. Is that what I should be testing, with no involvement of the BT?

SteveMacenski, 2021-01-06:
Yes, exactly. We're looking to have a costmap up and then test that we can call those services and they have the expected results, shouldn't need a BT at all

tonynajjar, 2021-01-17:
I've spent quite some time on that but unfortunately I have no good results. There are still some things I need to learn to be able to troubleshoot my blockers and now vacation is over so I have less time. Sorry but will have to leave this ticket.

SteveMacenski, 2021-01-19:
Ok, thanks for letting us know!

SteveMacenski, 2021-04-06:
Closing- this belongs in the larger testing discussion and no longer requires unique attention

closed 2021-04-06 by SteveMacenski