ros-navigation/navigation2 #1847: [Foxy release] nav2  util missing dependencies

SteveMacenski, 2020-07-02:
http://build.ros2.org/job/Fbin_uF64__nav2_util__ubuntu_focal_amd64__binary/1/consoleFull

```

15:40:35 
15:40:35   Could not find a package configuration file provided by
15:40:35   "launch_testing_ament_cmake" with any of the following names:
15:40:35 
15:40:35     launch_testing_ament_cmakeConfig.cmake
15:40:35     launch_testing_ament_cmake-config.cmake
```

Make sure any other package missing test deps have them added and issue a re-release

SteveMacenski, 2020-07-02:
I think now its only nav2_util

closed 2020-07-06 by SteveMacenski

ParthasarathyBana, 2020-07-08:
I find that these dependencies are still missing. After cloning the latest branch, I get these errors on build.

$ colcon build --symlink-install

--- stderr: nav2_util                           
CMake Error at CMakeLists.txt:47 (find_package):
  By not providing "Findlaunch_testing_ament_cmake.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "launch_testing_ament_cmake", but CMake did
  not find one.

  Could not find a package configuration file provided by
  "launch_testing_ament_cmake" with any of the following names:

    launch_testing_ament_cmakeConfig.cmake
    launch_testing_ament_cmake-config.cmake

  Add the installation prefix of "launch_testing_ament_cmake" to
  CMAKE_PREFIX_PATH or set "launch_testing_ament_cmake_DIR" to a directory
  containing one of the above files.  If "launch_testing_ament_cmake"
  provides a separate development package or SDK, be sure it has been
  installed.

make: *** [cmake_check_build_system] Error 1
---
Failed   <<< nav2_util [0.72s, exited with code 2]

$ vcs pull
=== . (git) ===
Already up to date.

$ vcs export --exact
repositories:
  navigation2:
    type: git
    url: https://github.com/ros-planning/navigation2.git
    version: 8b8807a6043aff0225f7eca90428f0233633c658