ros-navigation/navigation2 #1860: no destructor executed on nav2 planner_server node termination

MarcoLm993, 2020-07-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - commit hash: 2bb99805
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Bash 1:
```
ros2 run nav2_planner planner_server --ros-args --params-file $(ros2 pkg prefix nav2_bringup --share)/params/nav2_params.yaml
```
Bash 2:
```
ros2 lifecycle set /nav2_planner configure
```
Once the Lifecycle Transition finishes go back to Bash 1 and terminate the planner node (`Ctrl + C`).

#### Expected behavior
The destructor in the planner node and the costmap plugins is executed. When this happens you should see in Bash 1 the following log messages being printed before termination:
```
[INFO] [nav2_planner]: Destroying
[INFO] [global_costmap.global_costmap]: Destroying
```
This happens only in 2 cases:
- When the user explicitly calls the `cleanup` transition on the planner node (by calling `ros2 lifecycle set /nav2_planner cleanup` in Bash 2)
- When the user starts the nav2_planner node and then terminates it without executing the `configure` lifecycle transition

#### Actual behavior
The destructor is not executed and the output of nav2_planner is the following:
```
[INFO] [nav2_planner]: Creating
[INFO] [global_costmap.global_costmap]: Creating Costmap
[INFO] [nav2_planner]: Configuring
[INFO] [global_costmap.global_costmap]: Configuring
[INFO] [global_costmap.global_costmap]: Using plugin "static_layer"
[INFO] [global_costmap.global_costmap]: Subscribing to the map topic (/map) with transient local durability
[INFO] [global_costmap.global_costmap]: Initialized plugin "static_layer"
[INFO] [global_costmap.global_costmap]: Using plugin "obstacle_layer"
[INFO] [global_costmap.global_costmap]: Subscribed to Topics: scan
[INFO] [global_costmap.global_costmap]: Initialized plugin "obstacle_layer"
[INFO] [global_costmap.global_costmap]: Using plugin "voxel_layer"
[INFO] [global_costmap.global_costmap]: Subscribed to Topics: pointcloud
[INFO] [global_costmap.global_costmap]: Initialized plugin "voxel_layer"
[INFO] [global_costmap.global_costmap]: Using plugin "inflation_layer"
[INFO] [global_costmap.global_costmap]: Initialized plugin "inflation_layer"
[INFO] [nav2_planner]: Created global planner plugin GridBased of type nav2_navfn_planner/NavfnPlanner
[INFO] [nav2_planner]: Configuring plugin GridBased of type NavfnPlanner
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
```
In this case no destructor is executed.

#### Additional information
This behaviour happens every time the nav2_planner node is not in the `unconfigured` state before termination.
This means that even if we perform `ros2 lifecycle set /nav2_planner shutdown`, if we don't make the cleanup transition on the planner node first we will have the destructor not being executed at node termination time.

This kind of behaviour prevents the programmer to add any operation in the destructor of planners / costmaps plugins.

SteveMacenski, 2020-07-08:
Keep in mind this print statement could also just not be appearing in the terminal since the node has been / in the process of shut down. Did you check your log files if it appears?

MarcoLm993, 2020-07-09:
The log file looks to have the same content as the shell output.
Looked also at the transitions through the `/rosout` topic, if I kill the planner_server node while it is in `configured` state nothing is reported from there.

SteveMacenski, 2020-07-09:
I'm not exactly sure. The node is made as a shared pointer in the `main` function, so the memory management to call that on a clean exit should be on the shared pointer, not something we manually handle. I suppose it could be an error in our lifecycle wrapper or the ROS2 lifecycle node headers / transitions. 

I'm not exactly sure. I'd be open to your thoughts or experiments to see what we can do about it.

MarcoLm993, 2020-07-13:
I agree with you that Managed Nodes should be finalized before being killed. On the other hand operations in the destructor should be executed also in such cases when a clean transition back to unconfigured is not possible.

One possible source of error I found in the Costmap2DROS code. There the Costmap plugins are loaded from the LayeredCostamap that is made as a raw pointer (here: https://github.com/ros-planning/navigation2/blob/eloquent-devel/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_ros.hpp#L267).
This particular variable gets deleted only on_cleanup (here: https://github.com/ros-planning/navigation2/blob/eloquent-devel/nav2_costmap_2d/src/costmap_2d_ros.cpp#L248) but not in case the Costmap2DROS object itself gets destructed.

I did not try to modify the nav2 code to delete the LayeredCostmap instance explicitly in the costmap destructor and see if this solves the problem I reported, but I will try that tomorrow morning and report if this solves the problem or not.

SteveMacenski, 2020-07-13:
Again, when shutting down, you should always finalize before hitting control C. There's not a way around that issue you mention in costmap_2d, that's how you properly use the lifecycle interfaces. 

Is that what's actually causing the destructor not to be called? I have a hard time believing that because you're not mentioning a problem with the controller server which also has a costmap object.

iluetkeb, 2020-07-14:
`Costmap2DROS` holds a plain pointer to `LayeredCostmap`. Such pointers have to be deleted explicitly. Currently the only point this pointer is deleted is in `Costmap2DROS::on_cleanup`, which is *not* called in `~Costmap2DROS`.

The fact that `Costmap2DROS` does not `delete` its members in its destructor is a bug, IMHO. It's also easy to fix. Whether it's the root cause of the original problem is a different question ;-)

MarcoLm993, 2020-07-14:
I tried to delete explicitly the LayeredCostmap raw pointer content, it seems that was not the root cause of the error.

After some more tests using gdb breakpoints I got the final confirmation that the destructor method `PlannerServer::~PlannerServer` is not getting called when it receives a `signal 2` and it is not in the unconfigured state.
I also tried to set a breakpoint when a `LifecycleNode::~LifecycleNode` instance gets called to see if this was happening on `signal 2` but I observed the same behaviour as for the `PlannerServer::~PlannerServer` method.

In order to understand if this is a behaviour of the PlannerServer only or a behaviour of all the nodes that are built on the `nav2_util::LifecycleNode` class (as you were suspecting at the beginning), I decided to perform the same test on another node that is completely decoupled from the PlannerServer: the `MapServer` from `nav2_map_server`. What I found out is that this node has a very similar behaviour compared to the one of the PlannerServer. The only difference is that in this case the destructor is not executed even if we transition back to uncofigured. Instead, if we start the node and we kill it right after, the `~LifecycleNode` method gets called.
To me this indicates that it is very likely for the bug to be coming from the `LyfecycleNode` implementation in nav2_util (or even in the rclcpp API, I have no clue on this).

In order to reproduce this error (using GDB) the following steps can be followed:
```
ros2 run --prefix 'gdb --args' nav2_map_server map_server --ros-args -p yaml_filename:=/path_to_map/map.yaml
```
This will open a GDB shell where you can set the breakpoints and run the node:
```
b LifecycleNode::~LifecycleNode
run
```

In a separated terminal execute some lifecycle transitions. It doesn't matter if you transition back to `unconfigured` or not in this case:
```
ros2 lifecycle set /map_server configure
ros2 lifecycle set /map_server activate
ros2 lifecycle set /map_server deactivate
ros2 lifecycle set /map_server cleanup
```

Finally, go back to the GDB shell and send a `signal 2` to the running node.
To do this you need to first press Ctrl+C (this will be handled by GDB only, so the node will not handle this and will be simply paused by GDB without being killed).
Then in the GDB shell you can type:
```
signal 2
```
this command sends a `signal 2` to the node and starts executing it again. If the node will pass through some LifecycleNode destructors you will see that because of the breakpoints we set earlier, otherwise the process will reach the end and exit normally. In this case the destructor is not executed and we observe again the reported behaviour.

As a final note, I gave a look at the nav2_utils::LifecycleNode::~LifecycleNode implementation (here: https://github.com/ros-planning/navigation2/blob/eloquent-devel/nav2_util/src/lifecycle_node.cpp#L62) and I've found that there the termination of an active node is explicitly accounted.

SteveMacenski, 2020-07-14:
@iluetkeb per https://design.ros2.org/articles/node_lifecycle.html we clear all that info in the state transition where it should be so we can deallocate and bring the system back up. We have tested lifecycle bringup and shutdown at length to ensure that we're in the best compliance with that standard we can be.

As I've said in this thread, there is **no** support for shutting down nav without putting the lifecycle state machine in the finalized state _as is necessary when using lifecycle nodes_. 

> The fact that Costmap2DROS does not delete its members in its destructor is a bug, IMHO. It's also easy to fix. Whether it's the root cause of the original problem is a different question ;-)

It is not a bug because that is not where deallocation of state information should be in the lifecycle state machine. We need the layered costmap destroyed in cleanup so that it can be reconfigured to be brought up again in the same session when its reconfigured and a new layered costmap object is instantiated. The destructor is only called on absolute killing of the server which is not an acceptable solution. @MarcoLm993 is misusing managed nodes / navigation servers, as I've said above a couple different times.

SteveMacenski, 2020-07-14:
https://github.com/ros-planning/navigation2/blob/master/nav2_util/src/lifecycle_node.cpp#L63

I suspect that this isn't being called then to transition back down from misuse?

iluetkeb, 2020-07-14:
Firstly, I agree that what @MarcoLm993 is doing should be done during shutdown (we work together and have spoken about this privately).

That said, there is no reason not to clean up resources during destruction, and I would say that this is what most people would expect -- including yourself, if I interpret your initial comment correctly, and also the code in the destructor from LifecycleNode does it. If you agree, we can submit a PR.

SteveMacenski, 2020-07-14:
The resources should have long since been cleaned up in the `on_cleanup` state transition so there should not be anything to clean up in the destructor. If you wanted to guard all the pointers (check if they're `nullptr` before deallocating them) you could update the `on_cleanup`s to be directly called on the destructor so it'll run over it twice for a well formed application and once otherwise. 

PR for what exactly? Making the destructor of the lifecycle node wrapper work to put into finalized when it wasn't correctly? I'd be OK with that, yes.

SteveMacenski, 2020-07-16:
@iluetkeb @MarcoLm993 can you specify what you would like to propose as a solution?

MarcoLm993, 2020-07-17:
As far as I got from the discussion here @iluetkeb wanted to submit a PR to put the lifecycle node in the finalized state when `~LifecycleNode()` is executed, no matter what the node's state was at shutdown time.

Anyway, the problem that the `~LifecycleNode()` method won't be executed in some cases (see the long comment I wrote in Jul 14) stays, so you can handle everything you want in the destructor, but if it doesn't get called this won't help much.

The idea of opening this Bug Report was simply to report this weird behaviour I observed in the pipeline and made you guys aware of it.
Unfortunately I don't have a working solution yet, but if you have any idea on where the problem could be we can work on a solution together (unless you consider this to be not a bug but a desired behaviour, in this case there'll be probably nothing to do).

SteveMacenski, 2020-07-17:
OK, unfortunately, we're largely volunteer group and I am uncertain to when anyone will have time to look into this specific issue given it manifests primarily when a user doesn't properly finalize the state machines. I'm sure there's a solution someone can find down the line.

naiveHobo, 2020-07-31:
This turned out to be a huge memory leak in `planner_server`, `controller_server`, `recovery_server`, and `costmap_2d`, basically all the nodes that use pluginlib. The plugins loaded in these servers are given a `shared_ptr` to the server node for processing. When the system is brought down, the pluginlib class loader is supposed to unload the shared plugin libraries AFTER the objects created by the pluginlib class loader have been destroyed successfully, `pluginlib` usually warns us about this memory leak.

`SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.`

However, in our case the objects created by the pluginlib class loader will only be destroyed when the server node which creates these objects is destroyed successfully. But since the loaded plugins hold a `shared_ptr` to the server node, the server node is never destroyed and the loaded plugin libraries are never unloaded. This is why the destructors for `planner_server`, `controller_server`, `recovery_server`, and `costmap_2d` nodes are never executed.

I have been looking into how to solve this and one solution I can come up with is passing a `weak_ptr` to the server node object to any plugin that needs it for processing. The loaded plugin will then be responsible for locking the `weak_ptr` and using it as a `shared_ptr` when required. Thoughts?

SteveMacenski, 2020-07-31:
Can you give a run down with one of them (lets say planner server to be simple) of the lines you're mentioning? I think the thought process we had is we have shared pointers so on destruction the reference counter goes to zero and it should be deleted. If that's not the case, I'm just curious who's holding a pointer reference. https://github.com/ros-planning/navigation2/blob/master/nav2_planner/src/planner_server.cpp#L195 on cleanup of the lifecycle state there's a pretty clear clean up and then a vector clear which should delete all the plugins (assuming no other references are stored which I don't think there are?).

naiveHobo, 2020-07-31:
The problem is, clean up doesn't happen when you bring down a lifecycle node with ctrl + c. A signal 2 means that whatever stage the lifecycle node is currently in, it will transition to the finalized state and get ready for destruction (please correct me if I'm wrong here).

As @MarcoLm993 said in the description
> This behaviour happens every time the nav2_planner node is not in the `unconfigured` state before termination.

When the node is in the `unconfigured` state, there are no plugins objects existing in the heap which is why the node can be brought down safely and the destructor is executed.

As an example, I'll take up the recovery server.

- On startup, `recovery_server` is in the `unconfigured` stage and `std::vector<pluginlib::UniquePtr<nav2_core::Recovery>> recoveries_;` which hold unique pointers to the required recovery objects is empty.

- When the recovery server is configured, `on_configure` calls `loadRecoveryPlugins()` [here](https://github.com/ros-planning/navigation2/blob/ec2603960816a3b39204efb6d77ef5dff3d0a838/nav2_recoveries/src/recovery_server.cpp#L92). We use the class loader to load the required recovery plugin objects [here](https://github.com/ros-planning/navigation2/blob/ec2603960816a3b39204efb6d77ef5dff3d0a838/nav2_recoveries/src/recovery_server.cpp#L109) and the created `unique_ptrs` to the recovery plugin objects are pushed into `recoveries_`. In the [next line](https://github.com/ros-planning/navigation2/blob/ec2603960816a3b39204efb6d77ef5dff3d0a838/nav2_recoveries/src/recovery_server.cpp#L110) we configure the recovery plugin object and pass it a `shared_ptr` to the `recovery_server`. After this point, the loaded plugins exist in the heap and each recovery plugin holds a `shared_ptr` reference to the `recovery_server` node.

- At this point `recovery_server` is in the `inactive` stage. If the system is brought down here for whatever reason (let's say a signal 2 after ctrl + c), the `cleanup` lifecycle transition is never executed. `cleanup` transition is the only place in the server where we explicitly clear the loaded plugin objects.
https://github.com/ros-planning/navigation2/blob/ec2603960816a3b39204efb6d77ef5dff3d0a838/nav2_recoveries/src/recovery_server.cpp#L157-L162

On a signal 2, the node should transition to the finalized state and be destroyed which is when the destructor is supposed to be called. However, at this point all the loaded recovery plugin objects still exist in the heap since the `cleanup` transition was never called. Each recovery plugin object holds a `shared_ptr` to its parent `recovery_server` node which is why the `recovery_server` is never destroyed and the destructor isn't called. Since, the `recovery_server` destructor is not called, the shared libraries loaded by the pluginlib class loader are never unloaded (if they were unloaded at any point, we would have been prompted with that SEVERE WARNING!! since the plugin objects would still exist in the heap).

SteveMacenski, 2020-07-31:
> The problem is, clean up doesn't happen when you bring down a lifecycle node with ctrl + c. A signal 2 means that whatever stage the lifecycle node is currently in, it will transition to the finalized state and get ready for destruction (please correct me if I'm wrong here).

So we're correct in saying that then the destructors and everything is properly dealt with when being transitioned down? This is **only** for control C in active state, is that correct? Just looking for explicit verification that the intended behavior is working as intended.

Ah, got it, its the `rclcpp::Node::SharedPtr` that causes this. I understand now your weak_ptr comments, that makes sense. I was mixed up on the idea of the server "node" and the literal "node" object. That sounds like a totally reasonable solution.

naiveHobo, 2020-08-02:
I think this is an issue present in several helper classes that navigation2 uses. We are passing around `rclcpp::Node::SharedPtr` objects all over the place and every time `rclcpp::Node::shared_from_this()` is used to pass a `shared_ptr` from the parent `rclcpp::Node` to a member, we have a memory leak due to a cyclic dependency when the system is brought down without proper cleanup (which happens a lot).

It should be apparent from the minimal example below, both `~TestNode()` and `~TestClass()` are never executed after ctrl + c. When `node->init();` in `main()` is removed, `~TestNode()` is executed.

```cpp
class TestClass
{
public:
  TestClass(rclcpp::Node::SharedPtr node)
  : node_(node)
  {
    RCLCPP_INFO(node_->get_logger(), "Created TestClass");
  }

  ~TestClass()
  {
    RCLCPP_INFO(node_->get_logger(), "Destroyed TestClass");
  }

private:
  rclcpp::Node::SharedPtr node_;
};

class TestNode : public rclcpp::Node
{
public:
  TestNode()
  : rclcpp::Node("test_node")
  {
    RCLCPP_INFO(get_logger(), "Created TestNode");
  }

  ~TestNode()
  {
    RCLCPP_INFO(get_logger(), "Destroyed TestNode");
  }

  void init()
  {
    test_class_ = std::make_shared<TestClass>(shared_from_this());
  }

private:
  std::shared_ptr<TestClass> test_class_;
};

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  auto node = std::make_shared<TestNode>();
  node->init();
  rclcpp::spin(node->get_node_base_interface());
  rclcpp::shutdown();
  return 0;
}
```

closed 2020-08-25 by SteveMacenski