ros-navigation/navigation2 #1862: Navigation2 crashes on bringup

ParthasarathyBana, 2020-07-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- ROS2 Version:
  - Eloquent
- Version or commit hash:
  - 2b073964
- DDS implementation:
  - Default

#### Steps to reproduce issue
```
source ~/colcon_ws/install/setup.bash
ros2 launch nav2_bringup nav2_tb3_simulation_launch.py
```

#### Expected behavior
NavStack runs cleanly in simulation

#### Actual behavior
partha@partha:~/colcon_ws$ ros2 launch nav2_bringup nav2_tb3_simulation_launch.py 
[INFO] [launch]: All log files can be found below /home/partha/.ros/log/2020-07-08-23-51-38-235508-partha-4756
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [4778]
[INFO] [gzclient-2]: process started with pid [4779]
[INFO] [rviz2-3]: process started with pid [4780]
[INFO] [robot_state_publisher-4]: process started with pid [4781]
[ERROR] [robot_state_publisher-4]: process has died [pid 4781, exit code 127, cmd '/opt/ros/eloquent/lib/robot_state_publisher/robot_state_publisher /home/partha/colcon_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_waffle.urdf --ros-args -r __node:=robot_state_publisher --params-file /tmp/launch_params_p9gkngob        '].
[INFO] [lifecycle_manager-5]: process started with pid [4782]
[INFO] [map_server-6]: process started with pid [4783]
[INFO] [amcl-7]: process started with pid [4784]
[ERROR] [amcl-7]: process has died [pid 4784, exit code 127, cmd '/home/partha/colcon_ws/install/nav2_amcl/lib/nav2_amcl/amcl --ros-args -r __node:=amcl --params-file /tmp/tmp5jqagy_j        '].
[INFO] [controller_server-8]: process started with pid [4785]
[ERROR] [controller_server-8]: process has died [pid 4785, exit code 127, cmd '/home/partha/colcon_ws/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmp4vcddqth        '].
[INFO] [planner_server-9]: process started with pid [4786]
[INFO] [recoveries_server-10]: process started with pid [4788]
[INFO] [bt_navigator-11]: process started with pid [4794]
[INFO] [waypoint_follower-12]: process started with pid [4795]
[lifecycle_manager-5] [INFO] [lifecycle_manager]: Creating
[ERROR] [planner_server-9]: process has died [pid 4786, exit code 127, cmd '/home/partha/colcon_ws/install/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmp6bhiay2n        '].
[robot_state_publisher-4] /opt/ros/eloquent/lib/robot_state_publisher/robot_state_publisher: symbol lookup error: /opt/ros/eloquent/lib/robot_state_publisher/robot_state_publisher: undefined symbol: _ZN7tf2_ros26StaticTransformBroadcaster13sendTransformERKSt6vectorIN13geometry_msgs3msg17TransformStamped_ISaIvEEESaIS6_EE

[ERROR] [recoveries_server-10]: process has died [pid 4788, exit code 127, cmd '/home/partha/colcon_ws/install/nav2_recoveries/lib/nav2_recoveries/recoveries_server --ros-args -r __node:=recoveries_server --params-file /tmp/launch_params_diqzbibj        '].
[amcl-7] /home/partha/colcon_ws/install/nav2_amcl/lib/nav2_amcl/amcl: symbol lookup error: /home/partha/colcon_ws/install/nav2_amcl/lib/libamcl_core.so: undefined symbol: _ZTVN7tf2_ros14CreateTimerROSE

[controller_server-8] /home/partha/colcon_ws/install/nav2_controller/lib/nav2_controller/controller_server: symbol lookup error: /home/partha/colcon_ws/install/nav2_costmap_2d/lib/libnav2_costmap_2d_core.so: undefined symbol: _ZTVN7tf2_ros14CreateTimerROSE

[lifecycle_manager-5] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
[planner_server-9] /home/partha/colcon_ws/install/nav2_planner/lib/nav2_planner/planner_server: symbol lookup error: /home/partha/colcon_ws/install/nav2_costmap_2d/lib/libnav2_costmap_2d_core.so: undefined symbol: _ZTVN7tf2_ros14CreateTimerROSE
[recoveries_server-10] /home/partha/colcon_ws/install/nav2_recoveries/lib/nav2_recoveries/recoveries_server: symbol lookup error: /home/partha/colcon_ws/install/nav2_costmap_2d/lib/libnav2_costmap_2d_core.so: undefined symbol: _ZTVN7tf2_ros14CreateTimerROSE
[map_server-6] [INFO] [map_server]: Creating
[bt_navigator-11] /home/partha/colcon_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator: symbol lookup error: /home/partha/colcon_ws/install/nav2_bt_navigator/lib/libbt_navigator_core.so: undefined symbol: _ZN7tf2_ros17TransformListener21subscription_callbackESt10shared_ptrIN8tf2_msgs3msg10TFMessage_ISaIvEEEEb
[ERROR] [bt_navigator-11]: process has died [pid 4794, exit code 127, cmd '/home/partha/colcon_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator --params-file /tmp/tmp_8zw2sp_        '].
[waypoint_follower-12] [INFO] [waypoint_follower]: Creating
[rviz2-3] [rospack] Error: package 'rviz_common' not found
[rviz2-3] [librospack]: error while executing command
[gzserver-1] gzserver: symbol lookup error: /opt/ros/eloquent/lib/libgazebo_ros_init.so: undefined symbol: _ZN10gazebo_ros9ThrottlerC1Ed
[ERROR] [gzserver-1]: process has died [pid 4778, exit code 127, cmd 'gzserver -s libgazebo_ros_init.so /home/partha/colcon_ws/install/nav2_bringup/share/nav2_bringup/worlds/waffle.model'].
[rviz2-3] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [rospack] Error: package 'rviz_common' not found
[rviz2-3] [librospack]: error while executing command
[rviz2-3] [rospack] Error: package 'rviz_common' not found
[rviz2-3] [librospack]: error while executing command
[rviz2-3] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] free(): invalid pointer
[ERROR] [rviz2-3]: process has died [pid 4780, exit code -6, cmd '/opt/ros/eloquent/lib/rviz2/rviz2 -d /home/partha/colcon_ws/install/nav2_bringup/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2          '].
[INFO] [waypoint_follower-12]: sending signal 'SIGINT' to process[waypoint_follower-12]
[INFO] [map_server-6]: sending signal 'SIGINT' to process[map_server-6]
[INFO] [lifecycle_manager-5]: sending signal 'SIGINT' to process[lifecycle_manager-5]
[INFO] [gzclient-2]: sending signal 'SIGINT' to process[gzclient-2]
[waypoint_follower-12] [INFO] [rclcpp]: signal_handler(signal_value=2)
[waypoint_follower-12] [INFO] [waypoint_follower]: Destroying
[map_server-6] [INFO] [rclcpp]: signal_handler(signal_value=2)
[map_server-6] [INFO] [map_server]: Destroying
[INFO] [waypoint_follower-12]: process has finished cleanly [pid 4795]
[lifecycle_manager-5] [INFO] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-5] [INFO] [lifecycle_manager]: Destroying
[INFO] [map_server-6]: process has finished cleanly [pid 4783]
[ERROR] [gzclient-2]: process has died [pid 4779, exit code -2, cmd 'gzclient'].
[INFO] [lifecycle_manager-5]: process has finished cleanly [pid 4782]

SteveMacenski, 2020-07-09:
The first failure is 

```
 [ERROR] [robot_state_publisher-4]: process has died [pid 4781, exit code 127, cmd '/opt/ros/eloquent/lib/robot_state_publisher/robot_state_publisher /home/partha/colcon_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_waffle.urdf --ros-args -r __node:=robot_state_publisher --params-file /tmp/launch_params_p9gkngob '].
```

Of which you can see the error:

```
[robot_state_publisher-4] /opt/ros/eloquent/lib/robot_state_publisher/robot_state_publisher: symbol lookup error: /opt/ros/eloquent/lib/robot_state_publisher/robot_state_publisher: undefined symbol: _ZN7tf2_ros26StaticTransformBroadcaster13sendTransformERKSt6vectorIN13geometry_msgs3msg17TransformStamped_ISaIvEEESaIS6_EE
```

So, this isn't an issue with Navigation2, this is an issue with your robot state publisher / URDF or similar. Do you have an appropriate version of TF2 and robot state publisher being used? It seems like you might have a version mismatch. 

I'm closing since its not a Navigation2 issue, but you're welcome to follow up and we can keep working through it.

closed 2020-07-09 by SteveMacenski

ParthasarathyBana, 2020-07-09:
Thanks Steve! That was very helpful and it was the dependencies issue indeed.

The following two instructions helped me solve the issue:
```
# Add these two lines in the bashrc script
export LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/opt/ros/eloquent/lib:$LD_LIBRARY_PATH
```

SteveMacenski, 2020-07-09:
That shouldn't be needed if you sourced your opt space.