ros-navigation/navigation2 #1863: [Eloquent] Error building nav2_behavior_tree from source

wilcobonestroo, 2020-07-09:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- ROS2 Version: Eloquent
- Version or commit hash: 2b073964f166f00b0ed29c04bbcede3b278135b7
- DDS implementation: Fast-RTPS

I tried to build Navigation 2 for eloquent on a fresh Ubuntu install. I followed the steps described on: https://navigation.ros.org/build_instructions/index.html#build

`mkdir -p ~/navigation2_ws/src`
`cd ~/navigation2_ws/src`
`git clone https://github.com/ros-planning/navigation2.git --branch eloquent-devel`
`cd ~/navigation2_ws`
`rosdep install -y -r -q --from-paths src --ignore-src --rosdistro eloquent`
`colcon build --symlink-install`

The `colcon build` command gives the following error:

`/home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/src/behavior_tree_engine.cpp: In member function ‘nav2_behavior_tree::BtStatus nav2_behavior_tree::BehaviorTreeEngine::run(BT::Tree*, std::function<void()>, std::function<bool()>, std::chrono::milliseconds)’:`
`/home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/src/behavior_tree_engine.cpp:50:13: error: ‘struct BT::Tree’ has no member named ‘rootNode’; did you mean ‘root_node’?`
`       tree->rootNode()->halt();`
`             ^~~~~~~~`
`             root_node`
`/home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/src/behavior_tree_engine.cpp:54:20: error: ‘struct BT::Tree’ has no member named ‘rootNode’; did you mean ‘root_node’?`
`     result = tree->rootNode()->executeTick();`
`                    ^~~~~~~~`
`                    root_node`

It looks related to the closed issue: https://github.com/ros-planning/navigation2/issues/1828

#### Expected behavior
Successful build

#### Actual behavior
Failing build

SteveMacenski, 2020-07-09:
You're using an old version of BT.CPP, please update

wilcobonestroo, 2020-07-09:
Before doing the steps above, I installed ROS 2 by following the steps described on: https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/

When I now enter `sudo apt install ros-eloquent-behaviortree-cpp-v3` I get the message `ros-eloquent-behaviortree-cpp-v3 is already the newest version (3.1.1-1bionic.20191213.041642).` Am I using and old repo? And how can I use a newer one?

SteveMacenski, 2020-07-09:
did you apt update? https://github.com/ros/rosdistro/blob/master/eloquent/distribution.yaml#L247 3.5.1 is what eloquent has released

wilcobonestroo, 2020-07-10:
I did perform an apt update. However, I still get the 3.1.1 version. Thank you for the information. I am going to develop on a different machine, so I will try it there from scratch. If it is still not working, I will build the behavior tree from source and see if that works.

I will describe my finding here.

fmrico, 2020-07-10:
Hi @wilcobonestroo 

As @SteveMacenski said yesterday, if you want to contribute to Navigation2, you should use ROS2 Foxy.

Your problems are related to the version of the Behavior Tree library in Eloquent, different from the version in Foxy.

Don't waste your time. Update to Foxy.

wilcobonestroo, 2020-07-10:
As I understand it correctly, Steve said that I had to update the **behavior tree package** (BT.CPP).

I was aiming at Eloquent because there is a `ros-eloquent-gazebo-ros-pkgs` and no `ros-foxy-gazebo-ros-pkgs` available and we want to use Gazebo for simulation. However, according to the [documentation](http://gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros), the `ros2` branch of `gazebo_ros_pkgs` points at Foxy.

Am I correct that I can use Gazebo for simulation on Foxy if I build `gazebo_ros_pkgs` from source?

SteveMacenski, 2020-07-10:
Correct. I’m not sure why you’re not getting 3.5.1 in eloquent. Maybe there hasnt been a sync yet?

Maxcode19, 2020-07-16:
Hello everybody. I'm a beginner with ROS/ROS2 and I hope I'm joining the discussion correctly. I have:

- Operating System: Ubuntu 18.04
- ROS2 Version: Eloquent (from source)

I'm trying to build Navigation2 from source and I have bumped into a similar problem (not exactly the same) with nav2_behavior_tree. I have followed all the steps indicated in the tutorials but the build process fails. The error message is the following:

         --- stderr: nav2_behavior_tree                                         
        CMake Error at CMakeLists.txt:14 (find_package):
        By not providing "Findbehaviortree_cpp_v3.cmake" in CMAKE_MODULE_PATH this
        project has asked CMake to find a package configuration file provided by
        "behaviortree_cpp_v3", but CMake did not find one.

        Could not find a package configuration file provided by
        "behaviortree_cpp_v3" with any of the following names:

        behaviortree_cpp_v3Config.cmake
        behaviortree_cpp_v3-config.cmake

       Add the installation prefix of "behaviortree_cpp_v3" to CMAKE_PREFIX_PATH
       or set "behaviortree_cpp_v3_DIR" to a directory containing one of the above
       files.  If "behaviortree_cpp_v3" provides a separate development package or
       SDK, be sure it has been installed.

        ---
        Failed   <<< nav2_behavior_tree [6.69s, exited with code 1]
        Aborted  <<< nav2_lifecycle_manager [12.6s]            
        Aborted  <<< nav_2d_utils [14.6s]                      
        Aborted  <<< nav2_map_server [16.0s]                                    

        Summary: 7 packages finished [3min 19s]
        1 package failed: nav2_behavior_tree
        3 packages aborted: nav2_lifecycle_manager nav2_map_server nav_2d_utils
        1 package had stderr output: nav2_behavior_tree
        18 packages not processed

It seems that I have forgot the dependencies. However, before building, I had run:
              rosdep install -y -r -q --from-paths src --ignore-src --rosdistro eloquent

as required and I had got:
              #All required rosdeps installed successfully

Reading the operations performed by rosdep I have found:

...
executing command [sudo -H apt-get install -y -qq ros-eloquent-behaviortree-cpp-v3]
Selecting previously unselected package ros-eloquent-behaviortree-cpp-v3.
(Reading database ... 274576 files and directories currently installed.)
Preparing to unpack .../ros-eloquent-behaviortree-cpp-v3_3.1.1-1bionic.20191213.041642_amd64.deb ...
Unpacking ros-eloquent-behaviortree-cpp-v3 (3.1.1-1bionic.20191213.041642) ...
Setting up ros-eloquent-behaviortree-cpp-v3 (3.1.1-1bionic.20191213.041642) ...
...

I have got the version 3.1.1 version and running:
                sudo apt install ros-eloquent-behaviortree-cpp-v3
it does not change anything (I get the same result of @wilcobonestroo ). 

I don't know if I did something else wrong but I would really appreciate if you could help me.
Thank you in advance

wilcobonestroo, 2020-07-17:
I still have the same results. After `sudo apt update`, I see the following links for ros:

- Hit:5 http://packages.ros.org/ros/ubuntu bionic InRelease                                                                             
- Hit:6 http://packages.ros.org/ros2/ubuntu bionic InRelease

After `sudo apt install ros-eloquent-behaviortree-cpp-v3` I get the message `ros-eloquent-behaviortree-cpp-v3 is already the newest version (3.1.1-1bionic.20191213.041642).`

I don't how I can change or update that part.

wilcobonestroo, 2020-07-17:
I thought I had found my error, because I completely missed the part about the [Navigation2 dependencies](https://navigation.ros.org/build_instructions/index.html#build) in de documentation. So, I checked out the dependencies for eloquent. It turned out that versions in the [dependencies.repos](https://github.com/ros-planning/navigation2/blob/eloquent-devel/tools/ros2_dependencies.repos) file are not correct for eloquent. So, I manually set each of them to the appropriate  version (BehaviorTree -> master, gazebo_ros_packages -> eloquent, image_common -> eloquent). After building the dependencies and sourcing them, I still get the same error when I build the navigation2 packages. The strange thing is that the error is not referring to the BehaviorTree version that I have build in the navigation dependencies, but it refers to the one that is installed, e.g. `/opt/ros/eloquent/include/behaviortree_cpp_v3/control_node.h` and that is the old one.

The errors are

```
/home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp: In member function ‘virtual BT::NodeStatus nav2_behavior_tree::PipelineSequence::tick()’:
/home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp:85:35: error: no matching function for call to ‘nav2_behavior_tree::PipelineSequence::haltChildren()’
         ControlNode::haltChildren();
                                   ^
In file included from /home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp:18:0:
/opt/ros/eloquent/include/behaviortree_cpp_v3/control_node.h:47:10: note: candidate: void BT::ControlNode::haltChildren(size_t)
     void haltChildren(size_t i);
          ^~~~~~~~~~~~
/opt/ros/eloquent/include/behaviortree_cpp_v3/control_node.h:47:10: note:   candidate expects 1 argument, 0 provided
/home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp:109:29: error: no matching function for call to ‘nav2_behavior_tree::PipelineSequence::haltChildren()’
   ControlNode::haltChildren();
                             ^
In file included from /home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp:18:0:
/opt/ros/eloquent/include/behaviortree_cpp_v3/control_node.h:47:10: note: candidate: void BT::ControlNode::haltChildren(size_t)
     void haltChildren(size_t i);
          ^~~~~~~~~~~~
/opt/ros/eloquent/include/behaviortree_cpp_v3/control_node.h:47:10: note:   candidate expects 1 argument, 0 provided
make[2]: *** [CMakeFiles/nav2_pipeline_sequence_bt_node.dir/plugins/control/pipeline_sequence.cpp.o] Error 1
make[1]: *** [CMakeFiles/nav2_pipeline_sequence_bt_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/control/recovery_node.cpp: In member function ‘virtual BT::NodeStatus nav2_behavior_tree::RecoveryNode::tick()’:
/home/ros/navigation2_ws/src/navigation2/nav2_behavior_tree/plugins/control/recovery_node.cpp:91:43: error: no matching function for call to ‘nav2_behavior_tree::RecoveryNode::haltChildren()’
                 ControlNode::haltChildren();
```
So, it is using the version from the /opt/ros folder and the one that I have built and sourced.

SteveMacenski, 2020-07-17:
> ros2_dependencies.repos

That file is for the CI system only. By the time a distribution has been released, all of those dependencies are available in binary form for release (or else we couldn't have released navigation2). You are correct though that it would be best if we updated the dependencies.repos file so that it contained the specific versions after they were released. A PR to resolve that with the updates you made would be appreciated.

> The strange thing is that the error is not referring to the BehaviorTree version that I have build in the navigation dependencies, but it refers to the one that is installed

Did you source that workspace / is it in the same workspace? Yeah, those are definitely the errors I saw when moving BT.CPP from 3.x.x  (don't recall specific number off hand) to 3.5.1.

Digging deeper from the release,  I found that the binary jobs we failing: http://build.ros2.org/job/Ebin_uB64__behaviortree_cpp_v3__ubuntu_bionic_amd64__binary/ so it was never released in binary form to apt. However, its been fixed now, so in the next Eloquent sync, you should get the updated BT.CPP library. So this seems to just be a temporary issue during the dependency release

wilcobonestroo, 2020-07-20:
> Did you source that workspace / is it in the same workspace?

@SteveMacenski I did source the workspace with the BehaviorTree package that I had just built. However, it kept using the default installed version from my `/opt/ros/eloquent` install. I could resolve the issue by removing the installed version with `sudo apt remove ros-eloquent-behaviortree-cpp-v3`. After removing that one, navigation2 builds successfully on my machine. I don't know why it kept  using the installed version even after I sourced the correct version. Also putting the BehaviorTree package in the same workspace as Navigation 2 solves the issue. In that case, I can have the 3.1.1 version installed and Navigation still builds using the 3.5.1 version from the workspace.

@Maxcode19 If you want to build Navigation 2 manually, the easiest way is to download the BehaviorTree repo and put it in the same workspace as the Navigation 2 workspace.

Ps. After `sudo apt update` I still get the `3.1.1-1bionic.20191213.041642` version of BehaviorTree.
Ps2. I am going to focus on porting everything to foxy.
Ps3. Although I don't completely understand what causes the issues on my machine, I think this issue can be closed.

SteveMacenski, 2020-07-20:
Can't speak to your set up / source paths to tell you exactly why. My guess is you may have sourced things out of order at some point and it stuck around until a clean rebuild. 

Again, look at the link above, you will have to wait for apt install BT.CPP until the sync happens.

closed 2020-07-20 by SteveMacenski

omerts, 2022-05-19:
Anyone getting here from a google search, for me it was solved by running:
```
 sudo apt-get install ros-galactic-behaviortree-cpp-v3
 sudo apt-get install ros-galactic-ompl
```