ros-navigation/navigation2 #1865: RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.

Michael-Equi, 2020-07-13:
##  Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy bianries
- Version or commit hash:
  - 69977cd83d9040df3422d8a2e564715b6002f3fb
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
launch the bringup_launch.py file

#### Expected behavior
No warnings indicating the `RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED` is ignored

#### Actual behavior
All nodes flag `RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED` as ignored and that the buffering of log messages should be handled with `RCUTILS_LOGGING_USE_STDOUT` and `RCUTILS_LOGGING_BUFFERED_STREAM`

SteveMacenski, 2020-07-13:
PRs are appreciated, but I'm not exactly sure what you're looking for.

Michael-Equi, 2020-07-13:
> PRs are appreciated, but I'm not exactly sure what you're looking for. 

It more or less is just bugging me. Do you know if those lines are still needed? If not I can just remove them.

SteveMacenski, 2020-07-13:
https://index.ros.org/doc/ros2/Tutorials/Logging-and-logger-configuration/#line-buffered-console-output

The best I can tell, its still required

Michael-Equi, 2020-07-13:
I removed the line in my custom launch files and the output still prints to the temrinal fine. The only issue I know that it might cause is if those launch files are used directly with launch_test that reads output from STDOUT to determine wether certain conditions are met.

SteveMacenski, 2020-07-13:
You're also probably running on a desktop machine, this impacts more resource constrained robots

Michael-Equi, 2020-07-13:
I'll go ahead and replace that line with RCUTILS_LOGGING_BUFFERED_STREAM then and it should have the same behavior as that line was intended to create.

closed 2020-07-15 by SteveMacenski