ros-navigation/navigation2 #1873: Change name of default branch to main

mkhansenbot, 2020-07-14:
### Feature description

Open Robotics is moving all their ROS repos to use 'main' as the default branch name, see post here: https://discourse.ros.org/t/consider-renaming-default-branches/15385/5

I would love to see navigation2 follow suit.

#### Implementation considerations

Not sure what this affects but guessing:
 - everyone would need to re-synch personal clones / forks of the repo
 - release tools might need updating

ruffsl, 2020-07-14:
Our CI will need some attention given how the docker images are tagged to match the branch name: 

https://github.com/ros-planning/navigation2/blob/fbdace2325084c9f65c7bda8a95c4eba3ec244f1/.circleci/config.yml#L324

https://github.com/ros-planning/navigation2/blob/fbdace2325084c9f65c7bda8a95c4eba3ec244f1/.circleci/config.yml#L333

This includes updating the DockerHub settings to change the tagging of the automated repo build. 

https://hub.docker.com/repository/docker/rosplanning/navigation2/builds/edit

There are also some updates that will be needed given the renaming of upstream branches:

https://github.com/ros-planning/navigation2/blob/dafcffa75ba7a1a5e1b1bdd86c80454ba11d2447/tools/source.Dockerfile#L5

https://github.com/ros-planning/navigation2/blob/dafcffa75ba7a1a5e1b1bdd86c80454ba11d2447/tools/source.Dockerfile#L52

https://github.com/ros-planning/navigation2/blob/d8b8a9a0c41d2a86795a365d49d833c3ecd2b3e4/tools/run_sanitizers#L4

Depending on how organizations other than OSRF proceed, we'll need to keep tabs on our dependencies as well:

https://github.com/ros-planning/navigation2/blob/69977cd83d9040df3422d8a2e564715b6002f3fb/tools/ros2_dependencies.repos#L2-L5

Also, we should consider if such fixes should backported to previous branches as well,
lest we forfeit reproducibility for previous release builds.

SteveMacenski, 2020-07-14:
I have no conceptual issue - just not sure exactly how long it would take before I could address it fully. If anyone has spare time they could help with this transition for, that would be helpful. Like Ruffin mentions, my main concerns are updates necessary to docker / CI / dockerhub / CircleCi / scripts. I don't think there's anywhere (that I can think of). 

Rather than `main` I was thinking `root` since that's in git tree theme

mkhansenbot, 2020-07-14:
My 2 cents is it should follow whatever Github does, which sounds like it will be `main`. That way it's consistent with other Github repos.

From what I understand, Open Robotics is moving to `main` for all their repos

allenh1, 2020-07-14:
@mkhansenbot thanks for continuing the conversation here!

> My 2 cents is it should follow whatever Github does, which sounds like it will be main.

I have been going with "main" for "main copy" (as well as consistency with other repos). 

> Rather than main I was thinking root since that's in git tree theme

This is also an interesting idea, and makes quite a bit of sense (being the "root" node). My only issue with "root" in this context is that it makes me think of the original commit (the root of the commit graph). But that's probably just a "me" perspective, and by no means correct in any way. :+1: from me for either

ruffsl, 2020-07-14:
If `main` is to be the global default in the community going forward, my preference would then be that for consistency.

SteveMacenski, 2020-07-14:
Sure, just throwing out a suggestion. Main it is.

ruffsl, 2020-07-14:
It looks like there currently isn't a method for renaming the default branch without effecting open PRs in GitHub. The new `main` branch can be forked from `master`, and then set to default; but if the `master` branch is then deleted, then PRs targeting `master` will be automatically closed. We don't have too may PRs, but it would be nice not to lose all the PR discussions.

https://stackoverflow.com/questions/20007578/renaming-a-branch-while-on-pull-request

Any suggestions? Is it anticipated that there will be a feature to facilitate the migration? Should maintainers manually modify open PRs over to `main` before removing `master`?

https://stackoverflow.com/questions/24159036/how-to-modify-a-pull-request-on-github-to-change-target-branch-to-merge-into

https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/changing-the-base-branch-of-a-pull-request

SteveMacenski, 2020-07-14:
I can also change the target of the PRs with admin permissions. If there's no other branching, it should resolve OK. 

The questions to resolve:
- When
- All the things we need to change, because we need to do it all at once
- Who

ruffsl, 2020-07-15:
> When

A [`main`](https://github.com/ros-planning/navigation2/tree/main) branch with the necessary changes to CI has been added, including the new automated build rules to our Docker repo. @SteveMacenski , you may change the target of the PRs and default branch with admin permissions when you are ready, as the CI has been migrated.

https://circleci.com/workflow-run/48e754cc-7be4-4ae8-a534-5f1ffbfc8989

> All the things we need to change, because we need to do it all at once

The current `main` branch includes only changes with respect to the renaming of the local default branch. More changes upon the renaming of default branches from upstream will be required in the future as OSRF makes such changes to it's own infrastructure.

> Who

There are other references to the term `master` throughout the code base, e.g. master grids, master costmaps, master arrays. I'll defer to other maintainers more familiar with those costmap plugins to decide if and how that should be handled. Perhaps best tracked in it's own ticket. 

One note, it looks like historic codecov stats we've accrued the last 6 months may be disrupted until the default branch is updated accordingly in GitHub. Just so no one is caught unaware:

https://docs.codecov.io/v4.3.0/docs/codecov-yaml#master-copy

Ibmurai, 2020-07-15:
> Rather than `main` I was thinking `root` since that's in git tree theme

Clearly SVN was right: `trunk` is the correct name. Though that could offend elephants. Actually we should refrain from using words, as words can offend some people.

SteveMacenski, 2020-08-11:
@ruffsl I see you made a main branch with some jobs - do you have a migration plan?

ruffsl, 2020-08-11:
The CI has been ready since my previous comment. I think we're just waiting for someone with admin access to retarget all the open PRs. By now, I think the HEAD of main branch will also need to pushed to match that of the current default as some commits have been made in the meantime.

SteveMacenski, 2020-08-11:
Yeah, I was planning on rolling that forward after I do the foxy sync. So we should be good to go after that with dockerhub too (both triggering and updating the docker image names to main.debug and main.release)?

ruffsl, 2020-08-11:
Dockerhub repo settings are already primed for the switch; I merely have the auto build setting for the `main` branch disabled to limit the backlog of build jobs for docker images. When the `main` branch is updated (it already includes commits for the CI + Dockerfile + README changes), then I'll re-enable it to update the respective `main` branch images.

SteveMacenski, 2020-08-11:
OK. Let me get this foxy release / cherry pick done and I can focus on getting main up to date + PRs retargetted. 

At that point, we'd be in a good position to switch default to main, delete master, and reactivate Dockerhub for main and disable for master. I think that should do it?

ruffsl, 2020-08-11:
That should do it. Just comment here when your finished with PR and branch changes, and I'll update the Dockerhub setting.

SteveMacenski, 2020-08-11:
My todo list
- get main up to date with master
- change all PRs to use main
- delete master
- Ask Ruffin to remove master jobs and replace with main jobs (dockerhub, nightly)
- Control F for master in files to change

SteveMacenski, 2020-08-11:
@ruffsl I think your master PR just broke everything. What did you change in main to setup what you needed for this change? Main is completely messed up now and we need to restart.

Reminder **never** merge pull requests without squashing them. This has made the process about an order of magnitude more challenging and we'll likely have to restart.

ruffsl, 2020-08-11:
I'm not sure how your attempting to update main to include master, but master simply merges fine into main with no conflicts, here it is my fork for example: https://github.com/ruffsl/navigation2/commits/main

Where you trying to rebase the last few commits original on the head of main on top of master?
These are what was relevant here:
* ceac373de9b67af182afe618c8cd2f7386e1b20e
* c19199175ec81061fc4d13b7e76cdd8a52c8da45
* 1b0a7b160245daafae074c85e168a50a9d5bacbf

> In my opinion, main should be just nuked and restarted on head.

Done. I've rebase from master, but let not stall for main to fall behind again before retargeting branches.

SteveMacenski, 2020-08-12:
> Done. I've rebase from master, but let not stall for main to fall behind again before retargeting branches.

Thank you, I got really flustered when my scripts for aligning the branches succeeded but when I looked the diff was thousands of lines long as the hashes didn't match. I went through logs and saw some merge commit related errors that didn't trigger a non-zero return code from git so the script continued and pushed to origin. 

Changing the target of all the PRs now and will plan on tentatively deleting the master branch from remote but keeping a local up to date copy just in case for the next week or two while we migrate. 

I'll comment back when we're ready for you to update the CI / dockerhub settings

SteveMacenski, 2020-08-12:
@ruffsl master has been deleted from origin and all the pending PRs are now targetted to main.

Can you make the changes we talked about? I think that should button us up and complete the migration

SteveMacenski, 2020-08-12:
Through an audit, the last references to master are singularly in the costmap 2d package

ruffsl, 2020-08-12:
> Can you make the changes we talked about?

Done.

SteveMacenski, 2020-08-12:
I think its safe to close this then!

closed 2020-08-12 by SteveMacenski