ros-navigation/navigation2 #1874: Unable to set square robot footprint

Michael-Equi, 2020-07-14:
## Bug report

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - 69977cd83d9040df3422d8a2e564715b6002f3fb
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Replace each instance of `robot_radius: 0.4` with `footprint: [ [0.1, 0.1], [-0.1, 0.1], [-0.1, -0.1], [0.1, -0.1] ]` in the nav2_param.yaml file. 
#### Expected behavior
The robot should have a square footprint.

#### Actual behavior
Error parsing the yaml file by the following nodes from all the nodes that utilize it.

#### Additional information

The rest of the error logs: 
```

[INFO] [launch]: All log files can be found below /home/michael/.ros/log/2020-07-14-13-02-46-338923-michael-desktop-20-04-33737
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [33739]
[INFO] [amcl-2]: process started with pid [33741]
[INFO] [lifecycle_manager-3]: process started with pid [33743]
[INFO] [controller_server-4]: process started with pid [33745]
[INFO] [planner_server-5]: process started with pid [33747]
[INFO] [recoveries_server-6]: process started with pid [33755]
[INFO] [bt_navigator-7]: process started with pid [33758]
[INFO] [waypoint_follower-8]: process started with pid [33763]
[INFO] [lifecycle_manager-9]: process started with pid [33767]
[map_server-1] [ERROR] [1594756966.634841491] [rcl]: Failed to parse global arguments
[map_server-1] 
[map_server-1] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[map_server-1] This error state is being overwritten:
[map_server-1] 
[map_server-1]   'Couldn't parse params file: '--params-file /tmp/tmpz6y59cqa'. Error: Sequences cannot be key at line 179, at /tmp/binarydeb/ros-foxy-rcl-yaml-param-parser-1.1.5/src/parser.c:1615, at /tmp/binarydeb/ros-foxy-rcl-1.1.5/src/rcl/arguments.c:390'
[map_server-1] 
[map_server-1] with this new error message:
[map_server-1] 
[map_server-1]   'context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.5/src/rcl/context.c:51'
[map_server-1] 
[map_server-1] rcutils_reset_error() should be called after error handling to avoid this.
[map_server-1] <<<
[map_server-1] [ERROR] [1594756966.634869715] [rclcpp]: failed to finalize context: context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.5/src/rcl/context.c:51
[map_server-1] [ERROR] [1594756966.634936977] [map_server]: failed to initialize rcl: error not set
[map_server-1] [ERROR] [1594756966.634941250] [map_server]: Exiting
[map_server-1] corrupted size vs. prev_size in fastbins
[amcl-2] [ERROR] [1594756966.632528933] [rcl]: Failed to parse global arguments
[amcl-2] 
[amcl-2] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[amcl-2] This error state is being overwritten:
[amcl-2] 
[amcl-2]   'Couldn't parse params file: '--params-file /tmp/tmpjc0k173n'. Error: Sequences cannot be key at line 179, at /tmp/binarydeb/ros-foxy-rcl-yaml-param-parser-1.1.5/src/parser.c:1615, at /tmp/binarydeb/ros-foxy-rcl-1.1.5/src/rcl/arguments.c:390'
[amcl-2] 
[amcl-2] with this new error message:
[amcl-2] 
[amcl-2]   'context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.5/src/rcl/context.c:51'
[amcl-2] 
[amcl-2] rcutils_reset_error() should be called after error handling to avoid this.
[amcl-2] <<<
[amcl-2] [ERROR] [1594756966.632552305] [rclcpp]: failed to finalize context: context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.5/src/rcl/context.c:51
[amcl-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[amcl-2]   what():  failed to initialize rcl: error not set
[lifecycle_manager-3] [INFO] [1594756966.632305136] [lifecycle_manager_localization]: Creating
[lifecycle_manager-3] [INFO] [1594756966.646978388] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[... 64 lines cut ...]
[waypoint_follower-8] [ERROR] [1594756966.657702841] [rclcpp]: failed to finalize context: context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.5/src/rcl/context.c:51
[waypoint_follower-8] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[waypoint_follower-8]   what():  failed to initialize rcl: error not set
[lifecycle_manager-9] [INFO] [1594756966.718751198] [lifecycle_manager_navigation]: Creating
[ERROR] [map_server-1]: process has died [pid 33739, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_map_server/lib/nav2_map_server/map_server --ros-args -r __node:=map_server --params-file /tmp/tmpz6y59cqa'].
[lifecycle_manager-9] [INFO] [1594756966.746830058] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[lifecycle_manager-9] [INFO] [1594756966.752703072] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-9] [INFO] [1594756966.752735306] [lifecycle_manager_navigation]: Configuring controller_server
[ERROR] [controller_server-4]: process has died [pid 33745, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmpusm99kx1'].
[ERROR] [planner_server-5]: process has died [pid 33747, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmp889vguws'].
[ERROR] [bt_navigator-7]: process has died [pid 33758, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator --params-file /tmp/tmpx5ldbw37'].
[ERROR] [recoveries_server-6]: process has died [pid 33755, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_recoveries/lib/nav2_recoveries/recoveries_server --ros-args -r __node:=recoveries_server --params-file /tmp/tmp7wefvudc'].
[ERROR] [amcl-2]: process has died [pid 33741, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_amcl/lib/nav2_amcl/amcl --ros-args -r __node:=amcl --params-file /tmp/tmpjc0k173n'].
[ERROR] [waypoint_follower-8]: process has died [pid 33763, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_waypoint_follower/lib/nav2_waypoint_follower/waypoint_follower --ros-args -r __node:=waypoint_follower --params-file /tmp/tmp5clxrdja'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[lifecycle_manager-3] [INFO] [1594757074.835698241] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-9] [INFO] [1594757074.835698232] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-9] terminate called after throwing an instance of 'std::runtime_error'
[lifecycle_manager-9]   what():  controller_server/change_state service client: interrupted while waiting for service
[lifecycle_manager-3] terminate called after throwing an instance of 'std::runtime_error'
[lifecycle_manager-3]   what():  map_server/change_state service client: interrupted while waiting for service
[ERROR] [lifecycle_manager-9]: process has died [pid 33767, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_lifecycle_manager/lib/nav2_lifecycle_manager/lifecycle_manager --ros-args -r __node:=lifecycle_manager_navigation --params-file /tmp/launch_params_v8opicdd --params-file /tmp/launch_params_ddow8w6k --params-file /tmp/launch_params_wp0z4i6w'].
[ERROR] [lifecycle_manager-3]: process has died [pid 33743, exit code -6, cmd '/home/michael/Github/br_core/ws/install/nav2_lifecycle_manager/lib/nav2_lifecycle_manager/lifecycle_manager --ros-args -r __node:=lifecycle_manager_localization --params-file /tmp/launch_params_i5anleri --params-file /tmp/launch_params_1d7rpd_k --params-file /tmp/launch_params_n5spz4ez'].

```

SteveMacenski, 2020-07-14:
That looks fun! We should definitely take a look at that.

Jconn, 2020-07-14:
The footprint has been working fine for me, but I need to send in my nested array as a string, e.g. `"[ [0.1, 0.1], [-0.1, 0.1], [-0.1, -0.1], [0.1, -0.1] ]"`

There is a `makeFootprintFromString` in here somewhere that handles the parsing. It's unclear to me why the array type in yaml couldn't manage this

Michael-Equi, 2020-07-14:
> The footprint has been working fine for me, but I need to send in my nested array as a string, e.g. `"[ [0.1, 0.1], [-0.1, 0.1], [-0.1, -0.1], [0.1, -0.1] ]"`

That works! Ill go ahead and add it to the documentation.

SteveMacenski, 2020-07-14:
@Jconn I think its because its an array of arrays, not an array of values (ex `[0.1, 0.1, 0.1]`). There's no arrays of arrays parser. Thanks for the tip!

closed 2020-07-16 by Michael-Equi