ros-navigation/navigation2 #1876: Tf error: Timed out waiting for transform from base_link to map to become available

Tahir-Rasheed-437, 2020-07-16:
I am using vrep simulator with ROS2 dashing. First i used SLAM toolbox to map my environment. worked like a charm. Then, i am launching the navigation launch file (https://github.com/ros-planning/navigation2/blob/dashing-devel/nav2_bringup/launch/nav2_bringup_launch.py) by replacing the 'turtlebot3_world.yaml' with my own map file generated through SLAM toolbox and use_sim_time to be true.  I get the following error.  

Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.00747 timeout was 1

The problem is, there is no tf generated from map->odom. For rviz2 i use the command:

ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz

My map is loaded in rviz but in the wrong position which make sense as there is no tf between map and odom.   I have run through some older issues and found out navigation stack generate the tf once the initial pose is given using "2D Pose Estimate" button on rviz. Once given, i got the following message in the terminal:

[rviz2-1] [INFO] [rviz2]: Setting goal: Frame:map, Position(-0.0126218, 0.0416158, 0), Orientation(0, 0, -0.714806, 0.699322) = Angle: -1.59269
[amcl-4] [INFO] [amcl]: initialPoseReceived
[amcl-4] [INFO] [amcl]: Setting pose (1471886.000000): -0.013 0.042 -1.593
[world_model-5] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1 timeout was 1.
[amcl-4] [INFO] [amcl]: Received a 434 X 454 map @ 0.050 m/pix
[amcl-4] [INFO] [amcl]: Setting pose (1471887.250000): -0.013 0.042 -1.593
[world_model-5] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.125 timeout was 1.
[world_model-5] [INFO] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: . canTransform returned after 1.125 timeout was 1.

No tf is generated and the error keeps on coming about the transform. i know such errors have been solved by giving the 2D pose estimate but its not solving my issue. Am i missing something ? something else is needed to continuously generate the tf? My scans are fine as i have successfully run the SLAM-toolbox. Exactly same behavior https://github.com/ros-planning/navigation2/blob/dashing-devel/nav2_bringup/launch/nav2_simulation_launch.py. The error is vanished as soon as i launch the SLAMN-toolbox again but then the map is refreshed and start from zero.

SteveMacenski, 2020-07-16:
> The problem is, there is no tf generated from map->odom

Are you launching AMCL and giving it the initial pose? https://navigation.ros.org/getting_started/index.html follow this tutorial by the letter (except the nav2_simulation_launch.py file instead of the turtlebot_simulation_launch.py since we renamed that in Eloquent) and let me know if it still fails. 

Overall, I don't provide support for Dashing. its too old and too many changes have been made since then for me to practically consider issues with Dashing. If you can replicate this on Eloquent or Foxy, I'd be open to considering more assistance.

Tahir-Rasheed-437, 2020-07-17:
yes i did follow this tutorial. sorry i didn't mentioned it. 
OK i am currently shifting on foxy. I'll update here asap

Tahir-Rasheed-437, 2020-07-20:
Hey steve, so i have checked on foxy now. i had some issues with SLAM-toolbox as well but i will try to generate an issue in the SLAM git. I have somehow managed to generate the map from the toolbox. So for the navigation, as soon as i launch the files either tb3_simulation_launch or bringup_launch in (https://github.com/ros-planning/navigation2/tree/foxy-devel/nav2_bringup/bringup/launch), I got the following:

[image]

I am unable to click on start. On the terminal, the errors are: Timed out waiting for transform from base_link to map to become available, tf error. I guess its waiting for me to give the 2D pose estimate. But its rviz is not allowing me to start. Am i missing something?

Secondly, you mentioned launching AMCL. this tutorial https://navigation.ros.org/getting_started/index.html says that  tb3_simulation_launch launches AMCL as well. So i don't have to separately launch AMCL right?

SteveMacenski, 2020-07-20:
Please ask on ROS answers.

closed 2020-07-20 by SteveMacenski

HomeLH, 2020-07-23:
```
[planner_server-9] [INFO] [1595481966.470863588] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-4] [INFO] [1595481966.478299289] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 83.079 for reason 'Unknown'
[rviz2-4] [INFO] [1595481966.670278195] [rviz2]: Message Filter dropping message: frame 'odom' at time 83.743 for reason 'Unknown'
[rviz2-4] [INFO] [1595481966.670593265] [rviz2]: Message Filter dropping message: frame 'odom' at time 0.000 for reason 'Unknown'
[rviz2-4] [INFO] [1595481966.702458231] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 83.278 for reason 'Unknown'
[amcl-6] [WARN] [1595481966.733056020] [amcl]: ACML cannot publish a pose or update the transform. Please set the initial pose...
[rviz2-4] [INFO] [1595481966.744029831] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1595481966.744422472] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[rviz2-4] [INFO] [1595481966.863085439] [rviz2]: Message Filter dropping message: frame 'odom' at time 83.881 for reason 'Unknown'
```
I met the same problem. 
System: Ubuntu 20.04
ros2 version: foxy.
The navigation 2 part in RVIZ shows "Navigation:unkown and Localization:unkown". Buttons are all gray.

Tahir-Rasheed-437, 2020-07-23:
@HomeLH, you have to launch first the TB3 in simulation, and then you have to launch the navigation stack (TB3 file does not contain any navigation stack nodes). It will enable the options and you start and can give the initial pose. 

In foxy, things are written a bit differently compared to dashing. hope it will help.

Tahir-Rasheed-437, 2020-07-24:
I finally succeeded in making navigation work for foxy for turtlebot in gazebo. Here are the steps i followed:

export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$path to gazebo models
ros2 launch nav2_bringup tb3_simulation_launch.py

You don't have to click on start, all the 3 buttons will be grey initially. Just give 2D initial pose in the environment, the turtlebot model along with the costmaps will be loaded and then you can give the 2D navigation goal. 
[image]

After this, the buttons on left bottom corner are available to be clicked.