ros-navigation/navigation2 #1877: Rviz must be started after nav2

Michael-Equi, 2020-07-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy Binaries
- Version or commit hash:
  - 69977cd83d9040df3422d8a2e564715b6002f3fb
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
1. Launch TB3 in simulation without the nav2 stack. 
2. Launch Rviz2 after having sourced the repository with nav2 build in order to have access to the lifecycle manager interface and nav2 goal pose. 
3. launch the nav stack.

#### Expected behavior

Initial position can be set in the Rviz2 instance that was launched in the second step above.

#### Actual behavior

Setting initial pose does not communicate with the nav2 stack and therefore has not effect. Solution is to close and reopen Rviz2 (without restarting the nav2 stack or simulation) with the same configuration and then set the initial pose.

SteveMacenski, 2020-07-16:
I think that's an issue for ROS2 or Rviz, not for navigation2 in particular. Do you see that same behavior with non-navigation2 things? If so, I think this should be closed and redirected towards rviz or ROS2 core things

Michael-Equi, 2020-07-16:
It seems mostly to be an issue setting initial pose and there is not another program I typically use if for other than nav2. I can go ahead and try it out with another DDS vendor later today and report back.

SteveMacenski, 2020-07-16:
Well that's a topic publisher discovery issue, I don't think that's anything to do with Nav2, but more of ROS2 / DDS. 

Keep in mind this happened also in ROS1, but to a much less frequent degree.

Michael-Equi, 2020-07-17:
Cyclone DDS does solve this issue although I am still facing a painfully low real-time factor with the simulation (0.5 for TB3 and 0.1 on my own robot).

SteveMacenski, 2020-07-17:
Ok. I think that closes this issue in Nav2- its not something under this project’s scope. I’d suggest filing a ticket with fast rtps’s rmw and githubs linking this issue as an example manifestation.

closed 2020-07-17 by SteveMacenski