ros-navigation/navigation2 #1879: Multiple definition error if tf_help.hpp is included in more than one source file.

ijsmallman, 2020-07-17:
I think transformPose definitions should be defined in a separate source file and only declared in tf_help.hpp

SteveMacenski, 2020-07-17:
@ijsmallman please properly fill in the issue template & provide specific links about what you're referencing. We're not going to dig through the entire source tree to guess about what you're talking about.

ijsmallman, 2020-07-17:
What issue template?

SteveMacenski, 2020-07-17:
The one that you deleted when you filed this ticket. https://github.com/ros-planning/navigation2/issues/new

ijsmallman, 2020-07-17:
It's a pretty straightforward fix - put function definitions in a separate cpp file. I can file a pull request when i get back from my break.

SteveMacenski, 2020-07-17:
OK - sounds good

SteveMacenski, 2020-07-23:
Any updates @ijsmallman

ijsmallman, 2020-07-24:
I'm on holiday this week, I'll update on Monday

closed 2020-07-28 by SteveMacenski