ros-navigation/navigation2 #1883: Navigation2 doesn't function well on Celeron

jeremylu0601, 2020-07-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- ROS2 Version:
  - Foxy binaries 
- Version or commit hash:
  - foxy-devel
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Use Celeron G3900TE to run Navigation2

#### Expected behavior

#### Actual behavior
    [ERROR] [tf_help]: Transform data too old when converting from odom to map
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2020-07-22:
Please ask on ros answers / soec out your machine for the resources you need

closed 2020-07-22 by SteveMacenski

jeremylu0601, 2020-07-23:
Got it.
Thanks for your response.