ros-navigation/navigation2 #1891: Navigation failing

Tahir-Rasheed-437, 2020-07-28:
I am currently using master branch in simulation with vrep. I am launching the navigation using:

`ros2 launch nav2_bringup bringup_launch.py use_sim_time:=True autostart:=True \map:=src/navigation2/nav2_bringup/bringup/maps/map1.yaml `

map1 is loaded. Then, i am launching nav2_default_view file of rviz, giving the initial pose, robot footprint and the cost maps are loaded and then i can be able to give the navigaiton2 Goal. It works fine until after some time it gives me this error:  

[ERROR] [planner_server-5]: process has died [pid 6869, exit code -11, cmd '/home/irt-tests/catkin_navigation2_wsb/install/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmp0qwuajkw -r /tf:=tf -r /tf_static:=tf_static'].

Sometimes the error is generated at the very start of the navigation, and sometimes later during the navigation. The same error is generated if i used tb3_simulation_launch.py for my own simulation. What could be the cause?

SteveMacenski, 2020-07-28:
Properly fill in the Issue template, please.

Tahir-Rasheed-437, 2020-07-29:
@SteveMacenski , apologies, i have filled the necessary information.

SteveMacenski, 2020-07-29:
@Tahir-Rasheed-437 please try again. You're still missing sections.

Tahir-Rasheed-437, 2020-07-30:
@SteveMacenski apologies for the delay, i have tried my best to upload as much information as possible. Still if something is missing please do let me know. i have tried alot to remove the crash but its keep on coming sooner or later. Kindly do try and please let me know if you can find anything. Thank you.

SteveMacenski, 2020-07-30:
> I have CoppeliaSim simulator generating all the necessary topics of /odom, /clock and /scan (bag file is attached).

Try gazebo and see if you can reproduce, we're not going to debug specific third-party simulator issues if they don't properly deal with ROS correctly. 

Try compiling with GDB and get a traceback, that should help us narrow down the reason for a crash

Tahir-Rasheed-437, 2020-07-30:
I did tried gazebo and no problems there. Maybe something is wrong on CoppeliaSim side not sure about it.
I tried to mix gazebo and CoppeliaSim and somehow it also did work out. Things start getting crash as soon as the transforms are not from gazebo but my own nodes (the error somehow also shows that its not happy with my tf's)

Just a quick query, are the tf's in foxy-navigation are used/published the same way as dashing-navigation? I am using the same package that generates the tf which i wrote for dashing and i hope no changes there. (sorry for this query, if its not directly related to navigation but just wondering about it)

Sure i'll try what you said. What is GDB stand for? sorry i am not very familiar with the notations. 
Thank you again for your tips.

SteveMacenski, 2020-07-30:
No changes from dashing to foxy on navigation2's use of TF2, but you'd have to look at TF2 itself to know if it changed. My guess is your sim isnt publishing the right info or isn't providing all the necessary information. But again, you'd need to get a traceback to know. 

https://www.gnu.org/software/gdb/

SteveMacenski, 2020-07-30:
We're welcome to keep debugging, but since we solved this as a non-nav2 problem, I'm closing the ticket. Feel free to keep commenting or start a thread in slack

closed 2020-07-30 by SteveMacenski