ros-navigation/navigation2 #1892: nav2_system_tests build failure on build.ros2.org

jacobperron, 2020-07-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - 0.4.1
- DDS implementation:
  - N/A

#### Steps to reproduce issue

See failures on the buildfarm.

Example failure: http://build.ros2.org/job/Fbin_uF64__nav2_system_tests__ubuntu_focal_amd64__binary/26

#### Additional information

Looks like there's an issue finding `nav2_bringup`:

```
23:04:53 CMake Error at CMakeLists.txt:10 (find_package):
23:04:53   By not providing "Findnav2_bringup.cmake" in CMAKE_MODULE_PATH this project
23:04:53   has asked CMake to find a package configuration file provided by
23:04:53   "nav2_bringup", but CMake did not find one.
23:04:53 
23:04:53   Could not find a package configuration file provided by "nav2_bringup" with
23:04:53   any of the following names:
23:04:53 
23:04:53     nav2_bringupConfig.cmake
23:04:53     nav2_bringup-config.cmake
23:04:53 
23:04:53   Add the installation prefix of "nav2_bringup" to CMAKE_PREFIX_PATH or set
23:04:53   "nav2_bringup_DIR" to a directory containing one of the above files.  If
23:04:53   "nav2_bringup" provides a separate development package or SDK, be sure it
23:04:53   has been installed.
```

I'm guessing that adding `nav2_bringup` as a build_depend to the package.xml should fix the issue.

jacobperron, 2020-07-28:
> I'm guessing that adding `nav2_bringup` as a build_depend to the package.xml should fix the issue.

Or, removing the `find_package(nav2_bringup)` line in the CMakeLists.txt if it is not required for linking.

SteveMacenski, 2020-07-28:
Yeah, I've seen that. I didn't think it was worth updating until the first foxy sync (https://github.com/ros-planning/navigation2/issues/1866) since its not functional run-time code. Thanks for bringing it to my attention though. I copied Moveit and added all the CI badges to the readme to help me track these issues. 

I plan to do a sync when #1869 is in (which could be sped up by some reviews on my PR in bond_core :smile: ).

SteveMacenski, 2020-08-11:
In progress, will be done by EOD

closed 2020-08-11 by SteveMacenski