ros-navigation/navigation2 #1895: Observations sources in obstacle layer documented as list of strings but is actually string with white space

Michael-Equi, 2020-07-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - c65e3aa388f31d6469970d91ce62e6a17c313d07
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue
Try sending list of strings as observation source for the obstacle layer. For example by setting the following as shown `observation_sources: ["scan", "bumper"]`.

#### Expected behavior
Sets the obstacle layers observations sources as scan and bumper.

#### Actual behavior

```
[controller_server-4] [ERROR] [1595980200.105718186] []: Caught exception in callback for transition 10
[controller_server-4] [ERROR] [1595980200.105739892] []: Original error: parameter 'obstacle_layer.observation_sources' has invalid type: expected [string] got [string_array]
[controller_server-4] [WARN] [1595980200.105760281] []: Error occurred while doing error handling.
```

The obstacle layers wants the observation sources set like the following: `observation_sources: scan bumper`

#### Additional information

This can either be fixed by updating the documentation which says that the variable is a vector<string> or by changing the obstacle layer to take in a list of strings. I feel like the latter option is the best and makes the most logical sense.

SteveMacenski, 2020-07-29:
https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/plugins/obstacle_layer.cpp#L92

You're right, someone ported it to support only 1 feed, that's not awesome. Its not a documentation problem, its a feature problem, its supposed to support multiple feeds. The documentation is probably correct.

Michael-Equi, 2020-07-29:
It looks like it does handle multiple input sources (just tested it) but it accepts them in a way other than a list of strings.

SteveMacenski, 2020-07-29:
Can you be more specific on that? Not sure I exactly understand. What I think you might mean is that is reads each from a long string rather than a vector of them?

Michael-Equi, 2020-07-29:
That is what I am getting at, yes.

For multiple sensor inputs you would provide `sensor1 sensor2` rather than `["sensor1", "sensor2"]`

SteveMacenski, 2020-07-29:
Got it - I agree then, we should update docs

SteveMacenski, 2020-08-05:
https://github.com/ros-planning/navigation.ros.org/pull/64

closed 2020-08-05 by SteveMacenski