ros-navigation/navigation2 #1902: Nav stack publishes Twist message to controllers except for TwistStamped?

bmagyar, 2020-08-03:
Hey guys,

As it was brought to my attention, the nav stack somehow ended up publishing Twist messages to controllers even though internally it's still using TwistStamped?

Gazebo toy environments won't showcase this but any nitty & gritty real world system will want stamps on everything, *especially* on messages going to controllers.

I wonder if you have any plans for addressing this, the extra header won't hurt anyone but the lack of it may force companies forking the entire stack just to fix the publishers...

Keep up the good work,
Bence

PS 
Ref: https://github.com/ros-controls/ros2_controllers/issues/80

SteveMacenski, 2020-08-03:
@bmagyar I think I need alot more context here. Also see #1594 where I started the discussion about moving Twist to TwistStamped.

SteveMacenski, 2020-08-05:
@bmagyar ping

bmagyar, 2020-08-10:
Thanks for the pointer to #1594 , I was unaware of this one. With this in mind all I can say is that I'm strongly in favour of *Stamped messages for controllers and that is how we are setting things up in `ros2_control` already. We may provide a non-stamped version for Foxy when the release happens (later this year) if there's a strong mandate for it.

SteveMacenski, 2020-08-10:
> We may provide a non-stamped version for Foxy when the release happens (later this year) if there's a strong mandate for it.

If you'd like to have ros2_control interact with the navigation2 stack, I think that's a strong mandate. Plus you will need to update all of the gazebo plugins that might be an analog to ros2_control for hardware so that the interfaces match, right now they're all not going to be Stamped,

I have 0% problem with changing to `Stamped`, but you've missed the boat for Foxy. You're not going to be able to change that API for Nav2, Gazebo plugins, rviz, turtlebot, etc after they've been released. I would be happy to work on this to target for Galactic though.

Like #1594 mentions, the major blocker is the ecosystem, not navigation2. We need to get a list of all of the packages that we need to update in one place and start knocking them out.

bmagyar, 2020-08-10:
Indeed it is ;) Thanks for the feedback!

closed 2020-08-10 by bmagyar

SteveMacenski, 2020-08-10:
@bmagyar if you want to get started on Galactic, we can start that process now, we just need to align efforts.

bmagyar, 2020-08-13:
I happen to be maintainer over at ros-teleop too, happy to help