ros-navigation/navigation2 #1905: Add parameter for map topic in AMCL

Michael-Equi, 2020-08-04:
## Feature request

#### Feature description
The AMCL node should have a parameter `map_topic` to configure the map topic similar to the parameter for the static costmap layer which would allow for easy configuration of the node's map input without using remapping. 

#### Implementation considerations
Documentation would need to be added in the following locations:
https://navigation.ros.org/configuration/packages/configuring-amcl.html#parameters
https://github.com/ros-planning/navigation2/blob/master/doc/parameters/param_list.md

SteveMacenski, 2020-08-04:
Didn't you submit a PR for this? I don't seem to be able to find it. 

Either way, I'd merge a PR with that

Michael-Equi, 2020-08-04:
I made a PR for the sensor_topic not the map_topic

SteveMacenski, 2020-08-04:
Got it. I'd merge that PR, also remember you have `remapping` you can use to remap the "map" topic to something else.

Michael-Equi, 2020-08-04:
Just makes things a bit cleaner having it all in one configuration file.

closed 2020-08-05 by SteveMacenski