ros-navigation/navigation2 #1906: Make the map_topic parameter used in static layers specific to each static layer

Michael-Equi, 2020-08-04:
## Feature request

#### Feature description
Rather than having `map_topic` be a global parameter for the entire costmap it should be specific to each layer so that multiple static layers can be used to include things like "invisible walls" where an invisible wall layer would be server by a separate map_server on a unique topic other than `/map`.

#### Implementation considerations
This one may be worth some discussion and I am not sure whether there is a better way to go about handling invisible walls but this one seems the most logical and flexible to me. It would be best if we make the `map_topic` parameter "private" to the layer that we remove the global one although I am not sure if this will break any other costmap layers (it doesn't look like it would affect any of the other official ones).

SteveMacenski, 2020-08-04:
I'd consider a PR for that.

Also keep in mind the work happening in #1882, you'll soon be able to have map-masks or "filters" to apply to your costmap with provided layers like keepout zones (which I imagine is why you might want multiple static layers) amongst others.

Michael-Equi, 2020-08-04:
Ok, I didn't see that issue but yes, it basically achieves what I am trying to do. I will probably still push a PR for this since its a rather simple change and I am trying to get this ability built into my robot over the next couple days. How close is #1882 to being merged in?

SteveMacenski, 2020-08-04:
Pretty close, just a few things I'm waiting on Alexey to finish up to merge (mostly around tests and documentation)

closed 2020-08-05 by SteveMacenski