ros-navigation/navigation2 #1911: Costmap locking while making a plan

daisukes, 2020-08-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy release src build

#### Steps to reproduce issue

There is an issue in making plans while updating costmap.
I have a large map and sometimes got undesired results of plans with incomplete costmaps.

#### Expected behavior

Guarantees making plans with complete costmaps.

#### Additional information

I could work around this problem by adding a lock before making a plan, 
https://github.com/CMU-cabot/navigation2/commit/ba6b87e60899e7f0ce960d74bef11ee21a2a01d0

I guess, it may be better to lock only while copying costmap data into the planner instead of the entire planning process, but this means each planner plugin needs to take care of it. 
https://github.com/CMU-cabot/navigation2/commit/04ceabeabe3e328641bd9ee6cbca82b8f0ece3cc

Could you please let me know your thoughts.

SteveMacenski, 2020-08-04:
> Guarantees making plans with complete costmaps.

What does that mean, exactly? What part of the costmap isn't "complete" when you try to plan?

daisukes, 2020-08-04:
The layered costmap which has four layers in my environment. Some of the layers, especially when CPU load is high, have not completed when the planner makes a plan.

SteveMacenski, 2020-08-04:
But that doesn't answer what an "incomplete" map is or what the bad planning behavior you're seeing is

daisukes, 2020-08-04:
I'm using a static layer, an obstacle layer, and two custom layers.
My custom layer provides cost where the robot should not go through even if there is no obstacle, but sometimes it makes a plan like ignoring that layer, but still, a plan based on the static layer.

SteveMacenski, 2020-08-05:
Got it. https://github.com/ros-planning/navigation/blob/827c796e171b4b4b476b4aade68910a1372d6c7e/move_base/src/move_base.cpp#L473

I'm seeing where this was in move-base so it looks like an oversight on the planner and controller servers in ROS2 port. I'd merge a PR to add those locks before getting a plan or a control effort in the planner / controller servers since both could have that issue. 

Minimizing the lock time though would be good since it could be getting 60 fps sensor feeds that might be locked up

closed 2020-08-05 by SteveMacenski