ros-navigation/navigation2 #1917: Planner and Controller Server unable to survive reset + startup

Michael-Equi, 2020-08-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - ROS2 binaries 
- Version or commit hash:
  - 4a009dea360a079f44b02445d95052e1ded9440e
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue
Start the navigation stack and have RViz 2 running with the Navigation2 panel. Startup the navigation stack for the first time (unless done by autostart) and begin using the navstack like normal. Hit the reset button in the nav2 stack pannel and attempt to re-startup the system.

#### Expected behavior

All the nav2 nodes should transition to an active state error free.

#### Actual behavior

In every attempt at restarting the nav2 system the controller or planner server emits an error similar to either of the following:

```
[planner_server-4] [INFO] [1596675843.557904809] [planner_server]: Configuring
[planner_server-4] [INFO] [1596675843.557931726] [global_costmap.global_costmap]: Configuring
[planner_server-4] [ERROR] [1596675843.558043453] []: Caught exception in callback for transition 10
[planner_server-4] [ERROR] [1596675843.558051673] []: Original error: parameter 'static_layer.plugin' has already been declared
[planner_server-4] [WARN] [1596675843.558068927] []: Error occurred while doing error handling.
```

```

[controller_server-3] [ERROR] [1596676033.151989476] []: Caught exception in callback for transition 10
[controller_server-3] [ERROR] [1596676033.151999489] []: Original error: parameter 'stvl_layer.observation_sources' has already been declared
[controller_server-3] [WARN] [1596676033.152020113] []: Error occurred while doing error handling.
```

SteveMacenski, 2020-08-06:
That's odd because I just very recently did a bunch of tests of this to deal with bond. Can you pull in master and see if any of those changes help?

> Original error: parameter 'stvl_layer.observation_sources' has already been declared

That's an STVL bug it seems like, please file a ticket over there and we can chat on it

> Original error: parameter 'static_layer.plugin' has already been declared

That shouldn't be happening and I've tested that at length. Can you pull master and see if still an issue? If so, we can work on that too

SteveMacenski, 2020-08-06:
STVL makes sense https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/foxy-devel/src/spatio_temporal_voxel_layer.cpp#L66 this function is called on on_configure() and clearly its made no attempt to deal with resets. It should be an easy fix by changing the declare to declare_if_not_declared

Michael-Equi, 2020-08-06:
I pulled the latest and it still is giving me the same issues.

SteveMacenski, 2020-08-06:
Confirmed, I see the static layer problem as well now. It looks like L308-312 in costmap_2d_ros.cpp needs to declare if not declared.

SteveMacenski, 2020-08-06:
#1919 resolves

closed 2020-08-06 by SteveMacenski