ros-navigation/navigation2 #1925: [Open question] Should we keep the nav2 framework in this repo and move plugins into another?

SteveMacenski, 2020-08-07:
There's an argument to have the main framework (servers, BT navigator, utils, etc) in the navigation2 repo since that is the "nav2 framework". Then have another repo where we have all the algorithm implementation options provided by us. 

I'm curious if anyone has strong opinions about this and what they might be. 

CC @davetcoleman since I know MoveIt has a similar discussion a couple of years ago.

mikeferguson, 2020-08-10:
I presume the main reason to split things is to reduce the dependency tree, right? In that case, you'd actually want to have each algorithm in it's own repo (for instance, someone not using an OMPL-based planner would then not depend on OMPL). The downside to this is keeping everything aligned - which was why MoveIt merged most of their repositories together.

My suggestion would be to keep all "first-class" algorithms in this repo - but spin up new repos for more experimental or less well-supported algorithms (or algorithms that depend on less well-supported packages - for instance, we removed robot_pose_ekf from the ROS1 navigation stack a while back because the underlying dependency BFL was unmaintained and often the blocker for getting navigation into a new ROS distribution).

SteveMacenski, 2020-08-10:
The main reason is to have the navigation2 "framework" and the navigation "implementations". This may help with people being less restricted in their thinking to existing plugins or think that the "default" plugins are somehow optimal or the only type of plugins that work (e.g. only works in 2D because we provide 2D-only plugins).

mikeferguson, 2020-08-10:
If the other "implementations" are all in one repo then they are still the "default" implementations.

Perhaps it'd be better to document the list of plugins on navigation.ros.org - and put them in tiers based on how complete and/or well-supported they are (similar to the push to put ROS2 packages into Tiers based on support levels).

closed 2020-08-10 by SteveMacenski