ros-navigation/navigation2 #1929: Navigation 2 run on ramps

Andrea2202, 2020-08-09:
## Bug report

**Required Info:**

- Operating System:
 Ubuntu 18.04
- ROS2 Version:
 Foxy source code
- Version or commit hash:
  - 
- DDS implementation:
  Fast RTPS
#### Steps to reproduce issue

```
## Feature request

Hi all, 
I'm using ros2 and navigation2 for an indoor differential drive robot.
Currently the main limit of my system is that I'm not able to run ramps.
I need to move on ramps up to 6%.

Do you know if there are modules or plugins for ros2 or navigation2 that allow to navigate on ramps?
I's under development this feature on navigation2?
Or if not have you in mind a way to integrate this feature in the cirrent nav2 implementation? 

Thanks

#### Feature description
<!-- Description in a few sentences what the feature consists of and what problem it will solve -->

#### Implementation considerations
<!-- Relevant information on how the feature could be implemented and pros and cons of the different solutions -->

SteveMacenski, 2020-08-10:
Not currently a supported feature

Andrea2202, 2020-08-11:
Hi @SteveMacenski, 
Thanks for your answer.
Have you in mind some way to easly integrate  this feature in the current navigation2 stack?
Or can you provide us a path to follow in order to implement ourself this feature?

Thanks in advance

SteveMacenski, 2020-08-11:
I don't at this time. This is duplicate of #1224 asking for non-flat navigation, so I'll close this ticket but we can move discussion back over there.

closed 2020-08-11 by SteveMacenski