ros-navigation/navigation2 #1931: InflationLayer update collide when footprint is updated

daisukes, 2020-08-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu20.04
- ROS2 Version:
  - Foxy src build
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-DDS

#### Steps to reproduce issue

- launch nav2 and make it active
- change footprint may collide update

#### Expected behavior

- keep working

#### Actual behavior

- An exception is thrown when the inflation layer try to emplace_back

#### Additional information

I suspected multithread issue and tried to fix
https://github.com/CMU-cabot/navigation2/commit/e5241327ed046ff2fb3347a032c6fb44ec10776e

SteveMacenski, 2020-08-11:
> change footprint may collide update

Please give us **far** more information, this is not a properly filled out template

daisukes, 2020-08-11:
@SteveMacenski 
I updated the description with some backtraces.
If the robot changes its footprint, it crashes while updating the inflation layer.

SteveMacenski, 2020-08-11:
That does sound like a resource locking problem. What are the 2 places trying to access at once? The compute cache and what else? That will tell us where the guards need to go

SteveMacenski, 2020-08-11:
Seems like we should lock the on footprint change and the updateCosts/updateBounds (which ever uses the cached info)

daisukes, 2020-08-11:
Thank you for the comments. I thought so too.
Once I confirmed that my fix is working well with a 100hz footprint update, I will make a PR.

closed 2020-08-12 by SteveMacenski