ros-navigation/navigation2 #1933: Replace TB3 + Sandbox world in CI for more realistic robot use-cases & world

SteveMacenski, 2020-08-11:
For instance the AWS warehouse world with the Rover robotics robot doing some pick and serve operation with the nav stack:

dock -> pose -> get box -> pose -> let box go -> back to dock.

Later to be expanded to include other hard test items like doors closed needs to open, things falling off shelf, narrow aisles, other robots or people in way, etc.

shreyasgokhale, 2020-08-14:
Hi @SteveMacenski ,

I am currently working on something similar: Amazon warehouse scenario using Navigation 2 and Loading and Unloading of pallets with the help of custom behaviour trees. I have used bt_waypoint_follower (#1928) and [ROS2 planning system](https://github.com/IntelligentRoboticsLabs/ros2_planning_system) as my references. However, for the robot, I am using a modified turtlebot3_waffle with a lift joint. But I think using a standard industrial warehouse robot would've been much better.

I have managed to get my custom Packages + Gazebo model with moving lift + BT working. But I am facing some issues with `controller_server` crashing. I have filed a bug report #1944 for this. 

[This is my repo](https://github.com/shreyasgokhale/Amazon-Robot) without local_costmap (thus often fails to reach the goal). I am still in process of improving BT logic and action server / message structure. I am also new to ROS2 and Navigation2 and any suggestions would be great help for me :-)

I don't know how helpful this would be regarding your issue but I'd be happy to contribute.

SteveMacenski, 2020-08-14:
I'm happy that you're making use of the AWS world and something analog to this ticket! Would you be interested in potentially taking on this ticket and migrating over to the new world and update the system tests accordingly?

shreyasgokhale, 2020-08-15:
I'd love to take up on the ticket but unfortunately I cannot work on it before November (Master thesis deadlines 😅). If this is okay then sure!

SteveMacenski, 2020-08-17:
Sure, at this point I don't see anyone getting to it before then :wink:.

joe28965, 2022-04-06:
I had a question about this, It's something I could take a look at. However, I saw there were a couple of other "demo" labels in the issues and you said this:

> Later to be expanded to include other hard test items like doors closed needs to open, things falling off shelf, narrow aisles, other robots or people in way, etc.

At this point I'm wondering if it would be smart to make a `nav2_demos` package to add these things to. That way it would be possible to have different launch files for different situations (like the situations you mentioned above), without having 10+ launch files in `nav2_bringup`. It would also be a great package to put the urdf/world files that may or may not be needed.

I'm also wondering what the best robot is to use for this, do you have any ideas? I could also just quickly make one for the simulation, but it won't really have any stl files, so it would be a lot of square boxes.

SteveMacenski, 2022-04-06:
There's ongoing work on it https://github.com/ros-planning/navigation2/pull/2797 but @lucabonamini would be the best to explain any remaining issues / things that need to be worked on before it can be merged. I believe its incredibly close, just needs a little more work on the review comments. 

> At this point I'm wondering if it would be smart to make a nav2_demos

What would you see the role of `nav2_demos` to be? In many ways `nav2_bringup` is the demo of how to create a package for a given robot system and includes launch files that can be used as the basis of people's `<robot>_navigation` configuration packages. What kinds of additional things would you think we should have added?

I'm not sure we need to have many situations setup, just 1-2 "good" ones. Right now our default is not good, this work is to make that first one to be "good". We also don't have the world files in the bringup itself (after this PR linked above) and we use the gazebo world provided by the package now that there are binaries available. 

On the best robot, that is also a point of open discussion. There was some discussion about @padhupradheep Neobotix professional robot and I think there are even binaries for it but I don't think I got things to quite work, but it was mostly there. Certainly once the turtlebot4 comes out, we'd use that over the Turtlebot3, but neither are the most "ideal" to showcase the maturity of the system, although would be convenient for hobbyists.

joe28965, 2022-04-06:
Ah, you're right. It seems I'm a bit late.

As for the role of `nav2_demos`. For my own projects I usually just create a launch file that includes `bringup_launch.py` in my own package (and localization_ navigation_ & rviz_launch.py since they're included in bringup) and add the necessary parameter files via the launch configuration. 

For that reason I thought it would be smart to split that from the demo part, but then again it does say to actually _mirror_ `nav2_bringup` in the readme, and not piggyback on it like I do, so maybe that's just my fault. I guess you can ignore that comment.

As for robots, I had hoped that Clearpath would have merged the PRs for Dingo at this time, then that one would have been a very interesting option for Nav2, but alas. Those PRs have been open for 2 months without any update.

padhupradheep, 2022-04-06:
> On the best robot, that is also a point of open discussion. There was some discussion about @padhupradheep Neobotix professional robot and I think there are even binaries for it but I don't think I got things to quite work, but it was mostly there. 

yes, the only problem with our simulation is that our robots cannot handle the default step size and update rate defined in the gazebo world. For each different worlds that the user decide to use, he/she has to change those parameters. But our directions have changed now and we are focusing on moving to ignition gazebo and taking things further there on.

lucabonamini, 2022-04-09:
> There's ongoing work on it #2797 but @lucabonamini would be the best to explain any remaining issues / things that need to be worked on before it can be merged. I believe its incredibly close, just needs a little more work on the review comments.

Hi @joe28965. As @SteveMacenski  told you, I'm working on a PR to migrate from Turtlebot to [AWS warehouse](https://github.com/aws-robotics/aws-robomaker-small-warehouse-world) world. There's still some minor work to be done (well, let's Steve say it 😄): readd _test_dynamic_obstacle_ with a modified version of warehouse world, and debug _multi_tb3_simulation_launch.py_ which is currently not working due to some error with yaml loading (😡).

joe28965, 2022-04-10:
@SteveMacenski I'm also assuming you'd like it to be a real robot, not just one that exists only in simulation?

@lucabonamini If you need help with either of those things, let me know and I can look at it next week.

lucabonamini, 2022-04-10:
> @lucabonamini If you need help with either of those things, let me know and I can look at it next week.

Great! I'll get back to you in case of need 😃

SteveMacenski, 2022-04-11:
> @SteveMacenski I'm also assuming you'd like it to be a real robot, not just one that exists only in simulation?

I guess it doesn't **have** to be real, but it has to look and feel real (so may as well _be_ real)

joe28965, 2022-04-11:
@SteveMacenski because I was thinking, it's not that difficult to make something a frame that looks OK in SolidWorks. Then I could make a URDF with the base_link, imu, sensors and whatnot, make three different URDFs with diff-drive, ackermann and holonomic. Basically creating the 3 driving types while using the same frame/sensors.

Dunno, thought it might be nice, LMK what you think about it.

SteveMacenski, 2022-04-11:
I think the plan is in the medium term to move to a real robot, maybe the neobotix robot, maybe the new turtlebot4. I don't think its worth creating a fake-bot when there are shortly going to be a couple of real-bots that meet our needs. Neobotix has both diff and omni platforms. Its true that we don't have any good ackermann examples right now, but I think its important that that one is real to add it (since otherwise it doesn't complete fix the concern that there isn't a 'real' example of ackermann in use)

SteveMacenski, 2022-06-13:
https://github.com/ros-planning/navigation2/issues/2997 supersedes in the placeholder, though this work is not yet complete, the plan in that ticket involves merging of https://github.com/ros-planning/navigation2/pull/2797 and migration to the AWS warehouse world. Closing to redirect conversions there