ros-navigation/navigation2 #1937: Multiple nodes with same name

mikeferguson, 2020-08-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy Binaries
- Version or commit hash:
  - 0.4.1-1

#### Steps to reproduce issue
Run the example navigation system, then run ros2 node list
```
ubr@ubr1:~$ ros2 node list
WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.
<redacted>
/waypoint_follower
/waypoint_follower
```

#### Expected behavior
There would be only one waypoint_follower node - or maybe a waypoint_follower and a waypoint_follower_rclcpp_node

#### Actual behavior
There are two waypoint_followers.

#### Additional information
I looked through the code, but I'm not sure where the second node is getting created

SteveMacenski, 2020-08-12:
Can you un-redact the nav2 nodes also you have up? Or are you just launching the waypoint follower node in this context?

mikeferguson, 2020-08-12:
> Can you un-redact the nav2 nodes also you have up? Might help if I can see what else has duplicates, if any. Or are you just launching the waypoint follower node in this context?

Updated - I was just trying to reduce the list - this is basically running https://github.com/mikeferguson/ubr_reloaded/blob/ros2/ubr1_navigation/launch/navigation.launch.py and https://github.com/mikeferguson/ubr_reloaded/blob/ros2/ubr1_bringup/launch/robot.launch.py

SteveMacenski, 2020-08-12:
https://github.com/ros-planning/navigation2/pull/1939 resolves

SteveMacenski, 2020-08-13:
Thanks for  the report~

closed 2020-08-13 by SteveMacenski