ros-navigation/navigation2 #1938: Multiple ComputePathToPose does not work as expected

daisukes, 2020-08-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy src build
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-DDS

#### Steps to reproduce issue

This is an example of BT tree that has multiple `ComputePathToPose`, which lets the robot stick to the initial plan until the path is changed based on `SomeCondition`.

```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <Sequence>
      <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
      <RecoveryNode number_of_retries="6" name="NavigateRecovery">
	<PipelineSequence name="NavigateWithReplanning">
          <RateController hz="1.0">
	    <SomeCondition>
              <RecoveryNode number_of_retries="1" name="ComputePathToPose">
		<ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
		<ClearEntireCostmap name="ClearGlobalCostmap-Context" service_name="global_costmap/clear_entirely_global_costmap"/>
              </RecoveryNode>
	    </SomeCondition>
          </RateController>
          <RecoveryNode number_of_retries="1" name="FollowPath">
            <FollowPath path="{path}" controller_id="FollowPath"/>
            <ClearEntireCostmap name="ClearLocalCostmap-Context" service_name="local_costmap/clear_entirely_local_costmap"/>
          </RecoveryNode>
	</PipelineSequence>
	<ReactiveFallback name="RecoveryFallback">
          <GoalUpdated/>
          <SequenceStar name="RecoveryActions">
            <ClearEntireCostmap name="ClearLocalCostmap-Subtree" service_name="local_costmap/clear_entirely_local_costmap"/>
            <ClearEntireCostmap name="ClearGlobalCostmap-Subtree" service_name="global_costmap/clear_entirely_global_costmap"/>
            <Spin spin_dist="1.57"/>
            <Wait wait_duration="5"/>
	    <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
          </SequenceStar>
	</ReactiveFallback>
      </RecoveryNode>
    </Sequence>
  </BehaviorTree>
</root>
```

#### Expected behavior

The second `ComputePathToPose` inside `SomeCondition` changes the path.

#### Actual behavior

The second `ComputePathToPose` does not affect at first tick because it has `first_time_` variable which controls setting blackboard `path_updated` variable to true.

https://github.com/ros-planning/navigation2/blob/c6bffe853edcbe90a886c25a5e2fbb359869b1b3/nav2_behavior_tree/plugins/action/compute_path_to_pose_action.cpp#L41-L45

#### Additional information

Also, with the default BT XML, `first_time_` works at the very first trial out of 6 trials. 
https://github.com/ros-planning/navigation2/blob/c6bffe853edcbe90a886c25a5e2fbb359869b1b3/nav2_bt_navigator/behavior_trees/navigate_w_replanning_and_recovery.xml#L8
It works though, but in the latter trials, `FollowPath` sends the path as an updated path immediately after it sends the path.

I think it would be better `ComputePathToPose` output `last_updated` time and then `FollowPath` can know whether the path is updated or not based on the `last_updated` time and its status. Or set `path_updated` to true every time.

What do you think?

SteveMacenski, 2020-08-12:
Please rename ticket to be specific to your issue so its searchable in the future. 

I’ll touch back on this later today

SteveMacenski, 2020-08-12:
Thanks, much better title :-) 

This PR introduced it long long ago https://github.com/ros-planning/navigation2/pull/809 by @mjeronimo. I admittedly don't follow his logic on that. If you have a path, and you don't set that flag, the follower won't know about it to set the new path as an option. Though I'd need to verify that with some testing to see what they mean. I don't see why this would cause an immediate preemption since I would think that would trigger the initial plan. With that said, current behavior has almost no delay, so I suspect things are working now so that means testing is required on the default behavior trees.

Separate of the specific BT node, you BT seems to be calling the planner at tick rate (10hz), then calling it at 1 hz throttled, and again after recoveries (and the next step is then to replan at the top of the BT in the next tick a new nanoseconds later). I wonder if this is really what you want. Some of this issue could be resolved with potentially changing your behavior tree but still not losing the same capabilities. I think that last call is redundant, and the 1hz call is completely redundant if you plan at tick rate in the first instance.

Anyhow, 

> I think it would be better ComputePathToPose output last_updated time and then FollowPath can know whether the path is updated or not based on the last_updated time and its status. Or set path_updated to true every time.

I think several of those options could be reasonable solutions. First though, I think we'd have to understand why this was added and what that's trying to protect from.

daisukes, 2020-08-13:
My problem happens in very specific cases like the following.

- Two (or more) `ComputePathToPose`, let's say CPTP-A and CPTP-B

Tick/State|CPTP-A<br>`first_time_`| CPTPe-B<br>`first_time_`| BlackBoard<br>`path_updated`| FollwoPath<br>status | Path
---|---|---|---|---|---
Initial|true|true|false|inactive|-
tick CPTP-A|true->false|true|false|inactive->active|Path-A
tick CPTP-B|false|true->false|false|active|Path-A (*1
tick CPTP-B|false|false|false->true|active|Path-B
FollowPath aborted|false|false|false|active->inactive|-
tick CPTP-A|false|false|false->true|inactive->active|Path-A, Path-A (*2

(*1: This happens because CPTP-B's `first_time_` was true. This is what I want to fix. CPTP-B is ticked for the first time, but the path should be preempted.
(*2: The current code still has a problem after abortion. CPTP `first_time_` is not reset by abortion, so it set `path_updated_`. It leads the situation #809 tried to solve.

SteveMacenski, 2020-08-13:
I understand now, I look a look over https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/action/follow_path_action.cpp and didn't realize we set the on_tick path regardless of the flag on the initial call, which makes sense what happens in the first -A tick, which is where I was confused how it could be working without 1hz delay. I'd support a change to fix this. I think your updated timestamp idea is good and removing the first time.

Though we'd need to do one of two things:
- Set some duration timeout to  reject new paths (if newer than N milliseconds, then ignore) to have the same `first_time_` behavior
- Have the timestamp set not as `now()` when called, but on the timestamp of the iteration of the current BT tick. That way there's only 1 timestamp for the `on_tick` and `on_wait_for_result` because they're part o the same `tick` cycle so the second attempt would be rejected. Though this has an issue if you have multiple planner calls in a given tick iteration and you genuinely want to update for them all (e.g. you have planner A create a path every cycle, but then planner B only does so when a specific situation occurs which needs to take priority over planner A).

So I dont think that solution would itself be the ultimate fix. What are your thoughts on it?

daisukes, 2020-08-14:
I got a simple solution, which does not use blackboard at all.
Please check the PR #1943

closed 2020-08-20 by SteveMacenski