ros-navigation/navigation2 #1944: local_costmap causes controller_server to crash when using custom BT

shreyasgokhale, 2020-08-14:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04  
- ROS2 Version:
 Foxy
- Version or commit hash:
  pull request #1928 fork branch [75af83](https://github.com/ros-planning/navigation2/commit/75af83510ac7bdea264824b2dda228f72eabe608). Based on master branch commit [909c6f](https://github.com/ros-planning/navigation2/commit/909c6f2da7423440d7bd4db3115a82cf8802249a) 

- DDS implementation:
  - rmw_fastrtps_cpp

#### Steps to reproduce issue
- Install Navigation2 master branch from Docu
- Clone custom packages and checkout `foxy-devel` branch. Source nav2 workspace and build.
```
cd ament_workspace/src
git clone https://github.com/shreyasgokhale/CustomRobots.git
git checkout foxy-devel
colcon build --symlink-install
. ./install/setup.sh
```
- Launch `amazon_warehouse_world.py` bringup launch file

```
ros2 launch amazon_robot_bringup amazon_warehouse_world.py use_sim_time:=True autostart:=True 
```
- Localize the robot once it is launched and give a Nav Goal in RViz

#### Expected behavior
Controller server, after receiving a goal, should set a plan which is given to `FollowPath` to follow it until the goal is reached.

#### Actual behavior
After setting up plan, when trying to pass the plan to `FollowPath` Plugin, controller server crashes with `exit code -11`.

```
Planner Path set
[controller_server-9] [INFO] [1597399062.024938368] [controller_server]: Passing new path to controller.
[ERROR] [controller_server-9]: process has died [pid 121717, exit code -11, cmd '/home/ubuntu-foxy/nav_workspace/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmpjdalv4yu -r /tf:=tf -r /tf_static:=tf_static'].
```

#### Additional information
My repo is based on the pre implementation of [bt_waypoint_follower](https://github.com/ros-planning/navigation2/pull/1928) to create my own application using custom behaviour tree and plugins to implement a warehouse scenario of loading and unloading pallets.

When local costmap is not working (disabled), the robot can navigate to the pose and follow the behaviour tree. However, often, the robot is stuck in final approach to achieve the goal pose. This is expected as there are no costmaps.

[These are my ROS node log and gdb backtrace](https://gist.github.com/shreyasgokhale/ddff95cc7db8786035d55f638cc72eac)

Please correct me if I am wrong in my logic. I am new to ROS2 and Navigation2.

*YouTube Videos*

[depicting crash](https://youtu.be/GjXmSB-xVz8)
[working but no local costmap](https://www.youtube.com/watch?v=F1WELXcg364)

SteveMacenski, 2020-08-14:
We don't support non-merged community contributed PRs. Please talk to @ymd-stella about your issues directly on the PR or appropriate ticket. It also sounds like you have a bunch of custom stuff going on, I can't possibly tell what's problematic from your custom setup vs. navigation2 itself. 

We also have this great tutorial on GDB https://navigation.ros.org/tutorials/docs/get_backtrace.html please provide actual tracebacks, a crash report doesn't tell us much.

shreyasgokhale, 2020-08-15:
Thank you for the comment, I'll look into it. 

I updated my navigation2 installation to the latest master branch and the problem seems to persist. I also tried it using docker (just to eliminate any memory issues when using VM). I tried namespacing critics in `dwa_planner` (`"dwb_critics::RotateToGoal"` etc) just in case as I saw the this mentioned in the backtrace.

This is the full backtrace I received after doing everything mentioned.
```
#0  0x00007fedf09b5dc2 in costmap_queue::CostmapQueue::enqueueCell(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int) () from /opt/overlay_ws/install/dwb_critics/lib/libdwb_critics.so
No symbol table info available.
#1  0x00007fedf09b60dc in costmap_queue::CostmapQueue::getNextCell() () from /opt/overlay_ws/install/dwb_critics/lib/libdwb_critics.so
No symbol table info available.
#2  0x00007fedf09a23d8 in dwb_critics::MapGridCritic::propogateManhattanDistances() () from /opt/overlay_ws/install/dwb_critics/lib/libdwb_critics.so
No symbol table info available.
#3  0x00007fedf09a63cb in dwb_critics::GoalDistCritic::prepare(geometry_msgs::msg::Pose2D_<std::allocator<void> > const&, nav_2d_msgs::msg::Twist2D_<std::allocator<void> > const&, geometry_msgs::msg::Pose2D_<std::allocator<void> > const&, nav_2d_msgs::msg::Path2D_<std::allocator<void> > const&) () from /opt/overlay_ws/install/dwb_critics/lib/libdwb_critics.so
No symbol table info available.
#4  0x00007fedf09a8b13 in dwb_critics::GoalAlignCritic::prepare(geometry_msgs::msg::Pose2D_<std::allocator<void> > const&, nav_2d_msgs::msg::Twist2D_<std::allocator<void> > const&, geometry_msgs::msg::Pose2D_<std::allocator<void> > const&, nav_2d_msgs::msg::Path2D_<std::allocator<void> > const&) () from /opt/overlay_ws/install/dwb_critics/lib/libdwb_critics.so
No symbol table info available.
#5  0x00007fedf0ac2e0c in dwb_core::DWBLocalPlanner::computeVelocityCommands(nav_2d_msgs::msg::Pose2DStamped_<std::allocator<void> > const&, nav_2d_msgs::msg::Twist2D_<std::allocator<void> > const&, std::shared_ptr<dwb_msgs::msg::LocalPlanEvaluation_<std::allocator<void> > >&) () from /opt/overlay_ws/install/dwb_core/lib/libdwb_core.so
No symbol table info available.
#6  0x00007fedf0abbd46 in dwb_core::DWBLocalPlanner::computeVelocityCommands(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::Twist_<std::allocator<void> > const&) ()
   from /opt/overlay_ws/install/dwb_core/lib/libdwb_core.so
No symbol table info available.
#7  0x000055aa5927a2dd in nav2_controller::ControllerServer::computeAndPublishVelocity() ()
No symbol table info available.
#8  0x000055aa5927afc6 in nav2_controller::ControllerServer::computeControl() ()
No symbol table info available.
#9  0x000055aa592a2486 in nav2_util::SimpleActionServer<nav2_msgs::action::FollowPath, rclcpp::Node>::work() ()
No symbol table info available.
#10 0x000055aa592a3450 in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::FollowPath, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::FollowPath> >)::{lambda()#1}> >, void> >::_M_invoke(std::_Any_data const&) ()
No symbol table info available.
#11 0x000055aa5928205d in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) ()
No symbol table info available.
#12 0x00007fedf7b7c47f in __pthread_once_slow (once_control=0x7feddc004838, init_routine=0x7fedf7a50b50 <__once_proxy>) at pthread_once.c:116
        _buffer = {__routine = 0x7fedf7b7c4d0 <clear_once_control>, __arg = 0x7feddc004838, __canceltype = 0, __prev = 0x0}
        val = <optimized out>
        newval = <optimized out>
#13 0x000055aa59289ee8 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::FollowPath, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::FollowPath> >)::{lambda()#1}> >, void>::_Async_state_impl(std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::FollowPath, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::FollowPath> >)::{lambda()#1}>&&)::{lambda()#1}> > >::_M_run() ()
No symbol table info available.
#14 0x00007fedf7a51cb4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#15 0x00007fedf7b73609 in start_thread (arg=<optimized out>) at pthread_create.c:477
        ret = <optimized out>
        pd = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140659156305664, -2334653617915902539, 140659958314014, 140659958314015, 140659958314016, 140659156301632, 2324458771923518901, 2324503409558341045}, 
              mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
#16 0x00007fedf773f103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

SteveMacenski, 2020-08-17:
> dwa_planner

I hope that's a typo, if not, that could be your issue. There is no dwa planner. 

> No symbol table info available.

You have this in every single line, I think you have some other issues going on unrelated to this. I have no idea given your custom changes what would cause this. I'd recommend starting with the getting started on the website section and making changes one thing at a time to your custom situation until you find your issue.

SteveMacenski, 2020-08-19:
It also occurs to me that you don't source your /opt/ros/foxy/setup.bash before building your workspace. While I think that would cause it to fail, maybe something odd caused it to build but then nothing correctly linked.

shreyasgokhale, 2020-08-20:
Hi @SteveMacenski ,

Thanks for your suggestions. Yes, `dwa_planner` was a typo.

I was able to trace the problem back to `clock`. Gazebo was not publishing clock because `libgazebo_ros_init.so` was not included in launch arguments. I also made sure that I am applying `use_sim_time` in every single place. But that did not solve all the issues.

Now, I can see local and global costmaps. But somehow, the robot position does not correspond to the position in gazebo and this causes the goal to be missed.

[image]

[image]

Moreover, rviz crashes shortly after, giving following errors.

```bash
[recoveries_server-11] [INFO] [1597944789.194517157] [rclcpp]: signal_handler(signal_value=2)
[recoveries_server-11] [foonathan::memory] Allocator foonathan::memory::new_allocator (at (nil)) leaked 12024 bytes.
[recoveries_server-11] [foonathan::memory] Allocator foonathan::memory::heap_allocator (at (nil)) leaked 843056 bytes.
[planner_server-10] [foonathan::memory] Allocator foonathan::memory::new_allocator (at (nil)) leaked 12024 bytes.
[planner_server-10] [foonathan::memory] Allocator foonathan::memory::heap_allocator (at (nil)) leaked 1443280 bytes.
```

 But I think it might be related to my custom stuff rather than navigation stack. So maybe we can close this if you think the same?

SteveMacenski, 2020-08-20:
> Now, I can see local and global costmaps. But somehow, the robot position does not correspond to the position in gazebo and this causes the goal to be missed.

What do you mean by that? 

Please provide more full logs of the incident, not just a couple of lines, that doesn't give us context.

> But I think it might be related to my custom stuff rather than navigation stack. So maybe we can close this if you think the same?

Perhaps, did you try with the default tree and such? If that works, then yes its your custom stuff and we should close (unless you plan to open source it)

SteveMacenski, 2020-08-24:
SLAM has nothing to do with your local costmap or custom BT node. Please stay on topic. Revert back to the demo world with the TB3 and see if you can reproduce your issues with your BT / costmap you initially described. We're not going to debug 5 problems at once you may be seeing.

shreyasgokhale, 2020-08-25:
Hi @SteveMacenski ,

I tried reverting back and it seems that I can't reproduce this issue. I assume that this is likely solved with the steps I tried earlier. We can, therefore, close this.

closed 2020-08-25 by SteveMacenski