ros-navigation/navigation2 #1948: [Eloquent] Undefined symbol when launching bt_navigator

QQting, 2020-08-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- ROS2 Version:
  - Eloquent
- Version or commit hash:
  - ros-eloquent-navigation2	0.3.5-1bionic.20200630.140800
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue

1. Get ADLINK NeuronBot2 source codes
```sh
mkdir -p ~/neuronbot2_ros2_ws/src
cd ~/neuronbot2_ros2_ws/
wget https://raw.githubusercontent.com/Adlink-ROS/neuronbot2_ros2.repos/eloquent-devel/neuronbot2_ros2.repos
vcs import src < neuronbot2_ros2.repos
```
2. Ignore BT.CPP source code which is the workaround to fix undefined symbol
```sh
touch ~/neuronbot2_ros2_ws/src/behavior_tree.cpp/COLCON_IGNORE
```

3.  Build code
```sh
source /opt/ros/eloquent/setup.bash
colcon build --symlink-install
```

4. Bringup Gazebo simulaton
```sh
ros2 launch neuronbot2_gazebo neuronbot2_world.launch.py
```

5. Bringup navigation
```sh
ros2 launch neuronbot2_nav bringup_launch.py use_sim_time:=true open_rviz:=true
```

6. You will see the error messages as below: 
`[bt_navigator-7] [ERROR] []: Original error: Could not load library: /opt/ros/eloquent/lib/libnav2_compute_path_to_pose_action_bt_node.so: undefined symbol: _ZTIN2BT14CoroActionNodeE
`
[image]

The way to solve this issue is to get the latest BehaviorTree.CPP and rebuild it:
```sh
rm ~/neuronbot2_ros2_ws/src/behavior_tree.cpp/COLCON_IGNORE
cd ~/neuronbot2_ros2_ws/
source /opt/ros/eloquent/setup.bash
colcon build --symlink-install
```

SteveMacenski, 2020-08-17:
We've brought this up in a number of places but I understand that we haven't had any high level directives to users about this.

http://build.ros2.org/job/Ebin_uB64__behaviortree_cpp_v3__ubuntu_bionic_amd64__binary/

If you see, the BT.CPP released (https://github.com/ros/rosdistro/blob/master/eloquent/distribution.yaml#L247) is 3.5.1, which is what we support. This was released and sent to the build farm for debians. However, as you can see from that job manifest, it was failing for a long time and only recently(ish) made to work. However, there's been no ROS2 Eloquent sync since then for the new binaries to be installable on your machine, so you need to build from source for now.

Ping Davide on BT.CPP about this to make sure its included on the next sync, but there's nothing further for us to do here. You've found the right / best solution on your own to this issue.

SteveMacenski, 2020-08-18:
Closing - known solution.

closed 2020-08-18 by SteveMacenski

Phymin, 2020-09-24:
Hi, what's the status of the problem now? I still encounter this problem now

ipa-kut, 2021-03-11:
> Hi, what's the status of the problem now? I still encounter this problem now

If you still need to get this to work, as steve mentions above, I tried to build both [navigation2 -b eloquent devel](https://github.com/ros-planning/navigation2/tree/eloquent-devel) and [BehaviorTree.cpp -b master](https://github.com/BehaviorTree/BehaviorTree.CPP) from source. This could also work for the neuronbot then.

dverstandig, 2022-08-23:
Hello,

We are building the BehaviorTree.cpp (3.7.0 - 8db5593a880c60211a8871bfe4da5b2507f726b3) package from source and getting the navigation packages from debian (eloquent-navigation2), but are encountering a malloc issue when launching the BT navigator:
`[bt_navigator-4] malloc(): memory corruption`

Any ideas what the issue could be?