ros-navigation/navigation2 #1950: nav2_bringup tb3_simulation_launch.py crashes randomly during execution

rightbot-abhinav, 2020-08-17:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- ROS2 Version:
  - Foxy source
- Version or commit hash:
- DDS implementation:
  -  PLATFORM INFORMATION
system           : Linux
platform info    : Linux-4.9.140-tegra-aarch64-with-Ubuntu-18.04-bionic
release          : 4.9.140-tegra
processor        : aarch64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : foxy
distribution type      : ros2
distribution status    : active
release platforms      : {'ubuntu': ['focal']}

   ROS 2 INFORMATION
distribution name      : foxy
distribution type      : ros2
distribution status    : active
release platforms      : {'ubuntu': ['focal']}

gazebo version 11 installed from binary 

#### Steps to reproduce issue

ros2 launch nav2_bringup tb3_simulation_launch.py
set pose estimation
set way point (in the gui) and it crashes randomly at different instances 

#### Expected behavior
nav2bringup should not crash randomly after setting waypoint or pose estimation 
#### Actual behavior
ros2 launch nav2_bringup tb3_simulation_launch.py 
should not crash randomly when running like putting a waypoint etc 

#### Additional information
Logs 
Logs below

[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [13225]
[INFO] [gzclient-2]: process started with pid [13226]
[INFO] [robot_state_publisher-3]: process started with pid [13227]
[INFO] [rviz2-4]: process started with pid [13228]
[INFO] [map_server-5]: process started with pid [13229]
[INFO] [amcl-6]: process started with pid [13230]
[INFO] [lifecycle_manager-7]: process started with pid [13231]
[INFO] [controller_server-8]: process started with pid [13232]
[INFO] [planner_server-9]: process started with pid [13234]
[INFO] [recoveries_server-10]: process started with pid [13235]
[INFO] [bt_navigator-11]: process started with pid [13240]
[INFO] [waypoint_follower-12]: process started with pid [13241]
[INFO] [lifecycle_manager-13]: process started with pid [13244]
[robot_state_publisher-3] [WARN] [1597645737.442451776] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 7 children
[robot_state_publisher-3] Link camera_link had 2 children
[robot_state_publisher-3] Link camera_depth_frame had 1 children
[robot_state_publisher-3] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-3] Link camera_rgb_frame had 1 children
[... 573 lines cut ...]
[controller_server-8] 
[controller_server-8] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[controller_server-8] This error state is being overwritten:
[controller_server-8] 
[controller_server-8]   'rcl node's context is invalid, at /home/warry/ros2_foxy/src/ros2/rcl/rcl/src/rcl/node.c:441'
[controller_server-8] 
[controller_server-8] with this new error message:
[controller_server-8] 
[controller_server-8]   'publisher's context is invalid, at /home/warry/ros2_foxy/src/ros2/rcl/rcl/src/rcl/publisher.c:395'
[controller_server-8] 
[controller_server-8] rcutils_reset_error() should be called after error handling to avoid this.
[controller_server-8] <<<
[waypoint_follower-12] "/home/warry/ros2_foxy/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/listener_thread.cpp":__function__:157"__rmw_take failed": ros discovery info listener thread will shutdown ...
[INFO] [map_server-5]: process has finished cleanly [pid 13229]
[lifecycle_manager-7] [INFO] [1597645826.782433706] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-7] [INFO] [1597645826.792595008] [lifecycle_manager_localization]: Destroying
[amcl-6] [INFO] [1597645826.800759288] [rclcpp]: signal_handler(signal_value=2)
[amcl-6] [INFO] [1597645826.820157029] [amcl]: Destroying
[INFO] [amcl-6]: process has finished cleanly [pid 13230]
[map_server-5] [INFO] [1597645826.834356727] [rclcpp]: signal_handler(signal_value=2)
[map_server-5] [INFO] [1597645826.838548467] [map_server]: Destroying
[robot_state_publisher-3] [INFO] [1597645826.861574876] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [bt_navigator-11]: process has died [pid 13240, exit code -6, cmd '/home/warry/navigation2_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator --params-file /tmp/tmpzi07bl24 -r /tf:=tf -r /tf_static:=tf_static'].
[INFO] [gzserver-1]: process has finished cleanly [pid 13225]
[INFO] [gzclient-2]: process has finished cleanly [pid 13226]

rightbot-abhinav, 2020-08-17:
Tried changing the DDS middleware to cyclone issue does not pop up for now. @SteveMacenski

SteveMacenski, 2020-08-17:
>  [ERROR] [rviz2-4]: process has died [pid 13228, exit code -11, cmd '/home/warry/ros2_foxy/install/rviz2/lib/rviz2/rviz2 -d /home/warry/navigation2_ws/install/nav2_bringup/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2'].

This is an rviz error, your rviz session crashed bringing everything else down. This is not a navigation2 issue. Please file a ticket with any additional debug information (e.g. backtrace) to rviz2 itself.

closed 2020-08-17 by SteveMacenski

JaimeMartin, 2020-08-19:
Hi @SteveMacenski , 

Actually this user is reporting also crashes also with Cyclone DDS in navigation2 for example:

https://github.com/ros-planning/navigation2/issues/1955

As this is totally random and the stack trace does not show any hint that Fast DDS is failing, but rviz, I don't think you can assign the issue to Fast DDS.

rightbot-abhinav, 2020-08-19:
@JaimeMartin the issue I raised earlier #1955 was initially with fastrtps I changed to cyclonedds the crashing issue was solved. Currently I am facing separate other issues which dont make the stack crash but it generates weird behavior as I mentioned in #1955 . Also as @SteveMacenski rightly mentioned me changing names and raising issues wont solve hence I am trying to build a local stable stack for the officially not supported OS 18.04

rightbot-abhinav, 2020-08-19:
@JaimeMartin Created a different setup on Focal. The same issue occurs everything after a few mins of the launch. Changing to cyclonedds saves from crashing

rsanchez15, 2020-09-04:
Hi @rightbot-abhinav,
 
I've been running some tests in the scenario you describe in the issue with Ubuntu Focal (ROS 2 Foxy Tier 1 platform), and I haven't been able to reproduce the problem you mention with Fast DDS as the ROS 2 middleware implementation. 

To replicate your scenario I have followed the following steps.
* Installation:
  - ROS 2 Foxy from sources using the `ros2.repos` file updated in commit https://github.com/ros2/ros2/commit/2a98a560ec04786d00815bb1efea42220509cbc1.
  - navigation2 from binaries (latest stable version installed via apt).
  - TurtleBot3 from sources.
* Launch:
  - Fast DDS: `RMW_IMPLEMENTATION=rmw_fastrtps_cpp ros2 launch nav2_bringup tb3_simulation_launch.py`
  - Cyclone DDS: `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 launch nav2_bringup tb3_simulation_launch.py`
* Execution:
  - Set 2D Pose Estimation
  - Set Navigation2 Goal
* Duration of execution: ~1 hour.

It is true that I've found some navigation related error traces that occur using any of the DDS implementations, both with Fast DDS and Cyclone DDS, which indicates that the problem is not related to the middleware, as @JaimeMartin stated before. In any case, the execution of the program is never interrupted.