ros-navigation/navigation2 #1954: Collision detection tests are failing

SteveMacenski, 2020-08-18:
https://app.circleci.com/pipelines/github/ros-planning/navigation2/3629/workflows/4f8ce890-f57d-4718-9ab1-569449c08225/jobs/14222

This is failing regularly and its causing alot of hassle the last month or so. It didn't seem flaky before then.

CC @shivaang12

shivaang12, 2020-08-18:
@SteveMacenski @naiveHobo Still cannot reproduce it locally. I might further try creating a draft PR.

shivaang12, 2020-08-25:
After running tests in docker container, this test exhibit flaky behavior when all tests are executed `colcon test`. But passing without any problem when directly executed `colcon test --packages-select nav2_costmap_2d`

Need to see when this test when flaky.

shivaang12, 2020-08-31:
After some thorough inspection, I found that after 94615e7b412e3d947b888a6d799aa6e6f78a9031 commit this test started to failing. Still working on to find what change in this commit triggered this behavior.

mikeferguson, 2020-08-31:
In my experience, flaky tests often boil down to one of two main causes: timing issues with the test, or memory issues in either the test or tested code (such as uninitialized variables).

I'd suggest running valgrind on the test since memory issues will be easier to track down than timing ones. However, the fact that an entirely unrelated set of changes (as referenced above) appears to be the start of these failures could point towards timing issues.

mikeferguson, 2020-08-31:
Also, here's a nasty one (invalid iterator) that I found with valgrind during ROS1 noetic release. Doesn't seem to be happening on this repo (which is interesting), but I figured I would port it over: https://github.com/ros-planning/navigation2/pull/1969

shivaang12, 2020-09-12:
@mikeferguson Do you how can we integrate/use valgrind in to ros2's launch interface?

mikeferguson, 2020-09-12:
> Do you how can we integrate/use valgrind in to ros2's launch interface?

I have not figured out how to use an argument to select it at runtime, but you can see the commented out version for GDB here: https://github.com/mikeferguson/ubr_reloaded/blob/ros2/ubr1_bringup/launch/robot.launch.py#L76 - valgrind command should follow the same launch prefixes from ROS1

wilcobonestroo, 2020-09-15:
Steps to run valgrind manually:
build and source navigation in debug mode:
`colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Debug`
`. install/setup.bash`
`valgrind ./build/nav2_costmap_2d/test/integration/inflation_tests_exec`

Valgrind reported some `Conditional jump or move depends on uninitialised value(s)` issues in `range_test_exec`. I don't know if it is related to the issue described above. However, it definitely results in unpredictable behavior in the tests. I will make a separate issue for it with the error output.

SteveMacenski, 2020-09-15:
@wilcobonestroo does that relate to the ticket?

wilcobonestroo, 2020-09-16:
@SteveMacenski I still don't know. In general I see that the tests are not predictable at all at the moment. When I run them in isolation, they all succeed. However, when I run them together `colcon test` the results are highly unpredictable. Moreover, when I run 6 `colcon test`s at the same time I get even more failures. I think this issue of unpredictable behavior in tests is bigger than this specific ticket.

Solving the uninitialized values from #1987 did not help in making the tests more deterministic. I still get errors in the consmaps. I noticed that I get more failures when "ros things" are involved. For example, subscribing to topics, using actions and spinning in the tests results in sometimes failing and sometimes succeeding.

closed 2020-09-23 by SteveMacenski