ros-navigation/navigation2 #1961: ComputePathToPose action server freeze with deadlock

daisukes, 2020-08-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy release src build https://github.com/ros2/ros2/commit/4109e26bf96d0a0a9ef755364dabc851556f3089
- Version or commit hash:
  - main
- DDS implementation:
  - Fast-RTPS, cyclonedds

#### Steps to reproduce issue

- launch tb3 simulation and make it active
- repeat sending goal action and waiting for the result in a single loop, and wait for the deadlock 
https://gist.github.com/daisukes/46298b083a48e5db7016f3c0efba7e2d
  - I could reproduce the deadlock with this code usually in a few minutes

#### Expected behavior

- does not stop, keep working

#### Actual behavior

- the server will not accept goals when the deadlock happens (

#### Additional information
Here is gdb information
Thread 9 and 14 are trying to hold two mutex in different order
https://gist.github.com/daisukes/712316a97832f5d9ab851ad47c77ad98

Thread 9: server thread
Frame# 6 [nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::handle_goal](https://github.com/ros-planning/navigation2/blob/2e0697c64c265f31bd4ef6c5ed43374841242077/nav2_util/include/nav2_util/simple_action_server.hpp#L82)
Frame# 14 [rclcpp_action::ServerBase::execute_goal_request_received()](https://github.com/ros2/rclcpp/blob/4a5eed968ca6fa4979214db8ac945fb163919326/rclcpp_action/src/server.cpp#L213)

Thread 14: working thread [started here](https://github.com/CMU-cabot/navigation2/blob/db7cea3e5e/nav2_util/include/nav2_util/simple_action_server.hpp#L128)
Frame# 6 [rclcpp_action::ServerBase::notify_goal_terminal_state()](https://github.com/ros2/rclcpp/blob/4a5eed968ca6fa4979214db8ac945fb163919326/rclcpp_action/src/server.cpp#L545)
Frame# 11 [nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::succeeded_current](https://github.com/ros-planning/navigation2/blob/2e0697c64c265f31bd4ef6c5ed43374841242077/nav2_util/include/nav2_util/simple_action_server.hpp#L305)

This happens once in ten navigations in my environment. 
Luckily, I could find the deadlock loop, but I have no idea how to fix this so far.

daisukes, 2020-08-20:
I also checked the goal handle UUIDs in both threads and there are two different goal handles.
This means, somehow the server could accept new goal while finalizing the previous goal.

I have examined my BT and found there is one possible case calling ComputePathToPose twice in a row. 
I will test the fixed BT with my robot tomorrow and hope this will help me.

SteveMacenski, 2020-08-20:
I don't think this will fix the issue, but I'm curious if you removed the `current_handle_.reset();` after `current_handle_->succeed(result);` if that would resolve things. I think the `succeed()` is what allows for you to get a result and then you immediately replan, maybe you are able to catch it while still in the function because `reset()` is still happening at the same time your next immediate goal is being processed.

SteveMacenski, 2020-08-24:
How'd it go?

daisukes, 2020-08-24:
I removed that line, but it still happens.

```
#0  __lll_lock_wait (futex=futex@entry=0x55a9c4b83c70, private=0) at lowlevellock.c:52
#1  0x00007f0ed172d131 in __GI___pthread_mutex_lock (mutex=0x55a9c4b83c70) at ../nptl/pthread_mutex_lock.c:115
#2  0x00007f0ed1ca6a8f in __gthread_mutex_lock(pthread_mutex_t*) () from /opt/ros/foxy/rclcpp/lib/librclcpp.so
#3  0x00007f0ed1ca9d18 in __gthread_recursive_mutex_lock(pthread_mutex_t*) () from /opt/ros/foxy/rclcpp/lib/librclcpp.so
#4  0x00007f0ed1ca9e96 in std::recursive_mutex::lock() () from /opt/ros/foxy/rclcpp/lib/librclcpp.so
#5  0x00007f0ed1cab546 in std::lock_guard<std::recursive_mutex>::lock_guard(std::recursive_mutex&) () from /opt/ros/foxy/rclcpp/lib/librclcpp.so
#6  0x00007f0ed11149f0 in rclcpp_action::ServerBase::notify_goal_terminal_state() () from /opt/ros/foxy/rclcpp_action/lib/librclcpp_action.so
#7  0x00007f0ed20dc860 in rclcpp_action::Server<nav2_msgs::action::ComputePathToPose>::call_goal_accepted_callback(std::shared_ptr<rcl_action_goal_handle_t>, std::array<unsigned char, 16ul>, std::shared_ptr<void>)::{lambda(std::array<unsigned char, 16ul> const&, std::shared_ptr<void>)#1}::operator()(std::array<unsigned char, 16ul> const&, std::shared_ptr<void>) const (result_message=..., uuid=..., 
    this=<optimized out>) at /usr/include/c++/9/ext/atomicity.h:82
#8  std::_Function_handler<void (std::array<unsigned char, 16ul> const&, std::shared_ptr<void>), rclcpp_action::Server<nav2_msgs::action::ComputePathToPose>::call_goal_accepted_callback(std::shared_ptr<rcl_action_goal_handle_t>, std::array<unsigned char, 16ul>, std::shared_ptr<void>)::{lambda(std::array<unsigned char, 16ul> const&, std::shared_ptr<void>)#1}>::_M_invoke(std::_Any_data const&, std::array<unsigned char, 16ul> const&, std::shared_ptr<void>&&) (__functor=..., __args#0=..., __args#1=...) at /usr/include/c++/9/bits/std_function.h:300
#9  0x00007f0ed20b632d in std::function<void (std::array<unsigned char, 16ul> const&, std::shared_ptr<void>)>::operator()(std::array<unsigned char, 16ul> const&, std::shared_ptr<void>) const (
    __args#1=std::shared_ptr<void> (empty) = {...}, __args#0=..., this=0x7f0eac007880) at /usr/include/c++/9/bits/std_function.h:683
#10 rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose>::succeed (result_msg=..., this=0x7f0eac007820) at /opt/ros/foxy/rclcpp_action/include/rclcpp_action/server_goal_handle.hpp:194
#11 nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::succeeded_current (this=0x55a9c45930a0, 
    result=std::shared_ptr<struct nav2_msgs::action::ComputePathToPose_Result_<std::allocator<void> >> (use count 3, weak count 0) = {...})
    at /home/developer/nav2_ws/install/nav2_util/include/nav2_util/simple_action_server.hpp:309
```
Actually, it is locked in this line.
https://github.com/ros-planning/navigation2/blob/10598d562958432f5767b5bb624079e3b9b719fe/nav2_util/include/nav2_util/simple_action_server.hpp#L309

SteveMacenski, 2020-08-24:
Yeah its locked in the succeed, but I think the way to fix this is in the new goal processing line not the exiting logic. There's a new goal coming from your script before the last one was totally completed, that's the core issue, so I think the fix should be in fixing how new goals are processed or removing any delays in completion. 

I think this may be worth elevating to the rclcpp library since this will be an issue for other users too. You have to lock the goal handle and if the locking in the action process creates a deadlock, then the fix will probably have to come from both our sides. 

Any suggestions? (Other than adding some small sleep between sending a new goal and the last goal)

daisukes, 2020-08-25:
The problem is having a lock for managing multiple goals and a lock for updating goal state simultaneously.
I believe the goal handle itself is threadsafe, so only we need to do is managing multiple goals.

`SimleActionServer` has `current_handle_` and `pending_handle_` to manage incoming goals, but I would use a vector to keep all incoming goals. When it receives a goal, it locks for adding the goal into the vector. When it completes an action, it first changes the goal state without locking and then removing the goal from the vector with locking. This may help to separate managing multiple goals from updating goal state.

What do you think?

SteveMacenski, 2020-08-25:
I don't agree about the vector of incoming goals, I think that still has the same issues, you'd have to lock the vector to push and remove things, the deadlock case would still occur. If you think we don't need the locks for the goal handles at all (which I'd also have to do some digging on), then I don't see why the vector vs what's there now impacts things.

http://docs.ros2.org/dashing/api/rclcpp_action/classrclcpp__action_1_1ClientGoalHandle.html at least these docs don't mention anything about being thread safe.

I still think you should file a ticket in rclcpp. Even if we remove the mutexes completely, the fact that 1 goal handle can be both returning a result and a new action being processed seems like a bug. Neither of the functions you listed in the simple wrapper involve swapping out the pending -> current goals. In fact from your demo code, there's no use of the pending handle ever since there's never a pending goal requested until the last is completed.

daisukes, 2020-08-26:
I have tried to separate the lock. 
This does not cause any deadlock so far but, with the test program, it ends up with different problems. 
https://github.com/CMU-cabot/navigation2/commit/5ee6f7ef80d1d52b8f107f5ec3f04574bd5d7c0a

Now, I agree with filing a ticket in rclcpp.
rclcpp action calls user callbacks in their lock context, which causes the problem.

SteveMacenski, 2020-08-26:
Thanks, make sure to keep on it - sometimes `rclcpp` tickets don't get responded to very quickly.

SteveMacenski, 2020-10-06:
@daisukes can you give us and update on the status of this issue?

daisukes, 2020-10-07:
@SteveMacenski still working. I pushed a new commit for a small possible issue. I hope it will be merged soon.

SteveMacenski, 2021-01-14:
ROS 2 master has been updated! Thanks @daisukes for working through this for so long! Major issue averted

closed 2021-01-14 by SteveMacenski