ros-navigation/navigation2 #1963: Changing NavigateToPose behavior at runtime

karanchawla, 2020-08-20:
## Feature request
I've implemented waypoint following using a behavior tree. However, at this point, I'd like to change the behavior of the navigation stack (`controller` and `planner_server`) based on some external condition — is it possible to do this with the existing bt_navigator implementation? 

#### Implementation considerations
One way to do this would be to change the behavior tree at runtime by providing a path to the new XML file which loads a different set of plugins.

SteveMacenski, 2020-08-20:
EZ: https://github.com/ros-planning/navigation2/blob/main/nav2_msgs/action/NavigateToPose.action#L3 we already allow you to do that. Have your waypoint follower BT just specify a BT file path in the request, the BT navigator will then load the new tree courtesy of @fmrico

karanchawla, 2020-08-20:
Sweet! Thanks for the prompt response. Yes, I'm actually just starting to incorporate `navigation2` into our stack for ground autonomy :)

SteveMacenski, 2020-08-20:
Awesome! Let me know if you need anything else. Is this good to close?

karanchawla, 2020-08-21:
Thanks a ton! Will close this out and say hi in the slack group :)

closed 2020-08-21 by karanchawla