ros-navigation/navigation2 #1966: Server map_saver was unable to be reached after 4.00s by bond. This server may be misconfigured.

jediofgever, 2020-08-25:
## Bug report

**Required Info:**

- Operating System:
 Ubuntu 20.04
- ROS2 Version:
Foxy
- Version or commit hash:
  from source: output of `git -C navigation2 rev-parse HEAD

- DDS implementation:
   ( Fast-RTPS,.) 

#### Steps to reproduce issue

SteveMacenski, 2020-08-25:
Please add steps to reproduce.

jediofgever, 2020-08-25:
Hi @SteveMacenski , I am using `foxy` and `tutlebot3_simulations` binary package is not available for foxy, so this package needs to present in ws;
`git clone -b ros2 https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git`

After the build;
`ros2 launch turtlebot3_gazebo turtlebot3_house.launch.py `
which starts gazebo simulation with turtlebot3 already spawned in. 
 In a separate terminal I do; 

`ros2 launch nav2_bringup slam_launch.py `

which produces the following output; 
```

[INFO] [launch]: All log files can be found below /home/atas/.ros/log/2020-08-25-13-04-11-142975-atas-G7-7590-125471
[INFO] [launch]: Default logging verbosity is set to INFO
/home/atas/ros2_foxy/install/nav2_bringup/share/nav2_bringup/launch/slam_launch.py:77: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  start_map_saver_server_cmd = Node(
/home/atas/ros2_foxy/install/nav2_bringup/share/nav2_bringup/launch/slam_launch.py:83: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  start_lifecycle_manager_cmd = Node(
/home/atas/ros2_foxy/install/nav2_bringup/share/nav2_bringup/launch/slam_launch.py:83: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  start_lifecycle_manager_cmd = Node(
/opt/ros/foxy/share/slam_toolbox/launch/online_sync_launch.py:16: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  start_sync_slam_toolbox_node = Node(
[INFO] [sync_slam_toolbox_node-1]: process started with pid [125473]
[INFO] [map_saver_server-2]: process started with pid [125475]
[INFO] [lifecycle_manager-3]: process started with pid [125477]
[lifecycle_manager-3] [INFO] [1598331851.305335426] [lifecycle_manager_slam]: Creating
[sync_slam_toolbox_node-1] [INFO] [1598331851.317805102] [slam_toolbox]: Node using stack size 40000000
[lifecycle_manager-3] [INFO] [1598331851.324848853] [lifecycle_manager_slam]: Creating and initializing lifecycle service clients
[map_saver_server-2] [INFO] [1598331851.325848937] [map_saver]: 
[map_saver_server-2]    map_saver lifecycle node launched. 
[map_saver_server-2]    Waiting on external lifecycle transitions to activate
[map_saver_server-2]    See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_saver_server-2] [INFO] [1598331851.326112749] [map_saver]: Creating
[lifecycle_manager-3] [INFO] [1598331851.327124785] [lifecycle_manager_slam]: Starting managed nodes bringup...
[lifecycle_manager-3] [INFO] [1598331851.327202795] [lifecycle_manager_slam]: Configuring map_saver
[map_saver_server-2] [INFO] [1598331851.327747560] [map_saver]: Configuring
[lifecycle_manager-3] [INFO] [1598331851.332690493] [lifecycle_manager_slam]: Activating map_saver
[map_saver_server-2] [INFO] [1598331851.332844228] [map_saver]: Activating
[sync_slam_toolbox_node-1] [INFO] [1598331851.352416941] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1598331851.352655317] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[lifecycle_manager-3] [ERROR] [1598331853.335897684] [lifecycle_manager_slam]: Server map_saver was unable to be reached after 4.00s by bond. This server may be misconfigured.
[lifecycle_manager-3] [ERROR] [1598331853.335947569] [lifecycle_manager_slam]: Failed to bring up all requested nodes. Aborting bringup.
[sync_slam_toolbox_node-1] Info: clipped range threshold to be within minimum and maximum range!
[sync_slam_toolbox_node-1] [WARN] [1598331854.467245473] [slam_toolbox]: maximum laser range setting (20.0 m) exceeds the capabilities of the used Lidar (3.5 m)
[sync_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[lifecycle_manager-3] terminate called after throwing an instance of 'statemap::TransitionUndefinedException'
[lifecycle_manager-3]   what():  no such transition in current state
[ERROR] [lifecycle_manager-3]: process has died [pid 125477, exit code -6, cmd '/home/atas/ros2_foxy/install/nav2_lifecycle_manager/lib/nav2_lifecycle_manager/lifecycle_manager --ros-args -r __node:=lifecycle_manager_slam -r __node:=lifecycle_manager_slam --params-file /tmp/launch_params_vtpsp__s --params-file /tmp/launch_params_1g8nn0ph --params-file /tmp/launch_params_hb_6879p'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[map_saver_server-2] [INFO] [1598331888.001576401] [rclcpp]: signal_handler(signal_value=2)
[map_saver_server-2] [INFO] [1598331888.003269622] [map_saver]: Destroying
[sync_slam_toolbox_node-1] [INFO] [1598331888.001586354] [rclcpp]: signal_handler(signal_value=2)
[INFO] [sync_slam_toolbox_node-1]: process has finished cleanly [pid 125473]
[INFO] [map_saver_server-2]: process has finished cleanly [pid 125475]
```

SteveMacenski, 2020-08-25:
Try these changes https://github.com/ros-planning/navigation2/pull/1968 it looks like I missed a server in my bond PR

jediofgever, 2020-08-26:
Thanks , it addresses the issue

closed 2020-08-26 by SteveMacenski