ros-navigation/navigation2 #1967: sensor_msgs/PointCloud is deprecated in Foxy

mikeferguson, 2020-08-25:
Much of our visualization is based on the older sensor_msgs/PointCloud message type (costmap has a number of data visualization topics that use it, as does the DWB controller). In ROS Foxy this [message has been deprecated](https://github.com/ros2/common_interfaces/commit/b30ed13005098011836f8194f6a1a0221bf82240#diff-b6e1210de6d44bc9d6ff8df00dd4a5d4). I'm not sure when they anticipate removing it, but we should probably update to PointCloud2 at some point.

I don't have time to work on this right yet, but figured I would give it some visibility, and maybe work on a PR this fall.

SteveMacenski, 2020-08-25:
The 3 instances I can find are:
- [ ] DWB publisher 
- [ ] Costmap cloud publisher 
- [ ] Costmap voxel layer clearing ends

laiaaa0, 2021-01-20:
I was looking through the code and I saw a todo by @crdelsey to convert from pc to pc2. I'd like to have a go at this, any tips or resources I could check for how to do the conversion correctly ?

mikeferguson, 2021-01-21:
One example piece of code that I know of that uses the point cloud iterator to fill an XYZ point cloud in ROS2 is here: https://github.com/mikeferguson/ubr_reloaded/blob/ros2/ubr1_navigation/src/depth_layer.cpp#L297

SteveMacenski, 2021-01-21:
Oh, yeah, that's a really clean example, thanks for sharing that!

mikeferguson, 2021-01-21:
I think I've been involved in about half of the public ROS code that uses sensor_msgs iterator over the past few years, so I knew I had some examples around.

saching13, 2021-03-01:
I will be shot at this over the next 2 or 3 weeks.

SteveMacenski, 2021-03-10:
@saching13 any updates on this? We've chatted in Slack a bit, but just checking in :smile:

saching13, 2021-03-10:
Hey, Sorry was little busy at work this week. 
I tested  Costmap cloud publisher. This was easy. 
Now trying to test `Costmap voxel layer clearing ends` followed by ` DWB publisher`

SteveMacenski, 2021-03-10:
Awesome! Can you share your branch? Super happy to hear :smiley:

saching13, 2021-03-11:
[Here](https://github.com/saching13/navigation2/tree/sachin-pointcloud2-fix) is the link to the branch. I think I have added some prints to check. will remove them once I test. 

And I was not able to test `Cost map cloud publisher` but encountered some issues.
When I subscribed to the `clearing_endpoints` topic I key getting the warning that the publisher is not activated.
I tried to change the state of the node using service from the command line it didn't seem to work. 

And also the publisher is initialized with rclcpp publisher but the publisher is created and assigned from the lifecycle node. Just wanted to confirm if that is fine?

SteveMacenski, 2021-03-11:
If a publisher is made from a lifecycle node (even if the publisher isn't a lifecycle publisher object in the class variable definition), it gets assigned to a lifecycle manager. Since `rclcpp::Publisher` is the base class to `rclcpp_lifecycle::Publisher`, that conversion is valid (and annoying). 

So you'll have to `activate()` the publisher like you would for a normally declared lifecycle publisher

saching13, 2021-03-12:
Yeah. that conversion is the reason why I couldn't activate it. 
I changed that publisher to  `rclcpp_lifecycle::Publisher` and was able to get it to work. and tested it. `clearing_endpoints` is also working now with that change. 

Will test `DWB Publisher` tomorrow.

saching13, 2021-03-13:
DWB Publisher is also working. 
This was checkout from foxy-devel. Should I move everything to main before PR ? 
Based on that I will do linting checks and create a PR.

SteveMacenski, 2021-03-14:
Awesome! Yes, main before a PR would be good since main is what will roll into Galactic where PC(1) is deprecated. Super happy to hear 😄

saching13, 2021-03-14:
Prefect. Will do.