ros-navigation/navigation2 #1971: nav2_bt_navigator navigate_to_pose action server is not preepmtable

naiveHobo, 2020-09-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- ROS2 Version:
  - Foxy source
- Version or commit hash:
  - master

#### Steps to reproduce issue
Use the following behavior tree which implements the most basic navigation logic; compute path to pose and follow path:

```xml
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <Sequence name="NavigateWithReplanning">
      <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
      <FollowPath path="{path}"  controller_id="FollowPath"/>
    </Sequence>
  </BehaviorTree>
</root>
```

Send a navigation goal through rviz and the bot should start executing a path to this goal. Send another goal before the previous goal is reached.

#### Expected behavior
The previous goal should be preempted and the bot should start following a path to the new goal.

#### Actual behavior
The bot continues to move towards the previous goal and the new navigation goal is lost.

SteveMacenski, 2020-09-02:
I just created a clean workspace with only navigation2 main branch in it. I ran `ros2 launch nav2_bringup tb3_simulation_launch.py headless:=True` and then commanded through rviz to go to a location. I then, without stopping the robot, used the 2D Nav goal tool to command it to 3 other locations that the planner responded to and the robot followed. 

I think that shows that the navigation2 stack is working as we expect, that maybe there's something odd with the choice of BT for that simple application. 

I took that Bt and changed it:

```

<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <PipelineSequence name="NavigateWithReplanning">
      <RateController hz="1.0">
        <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
      </RateController>
      <FollowPath path="{path}"  controller_id="FollowPath"/>
    </PipelineSequence>
  </BehaviorTree>
</root>
```

to include a rate controller on compute path to pose, and it works then. Notable changes are `PipelineSequence` and the `RateController`. That's interesting that we see this issue. 

I added a print in `ComputePathToPoseAction::on_success` and see without the rate controller, I see that at 100hz (tick rate), which tells me that something is giving a result (though may be hold and async issues). So I made a debug print into an info in the planner server and cranked up the throttling from 1hz to 10hz. I see it preempting fine and on_success + planner server logging shows new plans every cycle.

I throttled the rate controller to 100hz, and I see the same thing, a bunch of on success prints matched with logs from the planner server that things are going as intended. Same with 1kHz. So I don't think its the rate at which we're calling the compute path to pose, maybe its the return code `SUCCESS` with `Sequence` vs `SequencePipeline`? Even when I have it rate controlled back down to only 1 hz (easily done) , it fails with `Sequence` to preempt. So I don't think from that evidence there's an issue (probably) with the planner server or the compute path to pose. 

I think that leaves the return code interaction differences between Sequence and Sequence Pipeline. Sequence isn't reactive by default in BT.CPP, so once a node is ticked and returns success, it won't again retick it until the follow path returns success (which only happens once its completed the navigation task). You need to use a `ReactiveSequence` so that it re-ticks successful nodes in tree runs (https://www.behaviortree.dev/sequencenode/). I find that with the reactive sequence, things work as expected.

So... the tl;dr of all that looking is that I think you're just using the wrong BT node, use `ReactiveSequence`, using just `Sequence` has the expected behavior because you've made it impossible for the compute path to pose BT node to be ticked again in the s ame navigation request. So therefore it's logical that you can't preempt it.

SteveMacenski, 2020-09-02:
So, fun puzzle, I think you just assumed the Sequence was reactive by default. Using the reactive sequence should resolve all of your issues.

naiveHobo, 2020-09-02:
I intentionally used Sequence instead of a reactive version of Sequence. The idea here was to show that the preemption logic in `nav2_bt_navigator` is solely dependent on the reticking of `ComputePathToPose` BT node. The `ComputePathToPose` node is reticked only when you want your behavior tree to implement some sort of a replanning logic, but that isn't necessary for every behavior tree right?

The reason why I used the BT I mentioned in the description was, what if we don't have replanning in a behaviour tree? I used the simplest navigation logic (plan path to goal once and follow this path) to show that the preemption logic doesn't actually live in `nav2_bt_navigator` which I think is an issue. Even with this basic behavior tree, I would expect the robot to stop pursuing the previous goal and start moving towards the new goal.

Even if we use a reactive version of the Sequence node and make sure `ComputePathToPose` is ticked again to enable replanning, it brings up another issue. What happens if you send a `NavigateToPose` goal with a different behavior tree (provide path to another BT file in the `behavior_tree` field in the action goal instead of leaving it empty and using the default BT) while the previous goal is still executing? With the current preemption logic in `nav2_bt_navigator`, only the goal is updated on the blackboard and the old behavior tree will be used to navigate to this new goal instead of loading the behavior tree that was actually requested by the new goal.

SteveMacenski, 2020-09-02:
> The idea here was to show that the preemption logic in nav2_bt_navigator is solely dependent on the reticking of ComputePathToPose BT node

I would say that that makes complete sense to me. If your behavior tree isn't "reactive" (sequence, or in any other control flow type) you can't expect your navigation behavior to be _reactive_ to dynamic changes. Its kind of like behavior tree node "latching" to focus on the current set of things until its complete, then to reprocess. That's the implicit meaning of a non-reactive behavior tree. 

I do, however, see the oddness that the "navigation reactiveness" is on the back of the "compute path". I don't think there's a way around that. Even if we had another BT node that dealt specifically with preemption requests at the beginning of each tree loop, if its unreactive, once it clears it once, it'll not be ticked to even try again  - same as now. 

How would you put it in BT navigator? I'd be open for a discussion on how that could be added there instead. However, all the ways I can think about it would involve canceling it, which isn't then preemption because we stop the entire navigation task and restart it - rather than having a smooth transition between goals. 

> what if we don't have replanning in a behaviour tree?

There isn't one now, but we could create a `tick once` node to wrap compute path to pose. Or the plugin could plan only once and then return the same plan over and over again. I think the BT node is better though. Though the tree you've made does that; it won't allow for replanning to a new goal, and you could think of it as attempts to replan to a new goal are simply rejected. Though its not an intuitive result.

naiveHobo, 2020-09-03:
> However, all the ways I can think about it would involve canceling it, which isn't then preemption because we stop the entire navigation task and restart it - rather than having a smooth transition between goals.

I completely agree with you here. But I think since navigation2 provides the option of choosing the behavior tree to run using the `behavior_tree` field of `NavigateToPose` action, every new navigation goal should have the ability to create a new tree and then use this behavior tree to achieve the new goal. Currently, a new behavior tree is loaded only for a goal that arrives when no previous goal is running. When a goal is already running and a new goal arrives, the current BT will be used to navigate to the new goal and the requested `behavior_tree` is just overlooked.

We don't notice this issue since we're using rviz to send navigation goals because of which it's always the default BT that's requested. This issue will become prominent when there are more complex behavior trees with multiple `NavigateToPose` nodes trying to run different navigation behaviors (like in the discussion happening in #1928).

I'm not sure what's the right solution here. We can make the "navigation reactiveness" independent of `ComputePathToPose` and also provide the ability to load a new behavior tree when a preempt is requested at the expense of losing the smooth transition between goals (cancelling the previous goal is the only option here). And the second option is to keep the smooth transitions but lose the ability to load a new tree when a preempt is requested.

In my opinion, the second option outright overlooks an important navigation2 feature that allows loading different behavior trees. I would go for the first option here.

SteveMacenski, 2020-09-03:
Well if you have a different navigate to pose with a different behavior tree, I agree, that doesn't look like a preemption though because there's no continuity. 

I think in that case, when we get a "preemption" with a different behavior tree, we do need to actually treat it like a cancel and rebuild the tree. Really, I think that should actually throw an error and say "impossible, must cancel before a new request with a different BT". That way there's no even impression that you can preempt with a new BT, you **must** cancel and re-submit a request as a new run entirely. 

But that seems like a completely different problem than the one this ticket starts to talk about with just preempting with an unreactive BT.

naiveHobo, 2020-09-03:
> The solution for both may be intertwined, but we shouldn't conflate one with another while working out a solution.

Yup, makes sense, sorry!

Keeping the issue of loading different BTs aside, I still think the navigation preemption logic should be independent of the behavior tree that's being executed. In my opinion, making `ComputePathToPose` reactive should only be done in the context of replanning a path to the current goal, not for performing a task that's actually supposed to be handled by a parent action server. My intuition suggests that even with the most basic behavior tree like the one mentioned in the description, you should be able to achieve *at least* what the bare-bones ROS1 navigation stack could achieve. Add to that the fact that canceling the previous goal and recreating the BT on preemption is still pretty seamless in terms of moving from one goal to another, I feel it provides for a more intuitive logic of how one would assume the navigation2 stack works.

naiveHobo, 2020-09-03:
PS: Really don't mean to press on the loading different BTs issue on this ticket but here are my final thoughts :sweat_smile: 

I can live with the navigation preemption logic being tied to `ComputePathToPose` but treating the BT loading behavior differently for a goal when the action server is inactive and for a goal when a previous goal is already running just doesn't make sense to me. As a user I would expect all my action goals to behave the same way regardless of whether or not the action server is already active (unless the action server explicitly states that preemption is not available).

SteveMacenski, 2020-09-03:
The problem is that we allow people to reconfigure themselves into a bad state, if they don't understand how behavior trees work. I don't think preemption is the start or end to the number of potential issues people could run into with a bad BT.

With that said, I agree that having the preemption logic dependent on the compute path to pose is a little brittle. I'm open for ideas on how to enable that in in the BT navigator itself. Most other issues of this form, I'll probably just close the ticket with "fix your BT" but preemption I think is a core skill that should work regardless of the specifics of the BT, however, I can't think of a way that would do that without disrupting a running tree. I really do think though that if someone intentionally doesn't use a `reactive` control flow type, then they're going to have to live with `non-reactive` behavior. 

---

Separately, the new request with different BTs should be handled outright. That needs to be resolved. My intuition says if we preempt with a new BT, reject it. New BTs can only be introduced between runs, else its not really a preemption, its a reset.

SteveMacenski, 2020-10-06:
@naiveHobo any thoughts here?

naiveHobo, 2020-10-08:
> I really do think though that if someone intentionally doesn't use a reactive control flow type, then they're going to have to live with non-reactive behavior.

In retrospect, I agree with you here. With a non-reactive BT you're essentially introducing a loop in the navigation task that doesn't account for preemption. You will need a reactive BT to be able to respond to preemptions, otherwise the only option is to halt the current BT and build a new BT when a new goal arrives.

About using the compute path to pose node to handle preemption, I think that's okay. One cleaner way of removing this logic from compute path to pose would be to use the goal updated condition node and have another branch which replans specifically when the goal is updated. This would be in addition to the current navigation branch which already includes a compute path to pose node. Though, I don't see too much value in doing this, it adds about 3-4 nodes per my estimation to do the task that's already being handled pretty well using compute path to pose.

This ticket can be closed if you're okay with keeping the preemption response in compute path to pose over having a set of nodes which specifically handle preemption.

> Separately, the new request with different BTs should be handled outright. That needs to be resolved. My intuition says if we preempt with a new BT, reject it. New BTs can only be introduced between runs, else its not really a preemption, its a reset.

I have some other issues regarding preemption with a different BT which I think I'll address in a ticket that's specific to that issue.

SteveMacenski, 2020-10-08:
> One cleaner way of removing this logic from compute path to pose would be to use the goal updated condition node and have another branch which replans specifically when the goal is updated. This [...]

One thing we could do is have the rate controller decorator turn into a rate controller condition node & have the failure condition have a subtree. So if we have a 1 Hz condition node, and when `true` it triggers planning, we could then have `false` tick the `if goal updated` node and if that outcome is true, tick `planning`. 

My head's been out of BTs for awhile, I don't know off hand actually how the 1 Hz decorator even allows for preemption if the replanning is delayed at 1hz which I don't see from using the stack. So maybe that logic is already implicitly taken care of on a more generic scale that isn't specific to the compute path to pose node. 

> I have some other issues regarding preemption with a different BT which I think I'll address in a ticket that's specific to that issue.

OK. 

I think we can close this - but if we want to decouple of the compute path to pose node from preemption, I'd be OK having that discussion.

naiveHobo, 2020-10-08:
> My head's been out of BTs for awhile, I don't know off hand actually how the 1 Hz decorator even allows for preemption if the replanning is delayed at 1hz which I don't see from using the stack. So [...]

Preemption is dependent on the re-ticking of compute path to pose. The 1hz rate controller does just that such that when a new goal arrives, the next time the rate controller allows compute path to pose to tick is when preemption is taken care of. Because we have the RateController node ticking ComputePathToPose at 1hz, there's usually not a visible delay in replanning a path to the new goal. But you'd notice it if you decrease the rate decorator to tick at say 0.1hz. In this case if a preempt is requested, the bot will continue to follow the old goal for 10 seconds until RateController allows ComputePathToPose to tick again and preemption to take effect.

> One thing we could do is have the rate controller decorator turn into a rate controller condition node & have the failure condition have a subtree. So if we have a 1 Hz condition node, and when true [...]

I like this. Plus there's already a BT node in navigation2 that throttles as a condition:
https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/plugins/condition/time_expired_condition.hpp

Though I think checking if the goal has been updated should be the first step. You'd want to replan as soon as you get a new goal. I imagine the navigation branch of the BT to look somewhat like the one described below.

The basic logic being:
If goal is not updated check if 1 second has passed. If goal is updated OR 1 second has passed, run the compute path to pose branch. Otherwise, always return running.

```xml
<PipelineSequence name="NavigateWithReplanning">
  <IfThenElse>
    <IfThenElse>
      <GoalUpdated/>
      <AlwaysSuccess/>
      <TimeExpired seconds="1"/>
    </IfThenElse>
    <RecoveryNode number_of_retries="1" name="ComputePathToPose">
      <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
      <ClearEntireCostmap name="ClearGlobalCostmap-Context" service_name="global_costmap/clear_entirely_global_costmap"/>
    </RecoveryNode>
    <AlwaysRunning/>
  </IfThenElse>
  <RecoveryNode number_of_retries="1" name="FollowPath">
    <FollowPath path="{path}" controller_id="FollowPath"/>
    <ClearEntireCostmap name="ClearLocalCostmap-Context" service_name="local_costmap/clear_entirely_local_costmap"/>
  </RecoveryNode>
</PipelineSequence>
```

SteveMacenski, 2020-10-08:
> Because we have the RateController node ticking ComputePathToPose at 1hz, there's usually not a visible delay in replanning a path to the new goal. But you'd notice it if you decrease the rate [...]

... is that true? If that's true, we should definitely work out a way to fix that. I don't want any delays in preemption, which I didn't think we had.

>  If goal is not updated check if 1 second has passed. If goal is updated OR 1 second has passed, run the compute path to pose branch. Otherwise, always return running.

I think either way works. The BT ticks stuff very, very fast so I doubt the << 1 millisecond delay is meaningful.  I'd say to do the one that's most compact or understandable.

SteveMacenski, 2020-10-14:
Updating the name to reflect the new nature of the ticket after discussion: dealing with BT changes in preemptions

mdtoom, 2020-10-17:
I am in favor of handling new BT's because if they would be rejected it would most likely mean that a user would abort the running  action and then send the new one. That is basically preemption with extra steps at the client side.
In #2039 I propose the following changes to reload when when a new action arrives (we might need to extend this so that it doesn't reload when the BT doesn't change):
```
@@ -240,7 +240,7 @@ BtNavigator::navigateToPose()
         return true;
       }
 
-      return action_server_->is_cancel_requested();
+      return action_server_->is_cancel_requested() || action_server_->is_preempt_requested();
     };
 
   auto bt_xml_filename = action_server_->get_current_goal()->behavior_tree;
@@ -260,11 +260,11 @@ BtNavigator::navigateToPose()
   std::shared_ptr<Action::Feedback> feedback_msg = std::make_shared<Action::Feedback>();
 
   auto on_loop = [&]() {
-      if (action_server_->is_preempt_requested()) {
-        RCLCPP_INFO(get_logger(), "Received goal preemption request");
-        action_server_->accept_pending_goal();
-        initializeGoalPose();
-      }
+      // if (action_server_->is_preempt_requested()) {
+      //   RCLCPP_INFO(get_logger(), "Received goal preemption request");
+      //   action_server_->accept_pending_goal();
+      //   initializeGoalPose();
+      // }
       topic_logger.flush();
 
       // action server feedback (pose, duration of task,
@@ -304,7 +304,14 @@ BtNavigator::navigateToPose()
 
     case nav2_behavior_tree::BtStatus::CANCELED:
       RCLCPP_INFO(get_logger(), "Navigation canceled");
-      action_server_->terminate_all();
+      if (action_server_->is_preempt_requested())
+      {
+          action_server_->terminate_current();
+      }
+      else
+      {
+          action_server_->terminate_all();
+      }
       break;
   }
 }
```
However this gives the side effect described in that issue. I still don't know why.
I am interested in what you think about this.

doisyg, 2021-02-26:
Jumping in as I have the same issue / need for nav2, and dropping my 2 cents.
We have a similar implementation of the bt_navigator in ROS1 which is too an action dedicated to go to a goal, and which behavior can be changed depending on the .xml file loaded (funny to discover that we both developed independently the same approach).
What we do is to halt the tree on preemption (no need to rebuild the tree after). As long as each node halts properly with (halt() override may needed, mostly for BT::AsyncActionNode), everything is smooth.
But by doing so we consider that preempting is similar to canceling (I see halt() at the BT equivalent of canceling). In my view this is the case for a navigation task: a new goal functionally cancels the previous one. But I can understand this is debatable.
However, what I experimented with on my forked bt_navigator is the following:
```
  void
  onPreempt()
  {
    RCLCPP_WARN(get_logger(), "Received goal preemption request");
    bt_action_server_->tree_.rootNode()->halt();

    initializeGoalPose(bt_action_server_->acceptPendingGoal());
  }

```
I had to pass tree_ as a public member. This seems to work and the previous goal receives the ABORTED result.

One alternative to explore would be to use `CoroActionNode` https://www.behaviortree.dev/tutorial_09_coroutines/ and yield. But I personally will use the halt approach on new goals. If somebody wants to experiment with `CoroActionNode` beware though that behavior_tree_cpp needs to be compiled from src for it to work (at least on my setup). Maybe related: https://github.com/BehaviorTree/BehaviorTree.CPP/issues/196

SteveMacenski, 2021-03-10:
I'd accept one for a public function `getTree()` and also a `haltTree()`

PR #2261 (2021-04-02): Handle navigate_to_pose preemptions with new BT

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | closes #1971 |
| Primary OS tested on | Ubuntu 20.04 |
| Robotic platform tested on | TurtleBot3 Gazebo simulation |

---

## Description of contribution in a few bullet points

* `bt_navigator` is configured to handle navigation preemption requests with a different BT than the one that the current running action is executing
* If a preemption request uses the same BT as the current action, current BT is not halted. The new goal is  [... cut]
closed 2021-04-01 by SteveMacenski