ros-navigation/navigation2 #1976: Costmap footprint parameter crashes whole navigation stack

simutisernestas, 2020-09-03:
## Bug report

**Required Info:**

- Operating System:
  - Distributor ID:	Ubuntu
  - Description:	Ubuntu 20.04.1 LTS
  - Release:	20.04
  - Codename:	focal
- ROS2 Version:
  - Foxy binaries
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Set footprint parameter as provided below (example from configuration guide).
```
global_costmap:
  global_costmap:
    ros__parameters:
      global_frame: map
      robot_base_frame: tank_base_link
      footprint: [ [0.1, 0.1], [0.1, -0.1], [-0.1, -0.1], [-0.1, 0.1] ]
      footprint_padding: 0.03
```

#### Expected behavior
Navigation starts properly, exactly like using the robot_radius parameter instead of footprint.

#### Actual behavior
Navigation crashes after footprint is set.

#### Additional information 
I've tried to get a backtrace according to this: https://navigation.ros.org/tutorials/docs/get_backtrace.html?highlight=trace, but GDB prompt doesn't appear. Below are some logs I've managed to produce.

Launch log: 

[INFO] [launch]: All log files can be found below /home/tank8/.ros/log/2020-09-03-11-33-21-308211-tank8-2761
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [2763]
[INFO] [planner_server-2]: process started with pid [2765]
[INFO] [recoveries_server-3]: process started with pid [2767]
[INFO] [bt_navigator-4]: process started with pid [2769]
[INFO] [waypoint_follower-5]: process started with pid [2771]
[INFO] [lifecycle_manager-6]: process started with pid [2773]
[lifecycle_manager-6] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[bt_navigator-4] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[bt_navigator-4] [ERROR] [1599122001.609633285] [rcl]: Failed to parse global arguments
[bt_navigator-4] 
[bt_navigator-4] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[bt_navigator-4] This error state is being overwritten:
[bt_navigator-4] 
[bt_navigator-4]   'Couldn't parse params file: '--params-file /tmp/tmpwf6xkpqw'. Error: Sequences cannot be key at line 142, at /tmp/binarydeb/ros-foxy-rcl-yaml-param-parser-1.1.7/src/parser.c:1615, at /tmp/binarydeb/ros-foxy-rcl-1.1.7/src/rcl/arguments.c:391'
[bt_navigator-4] 
[bt_navigator-4] with this new error message:
[bt_navigator-4] 
[bt_navigator-4]   'context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.7/src/rcl/context.c:51'
[bt_navigator-4] 
[bt_navigator-4] rcutils_reset_error() should be called after error handling to avoid this.
[bt_navigator-4] <<<
[bt_navigator-4] [ERROR] [1599122001.609674659] [rclcpp]: failed to finalize context: context is zero-initialized, at /tmp/binarydeb/ros-foxy-rcl-1.1.7/src/rcl/context.c:51
[waypoint_follower-5] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[waypoint_follower-5] [ERROR] [1599122001.610457408] [rcl]: Failed to parse global arguments
[waypoint_follower-5] 
[waypoint_follower-5] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[waypoint_follower-5] This error state is being overwritten:
[waypoint_follower-5] 
[waypoint_follower-5]   'Couldn't parse params file: '--params-file /tmp/tmp20ujz50s'. Error: Sequences cannot be key at line 142, at /tmp/binarydeb/ros-foxy-rcl-yaml-param-parser-1.1.7/src/parser.c:1615, at /tmp/binarydeb/ros-foxy-rcl-1.1.7/src/rcl/arguments.c:391'
[waypoint_follower-5] 
[waypoint_follower-5] with this new error message:
[... 345 lines cut ...]
^C--Type <RET> for more, q to quit, c to continue without paging--

Thread 1 "controller_serv" received signal SIGINT, Interrupt.
futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x555555afc038)
    at ../sysdeps/nptl/futex-internal.h:183
183	../sysdeps/nptl/futex-internal.h: No such file or directory.
(gdb) backtrace
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x555555afc038)
    at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x555555afc040, cond=0x555555afc010)
    at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=0x555555afc010, mutex=0x555555afc040) at pthread_cond_wait.c:638
#3  0x00007ffff7961e20 in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff72289bf in rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_events_t*, rmw_wait_set_t*, rmw_time_t const*) ()
   from /opt/ros/foxy/lib/librmw_fastrtps_shared_cpp.so
#5  0x00007ffff7d58517 in rcl_wait () from /opt/ros/foxy/lib/librcl.so
#6  0x00007ffff7ed1c12 in rclcpp::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) ()
   from /opt/ros/foxy/lib/librclcpp.so
#7  0x00007ffff7ed1f53 in rclcpp::Executor::get_next_executable(rclcpp::AnyExecutable&, std::chrono::duration<long, std::--Type <RET> for more, q to quit, c to continue without paging--
ratio<1l, 1000000000l> >) () from /opt/ros/foxy/lib/librclcpp.so
#8  0x00007ffff7ed8afd in rclcpp::executors::SingleThreadedExecutor::spin() () from /opt/ros/foxy/lib/librclcpp.so
#9  0x00007ffff7ed65a8 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) ()
   from /opt/ros/foxy/lib/librclcpp.so
#10 0x0000555555592469 in main (argc=<optimized out>, argv=<optimized out>) at ./src/main.cpp:24

SteveMacenski, 2020-09-03:
Lets avoid sensational title and describe better what you're actually seeing. Clearly, its just the yaml parsing that's unhappy. 

Did you put quotes around it, like the configuration guide shows? https://navigation.ros.org/configuration/packages/configuring-costmaps.html#costmap2d-ros-parameters

simutisernestas, 2020-09-04:
Sorry for such title. 

Works just fine with quotes around it. Would contribution to parse it like array of numbers be welcome?

SteveMacenski, 2020-09-04:
The ros2 parameters api doesnt support this type, it must be a string.

closed 2020-09-04 by SteveMacenski