ros-navigation/navigation2 #1979: Create a module for mapping from BT to Action

ymd-stella, 2020-09-03:
## Feature request

Related to https://github.com/ros-planning/navigation2/pull/1928#issuecomment-685995092

#### Feature description

- A module for mapping from BT to Action (like BtActionNode mapping from Action to BT::ActionNodeBase)
- It is similar to BtNavigator, but it applies to general actions.

#### Implementation considerations

- Inherit: nav2_util::LifecycleNode
- Template arg: Action type (ActionT)
- Functions:
  - Create action server (nav2_util::SimpleActionServer<ActionT>)
  - Load BT from file specified by goal
  - Provide the interfaces below
    - on_initialize_blackboard: Initialize blackboard (ex. Set goals)
    - on_bt_loop; Publish feedback by action server
    - on_bt_shutdown: Return result by action server

SteveMacenski, 2020-09-04:
Apologies, I don't understand the context / what your goal is here. The navigation to pose action can field a behavior tree in its request to use, so is that not already a BT to action call map?

I'm thinking your referencing our discussion on some kind of generic application server that people can use to build their specific autonomy applications, from your link and the implementation considerations, but I suppose the "feature description" section is a little less clear to me.

@mikeferguson do you have thoughts on this version of it? I don't have a strong vision for how this might turn out. Ideally, I think the best situation would be to create a server with a sufficiently generic action message definition that can be used for creating autonomy applications taking in a BT path, which I describe in that PR. This way its all about the BT and the action message population and a user doesn't have to implement and potentially fork the codebase to create a new server with their custom action template.

I think the template thing for the BT nodes works because the BTs are small runtime setable plugins, so a BT designer doesn't need to fork navigation2 to add a new BT plugin, they just need to build their own package of them and add to BT and it should work. This I think would require someone to modify this codebase since the template would be part of the server, not a smaller plugin component of it (but might be possible to make it that way?).

SteveMacenski, 2020-09-08:
Thanks for the update. Have you done any thinking on them as to which could be more implementable / what the `Something` message might be?

ymd-stella, 2020-09-08:
> which could be more implementable

Plan A is easier to implement because there are fewer unknowns in the implementation.

> what the Something message might be?

My idea was to define Something as a string-key-value-pair.
It is assumed to be set in blackboard as a string value and converted to the proper type by convertFromString when it is read.
My concern is that I don't know whether or not convertFromString is called when reading from Blackboard to BT xml.

ymd-stella, 2020-09-22:
@SteveMacenski  Can I get your feedback?

closed 2021-01-13 by SteveMacenski