ros-navigation/navigation2 #1983: NavigateToPose action server bugs

alsora, 2020-09-11:
## Bug report

**Required Info:**

- Operating System:
  - Debian Buster on robot, Ubuntu 20.04 on laptop
- ROS2 Version:
  - Foxy
- DDS implementation:
  - cycloneDDS

#### Steps to reproduce issue

Hi, I'm experiencing some problems when using a client for the NavigateToPose action.

I'm running the navigation stack on a RaspberryPi connected to my robot.
The robot uses slam_toolbox to generate a map of the environment.
I want to send navigation goals to it from a ROS 2 node on my laptop.

I'm able to send goals from Rviz using the default 2D Goal Pose tool.

My understanding is that this tool sends the goal by publishing a PoseStamped message on the `goal_pose` topic.
https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/src/rviz_default_plugins/tools/goal_pose/goal_tool.cpp
Then a subscription in the BT navigator converts that into an action server goal
https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/src/bt_navigator.cpp#L333-L337

This works exactly as expected.
After sending a goal I can see 

```
[bt_navigator-7] [INFO] [1599833543.541138555] [bt_navigator]: Begin navigating from current location to (-2.14, -0.11)
[controller_server-4] [INFO] [1599833543.566888799] [controller_server]: Received a goal, begin computing control effort.
[controller_server-4] [INFO] [1599833543.668741361] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [1599833544.668735613] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [1599833545.668747565] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [1599833546.668731421] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [1599833547.768732245] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [1599833548.768758408] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [1599833549.768740105] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [1599833549.957532147] [DWBLocalPlanner]: Goal reached!
[controller_server-4] [INFO] [1599833549.957603220] [controller_server]: Reached the goal!
[bt_navigator-7] [INFO] [1599833549.993784818] [bt_navigator]: Navigation succeeded
```

However, I would like to use ROS 2 actions for this.
This would allow my node to monitor the execution of the goal, to stop it and to be notified when it succeeds (or fails).

#### Actual behavior

If I create a ROS 2 action client and I publish a goal, the BT navigator node ignores the goal location that I specify and it always drive towards (0, 0)

This is my code:
```
    nav2_msgs::action::NavigateToPose::Goal goal;
    goal.pose.pose.position.x = 1.0f;
    goal.pose.pose.position.y = 1.0f;
    goal.pose.pose.position.z = 0.0f;
    goal.pose.pose.orientation = nav2_util::geometry_utils::orientationAroundZAxis(0.0f);

    // Set callbacks to be notified about this asynchronous goal
    auto send_goal_options = NavigateActionClient::SendGoalOptions();
    send_goal_options.result_callback = std::bind(&MyNode::navigate_result_callback, this, _1);

    // Send the goal request
    RCLCPP_INFO(this->get_logger(), "Navigating to %f %f", goal.pose.pose.position.x, goal.pose.pose.position.y);
    m_navigate_client->async_send_goal(goal, send_goal_options);
```

This is what I see on the BT navigator node

```
[bt_navigator-7] [INFO] [1599833586.617336361] [bt_navigator]: Begin navigating from current location to (0.00, 0.00)
```

I also tried to use this nav2_test_utils node, obtaining the same result
https://github.com/fmrico/nav2_test_utils

```
./install/nav2_test_utils/lib/nav2_test_utils/nav2_client_util -2.14 -0.11
```

Moreover, I'm not notified about action feedbacks or action termination.
If I enable debug logs in my node I can see a lot of 

```
[nav2_navigate_to_pose_client.rclcpp_action]: Received feedback for unknown goal. Ignoring...
```
and eventually
```
[nav2_navigate_to_pose_client.rclcpp_action]: Received status for unknown goal. Ignoring...
```

Let me know if there are other information you may need

SteveMacenski, 2020-09-11:
What branch of nav2 are you using? It looks like you deleted that field. 

I've been working on main testing some new features for a bit and don't see this issue. 

Keep in mind, you haven't filled in the message even remotely completely, you can't expect defined behavior with undefined inputs :wink: I don't see a single instance in nav2 of a NavigateToPose request that doesn't fully fill in `geometry_msgs/PoseStamped pose`.

alsora, 2020-09-11:
I'm using the `foxy-devel` branch, no changes at all to the navigation stack code.
You are right, I should be filling timestamp and frame. Although I would expect different errors if that would be the main issue.
I will update my code with the missing pieces.

After the call to `async_send_goal`, my node goes back to spin. Is it required to explicitly spin on the future that is returned by that function?
Note that I'm receiving something from the action server
```
[nav2_navigate_to_pose_client.rclcpp_action]: Received feedback for unknown goal. Ignoring...
```

Lastly, I'm currently trying to debug this using the `nav2_client_util` node, which is much simpler and follows all your suggestions.
However, the behavior is unchanged
https://github.com/fmrico/nav2_test_utils/blob/master/src/nav2_client_util.cpp#L39-L84

SteveMacenski, 2020-09-11:
> Although I would expect different errors if that would be the main issue.

I would think so too, but lets see first and then if that's the issue, we can more specifically address that problem. My guess is we try to transform to a frame that's an empty string so it kicks back out an empty transform and we didn't check for an error code. Should be a 2 line fix to log and error and exit if that's the case. 

I can't speak for Francisco's tools. I've never used or looked at them.

alsora, 2020-09-11:
Well, I found the problem... 
The node sending the action goals was not using CycloneDDS, but it was using the default FastRTPS.

All the problems can probably be explained with some serialization/deserialization issues.

Thank you a lot for the help.

closed 2020-09-11 by alsora

SteveMacenski, 2020-09-11:
These are my favorite kind of issues, ones that don't require me to fix anything :wink: