ros-navigation/navigation2 #1987: Valgrind reports uninitialized variables in range test

wilcobonestroo, 2020-09-15:
## Bug report

- Operating System: Ubuntu 20.04 (in Virtual Box under Windows 10)
- ROS2 Version: Foxy source
- Version or commit hash: 79963a4267060e7b5b4730f3d96b757fe1aa4646
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue
```
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Debug
. install/setup.bash
valgrind ./build/nav2_costmap_2d/test/integration/inflation_tests_exec
```

#### Expected behavior
No errors detected.

#### Actual behavior
ERROR SUMMARY: 240 errors from 4 contexts

#### Additional information
```
==10800== Memcheck, a memory error detector
==10800== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==10800== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==10800== Command: ./build/nav2_costmap_2d/test/integration/range_tests_exec
==10800== 
Running main() from /opt/ros/foxy/src/gtest_vendor/src/gtest_main.cc
[==========] Running 3 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 3 tests from TestNode
[ RUN      ] TestNode.testClearingAtMaxRange
[INFO] [1600172254.653890039] [range_test_node]: 
	range_test_node lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1600172254.809128922] [range_test_node]: range: ALL as input_sensor_type given
[INFO] [1600172254.960455465] [range_test_node]: RangeSensorLayer: subscribed to topic /range/topic
==10800== Conditional jump or move depends on uninitialised value(s)
==10800==    at 0x53A5824: nav2_costmap_2d::RangeSensorLayer::updateCostmap(sensor_msgs::msg::Range_<std::allocator<void> >&, bool) (range_sensor_layer.cpp:374)
==10800==    by 0x53A4DEA: nav2_costmap_2d::RangeSensorLayer::processVariableRangeMsg(sensor_msgs::msg::Range_<std::allocator<void> >&) (range_sensor_layer.cpp:281)
==10800==    by 0x53A4B23: nav2_costmap_2d::RangeSensorLayer::processRangeMsg(sensor_msgs::msg::Range_<std::allocator<void> >&) (range_sensor_layer.cpp:238)
==10800==    by 0x53AE9E6: void std::__invoke_impl<void, void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&>(std::__invoke_memfun_deref, void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&) (invoke.h:73)
==10800==    by 0x53ADD45: std::__invoke_result<void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&>::type std::__invoke<void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&>(void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&) (invoke.h:95)
==10800==    by 0x53ACFA9: void std::_Bind<void (nav2_costmap_2d::RangeSensorLayer::*(nav2_costmap_2d::RangeSensorLayer*, std::_Placeholder<1>))(sensor_msgs::msg::Range_<std::allocator<void> >&)>::__call<void, sensor_msgs::msg::Range_<std::allocator<void> >&, 0ul, 1ul>(std::tuple<sensor_msgs::msg::Range_<std::allocator<void> >&>&&, std::_Index_tuple<0ul, 1ul>) (functional:400)
==10800==    by 0x53ABC5F: void std::_Bind<void (nav2_costmap_2d::RangeSensorLayer::*(nav2_costmap_2d::RangeSensorLayer*, std::_Placeholder<1>))(sensor_msgs::msg::Range_<std::allocator<void> >&)>::operator()<sensor_msgs::msg::Range_<std::allocator<void> >&, void>(sensor_msgs::msg::Range_<std::allocator<void> >&) (functional:484)
==10800==    by 0x53AA00B: std::_Function_handler<void (sensor_msgs::msg::Range_<std::allocator<void> >&), std::_Bind<void (nav2_costmap_2d::RangeSensorLayer::*(nav2_costmap_2d::RangeSensorLayer*, std::_Placeholder<1>))(sensor_msgs::msg::Range_<std::allocator<void> >&)> >::_M_invoke(std::_Any_data const&, sensor_msgs::msg::Range_<std::allocator<void> >&) (std_function.h:300)
==10800==    by 0x53A83DE: std::function<void (sensor_msgs::msg::Range_<std::allocator<void> >&)>::operator()(sensor_msgs::msg::Range_<std::allocator<void> >&) const (std_function.h:688)
==10800==    by 0x53A4A75: nav2_costmap_2d::RangeSensorLayer::updateCostmap() (range_sensor_layer.cpp:229)
==10800==    by 0x53A5D3F: nav2_costmap_2d::RangeSensorLayer::updateBounds(double, double, double, double*, double*, double*, double*) (range_sensor_layer.cpp:444)
==10800==    by 0x582A010: nav2_costmap_2d::LayeredCostmap::updateMap(double, double, double) (layered_costmap.cpp:143)
==10800== 
==10800== Conditional jump or move depends on uninitialised value(s)
==10800==    at 0x53A5979: nav2_costmap_2d::RangeSensorLayer::updateCostmap(sensor_msgs::msg::Range_<std::allocator<void> >&, bool) (range_sensor_layer.cpp:383)
==10800==    by 0x53A4DEA: nav2_costmap_2d::RangeSensorLayer::processVariableRangeMsg(sensor_msgs::msg::Range_<std::allocator<void> >&) (range_sensor_layer.cpp:281)
==10800==    by 0x53A4B23: nav2_costmap_2d::RangeSensorLayer::processRangeMsg(sensor_msgs::msg::Range_<std::allocator<void> >&) (range_sensor_layer.cpp:238)
==10800==    by 0x53AE9E6: void std::__invoke_impl<void, void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&>(std::__invoke_memfun_deref, void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&) (invoke.h:73)
==10800==    by 0x53ADD45: std::__invoke_result<void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&>::type std::__invoke<void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&>(void (nav2_costmap_2d::RangeSensorLayer::*&)(sensor_msgs::msg::Range_<std::allocator<void> >&), nav2_costmap_2d::RangeSensorLayer*&, sensor_msgs::msg::Range_<std::allocator<void> >&) (invoke.h:95)
==10800==    by 0x53ACFA9: void std::_Bind<void (nav2_costmap_2d::RangeSensorLayer::*(nav2_costmap_2d::RangeSensorLayer*, std::_Placeholder<1>))(sensor_msgs::msg::Range_<std::allocator<void> >&)>::__call<void, sensor_msgs::msg::Range_<std::allocator<void> >&, 0ul, 1ul>(std::tuple<sensor_msgs::msg::Range_<std::allocator<void> >&>&&, std::_Index_tuple<0ul, 1ul>) (functional:400)
==10800==    by 0x53ABC5F: void std::_Bind<void (nav2_costmap_2d::RangeSensorLayer::*(nav2_costmap_2d::RangeSensorLayer*, std::_Placeholder<1>))(sensor_msgs::msg::Range_<std::allocator<void> >&)>::operator()<sensor_msgs::msg::Range_<std::allocator<void> >&, void>(sensor_msgs::msg::Range_<std::allocator<void> >&) (functional:484)
==10800==    by 0x53AA00B: std::_Function_handler<void (sensor_msgs::msg::Range_<std::allocator<void> >&), std::_Bind<void (nav2_costmap_2d::RangeSensorLayer::*(nav2_costmap_2d::RangeSensorLayer*, std::_Placeholder<1>))(sensor_msgs::msg::Range_<std::allocator<void> >&)> >::_M_invoke(std::_Any_data const&, sensor_msgs::msg::Range_<std::allocator<void> >&) (std_function.h:300)
==10800==    by 0x53A83DE: std::function<void (sensor_msgs::msg::Range_<std::allocator<void> >&)>::operator()(sensor_msgs::msg::Range_<std::allocator<void> >&) const (std_function.h:688)
==10800==    by 0x53A4A75: nav2_costmap_2d::RangeSensorLayer::updateCostmap() (range_sensor_layer.cpp:229)
==10800==    by 0x53A5D3F: nav2_costmap_2d::RangeSensorLayer::updateBounds(double, double, double, double*, double*, double*, double*) (range_sensor_layer.cpp:444)
==10800==    by 0x582A010: nav2_costmap_2d::LayeredCostmap::updateMap(double, double, double) (layered_costmap.cpp:143)
==10800== 
==10800== Conditional jump or move depends on uninitialised value(s)
==10800==    at 0x53A598A: nav2_costmap_2d::RangeSensorLayer::updateCostmap(sensor_msgs::msg::Range_<std::allocator<void> >&, bool) (range_sensor_layer.cpp:383)
[... cut ...]
==10800==    indirectly lost: 3,381 bytes in 49 blocks
==10800==      possibly lost: 0 bytes in 0 blocks
==10800==    still reachable: 36,699 bytes in 120 blocks
==10800==         suppressed: 0 bytes in 0 blocks
==10800== Rerun with --leak-check=full to see details of leaked memory
==10800== 
==10800== Use --track-origins=yes to see where uninitialised values come from
==10800== For lists of detected and suppressed errors, rerun with: -s
==10800== ERROR SUMMARY: 240 errors from 4 contexts (suppressed: 0 from 0)
```

SteveMacenski, 2020-09-15:
PRs appreciated

wilcobonestroo, 2020-09-15:
This turned out to be one variable. I don't think it solves the unpredictability of the tests. But we are getting closer...

closed 2020-09-15 by SteveMacenski