ros-navigation/navigation2 #1989: Map Saver CLI test is failing

wilcobonestroo, 2020-09-15:
## Bug report

**Required Info:**

- Operating System: Ubuntu
- ROS2 Version: Foxy source
- Version or commit hash: 79963a4267060e7b5b4730f3d96b757fe1aa4646
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue
`colcon test`

#### Expected behavior
Tests succeeding

#### Actual behavior
Test MapSaverCLI.CLITest fails.

#### Additional information
The test below is giving the error. The mode fake_mode is wrong. The test checks for a return value of 0 but, it should be at least anything different than 0. What value should we check? Not equal to zero or to another specific value?

```
  RCLCPP_INFO(node->get_logger(), "Testing invalid mode...");
  command =
    std::string(
    "ros2 run nav2_map_server map_saver_cli --mode fake_mode");
  return_code = system(command.c_str());
  EXPECT_EQ(return_code, 0);
```

wilcobonestroo, 2020-09-15:
Now it is running fine. If I run the command `ros2 run nav2_map_server map_saver_cli --mode fake_mode` on the command line, I get an error. What should the result code be? (and how can I find that out?)

SteveMacenski, 2020-09-15:
https://github.com/ros-planning/navigation2/blob/main/nav2_map_server/src/map_saver/main_cli.cpp#L123

Fake mode defaults to `Trinary` so that code should be `0`.

wilcobonestroo, 2020-09-15:
Ok. Than it is fine. It now also succeeds on my machine. I don't know what the problem was.

closed 2020-09-15 by wilcobonestroo